lwfinger/rtw88 #3: ant_sel get ignored

donihalim, 2020-07-12:
I'm using rtl8723de and build the latest rtw88 driver. When I set `ant_sel=2` in modprobe.d directory, I got this from `dmesg`:
`rtw_8723de: unknown parameter 'ant_sel' ignored`

OS: Artix Linux
Kernel: 5.7.6-artix1-1

Thank You.

Lollixzc, 2020-07-12:
I think that there is not any parameter in the driver to choose the antenna. I don't know if it is necessary to add it, it depends on which laptop you have, because some crappy laptops do not have all the two antennas attached, so you need to change that parameter. I don't know if it is necessary in your laptop. You should see if the WiFi module has only one cable attached to it, but you need to open the laptop to see that. 

I also tried to apply this parameter when i load the module, and it gives me the same error, because it is a firmware option that has to be added, but I don't know if it is necessary.

lwfinger, 2020-07-12:
That is not an option coded into the driver. Is it needed? I have two copies of that PCIe device, and both have only ONE connector. Even HP cannot get that wrong!

As for module parameters, 'grep module_param *' will show you the list. Parameters lps_deep_mode, disable_msi, and disable_aspm are needed for some motherboards. Obviously beam-forming (support_bf) is meaningless for a single-antenna device, and the developers on the linux-wireless mailing list might ask you to set debug_mask to debug some special problem.

Do either of you have weak signal? If you are getting signals greater than -50 dBm, then the antenna is connected and is being used correctly.

Lollixzc, 2020-07-13:
On my test laptop, I have good signal, so I don't see the need to add this parameter. I don't know if anyone else has the problem of weak signal, I've never had any issue. 

I noticed that many time other guides refer to `ant_sel` parameter for slightly different pcie devices, like the `RTL8723BE` device(which is not included in this repository), so you must pay attention to the naming of your device, because not every device has the same parameters baked into the code, and are necessary.

teal77, 2020-07-15:
I have an HP laptop with rtl8723de. I can confirm that ant_sel is no longer required in the driver that comes with 5.8 kernel.

Previous rtlwifi_new driver required that setting or the wifi would be painfully slow.

donihalim, 2020-07-18:
Turn out, I don't need the ant_sel anymore, signal seems fine. Closed

closed 2020-07-18 by donihalim