lwfinger/rtw88 #6: 8723DE monitor mode stun(freezes) all system

narcoloki, 2020-07-26:
Hello, sorry for opening this issue. I have installed your driver, works well. But when i'm  airmon-ng start wlan0 all OS is freezing. May be i'm not understand wich drivers must to be blacklisted. Can you give me some advise what to do
lsmod | grep rtw
rtw_8723de    16384  0
rtw8723d        57344 1 rtw8723de
rtw_pci            28672 1 rtw8723de
rtw_core       188416 2 rtw8723d,rtw_pci
mac80211    905216 3 rtw_core,rtl8xxxu,rtw_pci
cfg80211      872448 4 rtw_core,mac80211,rtl8xxxu,r8188eu

uname -srvo
Linix 5.7.0-kali-amd64 #1 Debian 5.7.6-1kali2  (2020-07-01) GNU/Linux

lwfinger, 2020-07-26:
Although they have nothing to do with the RTL8723DE, you should not have both rtl8xxxu and r8188eu installed. If you really want to use r8188eu, then rtl8xxxu should be blacklisted, or vice versa.

For the rtl8723de, driver rtw8723d is from the kernel, and not from this repo. If you want the newest driver, then it should be blacklisted. In that case, the ones you want are rtw_core,  rtw_pci, rtw_8723d, and rtw_8723de.

I have no idea if packet injection will work. I have not tested it.

Qammarbhat, 2020-08-18:
was your issue resolved? I have the same problem
here is output of lsmode | grep 
`rtw_8723de             16384  0
rtw_8723d              57344  1 rtw_8723de
rtw_pci                28672  1 rtw_8723de
rtw_core              196608  2 rtw_8723d,rtw_pci
mac80211              905216  2 rtw_core,rtw_pci
cfg80211              872448  3 wl,rtw_core,mac80211`

lwfinger, 2020-08-18:
Your blacklist file has an error. You need to fix that.

As for the system freeze when using monitor mode, that is not a problem that I can fix. Look at the last paragraph of README.md and follow those instructions.

closed 2020-08-18 by lwfinger