lwfinger/rtw88 #7: Poor reception of 2.4 GHz wifi

bsmith7581, 2020-07-28:
I have a laptop with an rtl8821ce wifi adapter. 5GHz works great with this driver, but 2.4 GHz has very weak reception. It won't even pick up my home router's 2.4 GHz signal from the next room. My phone's wifi tethering supports only 2.4 GHz. The phone can be right next to my laptop and register weak signal with about 1 Mbps download speed and 4 Mbps upload. My laptop worked well on 2.4 GHz wifi with the rtl8821ce driver provided by tomaspinho, so I know it's not a hardware issue.

buboleck, 2020-07-28:
I guess your card is "RFE 4" as mine, same issue.

bsmith7581, 2020-07-28:
No, my card is RTL8821CE.

buboleck, 2020-07-28:
> No, my card is RTL8821CE.

[    2.369297] rtw_core: loading out-of-tree module taints kernel.
[    2.369482] rtw_core: module verification failed: signature and/or required key missing - tainting kernel
[    2.384989] rtw_8821ce 0000:01:00.0: enabling device (0000 -> 0003)
[    2.386678] rtw_8821ce 0000:01:00.0: Firmware version 24.5.0, H2C version 12
[    2.410230] rtw_8821ce 0000:01:00.0: rfe 4 isn't supported
[    2.410334] rtw_8821ce 0000:01:00.0: failed to setup chip efuse info
[    2.410336] rtw_8821ce 0000:01:00.0: failed to setup chip information

Check your dmesg please

lwfinger, 2020-07-28:
@bsmith7581: Does your dmesg show that yo have an rfe 4 as shown above?

Run the command 'iw dev | grep Interface'. Using the device name shown as <name> below, run the command
 sudo iw dev <name> scan | egrep "signal|SSID|freq:"
Post the output.

bsmith7581, 2020-07-28:
My dmesg says nothing about rfe 4. 

`"
        freq: 5180
        signal: -59.00 dBm
        SSID: Rosewill06217C
        freq: 2412
        signal: -20.00 dBm
        SSID: Ben's phone hotspot
                 * SSID List
        freq: 5180
        signal: -69.00 dBm
        SSID: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
        freq: 5180
        signal: -66.00 dBm
        SSID: Blackcloud
                 * SSID List
        freq: 5180
        signal: -66.00 dBm
        SSID: xfinitywifi
        freq: 5180
        signal: -66.00 dBm
        SSID: 
                 * SSID List
        freq: 5180
        signal: -66.00 dBm
        SSID: XFINITY
                 * SSID List
        freq: 5180
        signal: -66.00 dBm
        SSID: 
                 * SSID List
        freq: 5745
        signal: -81.00 dBm
        SSID: 
                 * SSID List
        freq: 5745
        signal: -80.00 dBm
        SSID: PontaHome5GHZ
                 * SSID List
        freq: 5745
        signal: -80.00 dBm
        SSID: 
                 * SSID List
`

bsmith7581, 2020-07-28:
Ok so the above output shows strong signal for the 2.4 GHz network coming from my phone. Then why can I only get 0.52 Mbps down and 0.28 Mbps up on a speed test? With USB tethering, I get a minimum of 40 Mbps down and 20 Mbps up.

lwfinger, 2020-07-28:
Obviously it is not signal strength. On the other hand, -20 dBm is very strong and may be overloading the amplifiers. What happens if you double the distance to the phone?

Are all those hidden SSIDs at frequencies of 5180 and 5745 yours, or your neighbors? In either case, the configuration looks wrong.

bsmith7581, 2020-07-28:
The Rosewill is my home router, which also has a 2.4 GHz signal that my
laptop can't pick up in my room with this driver. The other SSIDs belong to
my neighbors.

I used to see many more 2.4 GHz SSIDs when I used tomaspinho's rtl8821ce
driver. Also I would get decent speeds on wifi tethering from my phone over
2.4 GHz.

>

lwfinger, 2020-07-28:
Why do you not use the other driver? I have not written any of them and I am only providing them. If you want this one improved, then read the last part of README.md and follow the instructions.

bsmith7581, 2020-07-28:
I just reinstalled Arch and decided to give this driver a try. I will probably be switching back to the other driver soon. 

Is there a github page for the people making this driver? I'd rather report on here than email them.

lwfinger, 2020-07-28:
No. If you are unwilling to expose your E-mail on a mailing list, create a new one just for that purpose.

Linux development is done through mailing list.

bsmith7581, 2020-07-28:
FYI, moving the phone slightly farther away did not improve the connection. I appreciate that you provided a way to try the newest Realtek wifi driver that will eventually be mainlined into the kernel. I just don't feel it is ready for general use yet. I'll close this issue now.

closed 2020-07-28 by bsmith7581