lwfinger/rtw88 #8: sudo make uninstall not working : make: *** [Makefile:112: uninstall] Error 1

shyamjos, 2020-07-30:
Hi ,

I installed this driver on my fedora 32 for testing purposes, But after testing when I tried to remove the driver using `make uninstall` its throwing below error

```
rm: cannot remove '/lib/modules/5.7.10-201.fc32.x86_64/kernel/drivers/net/wireless/realtek/rtw88/rtw_*.ko': No such file or directory
make: *** [Makefile:112: uninstall] Error 1
```
Also I noticed that my wireless stopped working after running  `make uninstall` , How can I copy back original rtw88 drivers shipped with my os  ?
PS: As a temporary fix I have switched to older kernel

lwfinger, 2020-07-31:
I fixed that error in Makefile and pushed the changes a few minutes ago.

With kernel 5.7.10, you should probably not be using this repo if you have either an RTL8822BE of RTL8822CE. The code here is the same as in the wireless_drivers_next repo, and what will be in kernel 5.8; however, the changes between that version and 5.7 are small.

If you are using the RTL8723DE device, the driver here is very different from any in a kernel.

To restore the old driver, use the command
sudo tar xPf backup_drivers.tar

If the file backup_drivers.tar does not exist, you will need to reinstall that kernel from the Fedora update repository.

shyamjos, 2020-07-31:
Thanks for the quick fix :+1:  I have already fixed the issue by upgrading my kernel.

My card is RTL8822BE and having disconnection issues when Bluetooth and wireless are used at the same time. This issue is the main reason why I try this repo every few months to check if issue got resolved.

Someone already reported this bug in kernel bugzilla but haven't received any response from kernel devs. It would be great if you can bring attention of any realtek devs about this strange disconnection issue. here is the bugzilla link : https://bugzilla.kernel.org/show_bug.cgi?id=208697

lwfinger, 2020-08-01:
There is a reason that my instructions in README.md do not mention the kernel bugzilla as a place to report problems. The devs do NOT read it. If you want to be sure that the devs see your complaint, the linux-wireless ML is the place to report problems!!

closed 2020-08-01 by lwfinger