lwfinger/rtw88 #10: RTL8822BE Not working on kernel 5.7.12

rhysperry111, 2020-08-06:
Hi,

My laptop has an RTL8822BE Wifi/bluetooth controller.

All througout the release of kernel 5.7 (almost every other subversion) my bluetooth has stopped showing up

Is this a known issue?

lwfinger, 2020-08-06:
No. What does dmesg say about that?

rhysperry111, 2020-08-06:
There doesn't seem to be anything about it in dmesg. I have attached the full log.

Would it help if I went back through all my 5.7.x kernels and tested to see which worked and which didn't?

[dmesg log](https://github.com/lwfinger/rtw88/files/5036478/dmesg.txt)

lwfinger, 2020-08-06:
It might help. Your log showed some USB errors. Please post the output of 'lsusb' and 'lsmod'.

rhysperry111, 2020-08-06:
It is a PCI device, so i have included `lspci -v`, `lsusb -v` and `lsmod`.
lspci -v:
```
00:00.0 Host bridge: Intel Corporation 8th Gen Core 4-core Processor Host Bridge/DRAM Registers [Coffee Lake H] (rev 07)
	Subsystem: Hewlett-Packard Company Device 8478
	Flags: bus master, fast devsel, latency 0
	Capabilities: [e0] Vendor Specific Information: Len=10 <?>
	Kernel driver in use: skl_uncore

00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 07) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 122
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 00005000-00005fff [size=4K]
	Memory behind bridge: b3000000-b40fffff [size=17M]
	Prefetchable memory behind bridge: 00000000a0000000-00000000b1ffffff [size=288M]
	Capabilities: [88] Subsystem: Hewlett-Packard Company Device 8478
	Capabilities: [80] Power Management version 3
	Capabilities: [90] MSI: Enable+ Count=1/1 Maskable- 64bit-
	Capabilities: [a0] Express Root Port (Slot+), MSI 00
	Capabilities: [100] Virtual Channel
	Capabilities: [140] Root Complex Link
	Capabilities: [d94] Secondary PCI Express
	Kernel driver in use: pcieport

00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 630 (Mobile) (prog-if 00 [VGA controller])
	DeviceName: Intel(R) UHD Graphics 630
	Subsystem: Hewlett-Packard Company Device 8478
	Flags: bus master, fast devsel, latency 0, IRQ 137
	Memory at b2000000 (64-bit, non-prefetchable) [size=16M]
	Memory at 80000000 (64-bit, prefetchable) [size=512M]
	I/O ports at 6000 [size=64]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: [40] Vendor Specific Information: Len=0c <?>
	Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
	Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable- 64bit-
	Capabilities: [d0] Power Management version 2
	Capabilities: [100] Process Address Space ID (PASID)
	Capabilities: [200] Address Translation Service (ATS)
	Capabilities: [300] Page Request Interface (PRI)
	Kernel driver in use: i915
	Kernel modules: i915

00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 07)
	Subsystem: Hewlett-Packard Company Device 8478
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Memory at b4610000 (64-bit, non-prefetchable) [size=32K]
	Capabilities: [90] MSI: Enable- Count=1/1 Maskable- 64bit-
	Capabilities: [d0] Power Management version 3
	Capabilities: [e0] Vendor Specific Information: Len=0c <?>
	Kernel driver in use: proc_thermal
	Kernel modules: processor_thermal_device

00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
	Subsystem: Hewlett-Packard Company Device 8478
	Flags: bus master, fast devsel, latency 0, IRQ 255
	Memory at b4620000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: [90] MSI: Enable- Count=1/1 Maskable- 64bit-
	Capabilities: [dc] Power Management version 2
	Capabilities: [f0] PCI Advanced Features

00:12.0 Signal processing controller: Intel Corporation Cannon Lake PCH Thermal Controller (rev 10)
	Subsystem: Hewlett-Packard Company Device 8478
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Memory at b4621000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: [50] Power Management version 3
	Capabilities: [80] MSI: Enable- Count=1/1 Maskable- 64bit-
	Kernel driver in use: intel_pch_thermal
	Kernel modules: intel_pch_thermal

00:14.0 USB controller: Intel Corporation Cannon Lake PCH USB 3.1 xHCI Host Controller (rev 10) (prog-if 30 [XHCI])
[... 1112 lines cut ...]
atkbd                  36864  0
libps2                 20480  2 atkbd,psmouse
aesni_intel           368640  6
crypto_simd            16384  1 aesni_intel
xhci_pci               20480  0
cryptd                 24576  3 crypto_simd,ghash_clmulni_intel
glue_helper            16384  1 aesni_intel
xhci_hcd              286720  1 xhci_pci
alcor_pci              20480  1 alcor
i8042                  32768  0
serio                  28672  6 serio_raw,atkbd,psmouse,i8042
raid6_pq              122880  1 btrfs
i915                 2617344  28
intel_gtt              24576  1 i915
i2c_algo_bit           16384  2 i915,nouveau
drm_kms_helper        253952  2 i915,nouveau
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
cec                    69632  2 drm_kms_helper,i915
rc_core                61440  1 cec
drm                   581632  12 drm_kms_helper,i915,ttm,nouveau
agpgart                53248  4 intel_gtt,ttm,nouveau,drm
```

lwfinger, 2020-08-06:
Please provide what I ask for - "lsusb", not "lsusb -v". I had to look through a lot of extra material. I did find what I wanted to know.

Although the wifi part of the RTL8822BE is PCI, the BT part is USB. I do not know what has happened, but the BT part does not show in the lsusb scan. As a result, the driver (btusb) was not loaded.

Your only USB devices are
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 04ca:707f Lite-On Technology Corp. HP Wide Vision HD Camera
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

rhysperry111, 2020-08-06:
I think I fixed the problem, although I am not aware of it's cause.

What I did was:
1. Power Off
2. Physically remove RTL8822BE
3. Power On
4. Put RTL8822BE back
5. Power on

closed 2020-08-06 by rhysperry111

lwfinger, 2020-08-06:
There was likely a contact problem.