lwfinger/rtw88 #12: rtl8723de hotspot :(

itzsnehasis, 2020-08-10:
thank you so much sir for this project , so much ! no linux has inbuilt driver for my wifi card , if you stop working on it then have to back to windows again :( , by the way I am using fedora 32 , and have succecfully installed the wifi driver from your repo , all working , but i cant make hotspot , fedora show that hotspot created by am not finding connections from others devices ,

BytEvil, 2020-08-11:
Hi. Same issue on rtl8822be chip.

itzsnehasis, 2020-08-14:
aww am sorry i have tried with some fedora expert advices and yes its working well , all working , very happy thank you for supporting rtl88xxx family , thank lwfinger

smekkley, 2020-09-09:
Same with 8821ce. 
As for 802.11ac, it looks like it's not even supported, which shouldn't be the case for 8821ce
```
hw vht capab: 0x3d07122, conf vht capab: 0x0
```

lwfinger, 2020-09-09:
Follow the last paragraph of README.md.

closed 2020-09-09 by lwfinger