lwfinger/rtw88 #15: modprobe: FATAL: Module rtl8723de not found.

naveenthakur0225, 2020-08-31:
while running this command getting module not found error. 
`(base) naveen@nv:~/rtw88$ sudo modprobe -r rtl8723de`

lwfinger, 2020-08-31:
That is NOT the name of any driver in this repo.

closed 2020-08-31 by lwfinger

naveenthakur0225, 2020-08-31:
http://ubuntuhandbook.org/index.php/2018/08/no-wifi-adapter-found-hp-laptops-ubuntu-18-04/
I have gone through this link and only this driver is mentioned here to install wifi package.
Please help me out with this, I am not able to use wifi.

lwfinger, 2020-08-31:
Are you sure you have an RTL8723DE? Use 'lspci -nn' to verify that.

The driver names in THIS repo are rtw_8723de.ko  rtw_8723d.ko  rtw_8821ce.ko  rtw_8821c.ko  rtw_8822be.ko  rtw_8822b.ko  rtw_8822ce.ko  rtw_8822c.ko  rtw_core.ko  rtw_pci.ko

Apparently you are using instructions that refer to an old repo and not looking at what is here.

If you have successfully done
make
sudo make install
sudo reboot

once the system comes up, the wifi should work.

naveenthakur0225, 2020-08-31:
Nope its not working either. 

this is the output of lspci -nn

``
00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers [8086:5914] (rev 08)
00:02.0 VGA compatible controller [0300]: Intel Corporation UHD Graphics 620 [8086:5917] (rev 07)
00:04.0 Signal processing controller [1180]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem [8086:1903] (rev 08)
00:14.0 USB controller [0c03]: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller [8086:9d2f] (rev 21)
00:14.2 Signal processing controller [1180]: Intel Corporation Sunrise Point-LP Thermal subsystem [8086:9d31] (rev 21)
00:15.0 Signal processing controller [1180]: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 [8086:9d60] (rev 21)
00:16.0 Communication controller [0780]: Intel Corporation Sunrise Point-LP CSME HECI #1 [8086:9d3a] (rev 21)
00:1c.0 PCI bridge [0604]: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 [8086:9d10] (rev f1)
00:1c.4 PCI bridge [0604]: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 [8086:9d14] (rev f1)
00:1c.6 PCI bridge [0604]: Intel Corporation Sunrise Point-LP PCI Express Root Port #7 [8086:9d16] (rev f1)
00:1f.0 ISA bridge [0601]: Intel Corporation Intel(R) 100 Series Chipset Family LPC Controller/eSPI Controller - 9D4E [8086:9d4e] (rev 21)
00:1f.2 Memory controller [0580]: Intel Corporation Sunrise Point-LP PMC [8086:9d21] (rev 21)
00:1f.3 Audio device [0403]: Intel Corporation Sunrise Point-LP HD Audio [8086:9d71] (rev 21)
00:1f.4 SMBus [0c05]: Intel Corporation Sunrise Point-LP SMBus [8086:9d23] (rev 21)
01:00.0 3D controller [0302]: NVIDIA Corporation GP108M [GeForce MX150] [10de:1d10] (rev a1)
02:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter [10ec:b822]
03:00.0 Non-Volatile memory controller [0108]: Sandisk Corp Device [15b7:5005] (rev 01)

``

lwfinger, 2020-08-31:
In your lspci output is "Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac", thus you have an 8822be. If you look at my previous comment, that driver is found in this repo. Do the "make" sequence I gave you.

naveenthakur0225, 2020-08-31:
(base) naveen@nv:~/rtw88$ sudo modprobe -r rtw_8822be
(base) naveen@nv:~/rtw88$ sudo modprobe rtw_8822be
modprobe: ERROR: could not insert 'rtw_8822be': Operation not permitted

I think this is the last error I am getting.

lwfinger, 2020-08-31:
Do you have secure boot enabled? That would prevent loading an external module.