lwfinger/rtw88 #16: An error is given out , Dunno much .. Please help

achallion, 2020-08-31:
[image]

It looks like a gcc error , I am not able to use wifi now
Thats so much of pain

Lollixzc, 2020-09-01:
What command have you launched on the console? It seems that there is a command-line option that it is not understood, you should usually do `git clone repo` then `make` to compile all the files

achallion, 2020-09-01:
[image]

It doesnt help also

lwfinger, 2020-09-01:
Your system is completely screwed up. I have no idea where that extra command-line option is coming from. It is a Kali problem - ask them.

achallion, 2020-09-01:
Ok 
Thank you

closed 2020-09-01 by achallion