lwfinger/rtw88 #17: GCC ERROR on kali linux 2020.3

Ajaysharma1703, 2020-09-02:
I cloned rtw88 (for rtl8723de) as you instructed and when I run make command it gives an error----

make -C /lib/modules/5.7.0-kali3-amd64/build M=/home/cooldude/rtw88 modules
make[1]: Entering directory '/usr/src/linux-headers-5.7.0-kali3-amd64'
  CC [M]  /home/cooldude/rtw88/main.o
gcc: error: unrecognized command-line option ‘--param=allow-store-data-races=0’; did you mean ‘--allow-store-data-races’?
make[3]: *** [/usr/src/linux-headers-5.7.0-kali3-common/scripts/Makefile.build:272: /home/cooldude/rtw88/main.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.7.0-kali3-common/Makefile:1745: /home/cooldude/rtw88] Error 2
make[1]: *** [/usr/src/linux-headers-5.7.0-kali3-common/Makefile:180: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.7.0-kali3-amd64'
make: *** [Makefile:80: all] Error 2

PLEASE HELP TO SORT THIS OUT!!

Lollixzc, 2020-09-02:
That's a problem not related to the driver but rather to the distro/system. Search your solution on Kali forums, not here. The drivers work fine on ubuntu and other distros. You can close this issue, as this issue has nothing to do with the driver.

Ajaysharma1703, 2020-09-02:
Thanks for such a quick reply!!!

closed 2020-09-02 by Ajaysharma1703