lwfinger/rtw88 #18: Compile Error in Kali

cmkeslin, 2020-09-03:
First, thank you so much for maintaining this repository.  I've been using the driver for quite some time and it's made my lenovo yoga run like a champ with Kali.

During the latest kernel upgrade I received the following gcc errors when running make

make -C /lib/modules/5.7.0-kali3-amd64/build M=/home/toor/rtw88 modules
make[1]: Entering directory '/usr/src/linux-headers-5.7.0-kali3-amd64'
  CC [M]  /home/toor/rtw88/main.o
gcc: error: unrecognized command-line option ‘--param=allow-store-data-races=0’; did you mean ‘--allow-store-data-races’?
make[3]: *** [/usr/src/linux-headers-5.7.0-kali3-common/scripts/Makefile.build:272: /home/toor/rtw88/main.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.7.0-kali3-common/Makefile:1745: /home/toor/rtw88] Error 2
make[1]: *** [/usr/src/linux-headers-5.7.0-kali3-common/Makefile:180: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.7.0-kali3-amd64'
make: *** [Makefile:80: all] Error 2

Any thoughts - I've not come accross this until this kernel.  Thanks again!

lwfinger, 2020-09-03:
It is a Kali problem. I have no idea where those gcc command-line options come from. I had one other Kali user report that same problem.

cmkeslin, 2020-09-03:
Perfect – thanks – that’s what I sort of figured out after working on it for a while.  I was able to get up and running with another driver until Kali fixes the headers for the latest kernel.

Cheers,
--Chris

mrizkimaulidan, 2020-09-05:
are there any fixes for this error?

mrizkimaulidan, 2020-09-05:
so are there any fixes for this issue?

lwfinger, 2020-09-05:
Have you asked Kali? I have used that compiler on openSUSE without ever running into that error, thus it is definitely a Kali problem!

closed 2020-09-05 by lwfinger

steev, 2020-09-06:
https://lore.kernel.org/patchwork/patch/472180/

mrizkimaulidan, 2020-09-06:
I have asked kali about this problem at here https://bugs.kali.org/view.php?id=6707, the problem is fixed now on mine. Just comment this line out at /usr/src/linux-headers-5.7.0-kali3-common/Makefile. Thank you so much for maintaining this repository sir, it helps alot!

[image]

bwhacks, 2020-09-08:
Kali is using the Debian kernel packaging. I think the problem is that:

- Debian kernel packaging sets a specific compiler version for building the kernel and all out-of-tree modules; currently gcc-9
- rtw88's Makefile sets CC = gcc; and that's now gcc 10

The result of this is that Kbuild will do compiler feature probing with gcc 9 but then rtw88 is built with gcc 10, and that no longer accepts this particular option.

Setting CC also breaks cross-compiling. So please don't set it.