lwfinger/rtw88 #19: RTL8723DE monitoring not working?

linuxindo, 2020-09-04:
after update to kernel 5.7.0-kali1
and got no wifi, i try use RTW88
and getting frezee use monitoring mode
before update kernel iam use rtlwifi_new and use monitor mode verry well?
any solution to use monitor mode again on this kernel?

lwfinger, 2020-09-04:
Not a problem I can help you with. Follow the instructions in the last paragraph of README.md.

closed 2020-09-04 by lwfinger

linuxindo, 2020-09-04:
Sorry for disgusting u, but iam really newer using linux. And thank u for helping me
But r the options look like this?
/etc/modprobe.d/rtw_8723.de
options rtw_8723de lps_deep_mode= 1
options rtw_8723de support_bf=Y 
options rtw_8723de debug_mask=1
options rtw_8723de disable_msi=Y 
options rtw_8723de disable_aspm=Y

Sorry 🙏😂
I try 1 by one and try it all still got freeze when starting monitor mode.
I spending 6 hours just for set modprobe 😂😅
And thank again for u best work to this repo.