lwfinger/rtw88 #20: RTL8821CE AP MODE not working

smekkley, 2020-09-10:
On kernel 5.9rc4
```
$ lsmod |grep -e rtw    
rtw88_8821ce           16384  0
rtw88_8821c            77824  1 rtw88_8821ce
rtw88_pci              28672  1 rtw88_8821ce
rtw88_core            147456  2 rtw88_pci,rtw88_8821c
mac80211             1019904  2 rtw88_pci,rtw88_core
cfg80211              876544  2 rtw88_core,mac80211
```  

APMode doesn't work is 80211.ac completely, 802.11g looks like it works from hostapd but SSID is not visible from any device I have.
```
hw vht capab: 0x3d07122, conf vht capab: 0x0
Completing interface initialization
Mode: IEEE 802.11a  Channel: 36  Frequency: 5180 MHz
DFS 0 channels required radar detection
Could not set channel for kernel driver
Interface initialization failed
wlp2s0: interface state COUNTRY_UPDATE->DISABLED
```

hostapd.conf (802.11ac)
```
# cat /etc/hostapd/ac.conf 
ssid=my-test-ap
wpa_passphrase=set-ap-password

country_code=US
# ieee80211d=1
# ieee80211h=1

interface=wlp2s0
driver=nl80211

wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

logger_syslog=0
logger_syslog_level=4
logger_stdout=-1
logger_stdout_level=0

hw_mode=a
ieee80211n=1
require_ht=1
ieee80211ac=1
require_vht=1

vht_oper_chwidth=1
channel=36
vht_oper_centr_freq_seg0_idx=42
```

Lollixzc, 2020-09-10:
This issue is not related to this repo. You should ask the maintainers of the chip to introduce the feature, as stated in the last paragraph of the [readme.md](https://github.com/lwfinger/rtw88/blob/master/README.md). I suggest you to close this issue.

smekkley, 2020-09-11:
I wonder if the source code is the same. Anyway I'll update it here once I get an update from them.

Lollixzc, 2020-09-11:
Yes, the source code is the same as the kernel. It is maintained in this repo for people who doesn't have the latest kernel.

jloqfjgk, 2020-12-31:
@smekkley 
How about checking the 5G channels (`iw list`) and switching to another channel?

smekkley, 2020-12-31:
Hi, Sorry forgot to update the issue.
Apparently there are variants with rtl8821ce. Only rev1 works. rev2 or higher isn't supported. So unofficial rtl8821 driver is recommended for now.

closed 2020-12-31 by smekkley