lwfinger/rtw88 #22: Current version is not buildable on Arch Linux?

mthw0, 2020-09-26:
I am on Arch Linux and I tried to build this, but for some reason I am unable. Here is the log:

> make -j6
make -C /lib/modules/5.8.10-zen1-1-zen/build M=/home/mthw/rtw88 modules
make[1]: Entering directory '/usr/lib/modules/5.8.10-zen1-1-zen/build'
  CC [M]  /home/mthw/rtw88/main.o
  CC [M]  /home/mthw/rtw88/phy.o
  CC [M]  /home/mthw/rtw88/mac.o
  CC [M]  /home/mthw/rtw88/tx.o
  CC [M]  /home/mthw/rtw88/efuse.o
  CC [M]  /home/mthw/rtw88/coex.o
/home/mthw/rtw88/main.c: In function ‘rtw_core_init’:
/home/mthw/rtw88/main.c:1623:2: error: implicit declaration of function ‘tasklet_setup’ [-Werror=implicit-function-declaration]
 1623 |  tasklet_setup(&rtwdev->tx_tasklet, rtw_tx_tasklet);
      |  ^~~~~~~~~~~~~
/home/mthw/rtw88/tx.c: In function ‘rtw_tx_tasklet’:
/home/mthw/rtw88/tx.c:600:27: error: implicit declaration of function ‘from_tasklet’ [-Werror=implicit-function-declaration]
  600 |  struct rtw_dev *rtwdev = from_tasklet(rtwdev, t, tx_tasklet);
      |                           ^~~~~~~~~~~~
  CC [M]  /home/mthw/rtw88/fw.o
/home/mthw/rtw88/tx.c:600:51: error: ‘tx_tasklet’ undeclared (first use in this function); did you mean ‘rtw_tx_tasklet’?
  600 |  struct rtw_dev *rtwdev = from_tasklet(rtwdev, t, tx_tasklet);
      |                                                   ^~~~~~~~~~
      |                                                   rtw_tx_tasklet
/home/mthw/rtw88/tx.c:600:51: note: each undeclared identifier is reported only once for each function it appears in
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:281: /home/mthw/rtw88/tx.o] Error 1
make[2]: *** Waiting for unfinished jobs....
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:281: /home/mthw/rtw88/main.o] Error 1
make[1]: *** [Makefile:1750: /home/mthw/rtw88] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.8.10-zen1-1-zen/build'
make: *** [Makefile:79: all] Error 2

I have been using this driver without issues for a while, but I just reinstalled my system and it doesn't work anymore. It literally worked yesterday.

mthw0, 2020-09-26:
This issue is caused by commit d868001. The last working version is 98e63c9.

lwfinger, 2020-09-26:
I failed to notice that kernel 5.9, which I use, changed the tasklet setup. The code is now testing for that case.

closed 2020-09-26 by lwfinger