lwfinger/rtw88 #23: Make error in function 'tasklet_setup'

gokhansolak, 2020-09-26:
I got a similar issue to #22, in Ubuntu 18.04 live usb. When I **make**:

```
make -C /lib/modules/5.4.0-42-generic/build M=/home/ubuntu/rtw88 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-42-generic'
  CC [M]  /home/ubuntu/rtw88/main.o
/home/ubuntu/rtw88/main.c: In function ‘rtw_core_init’:
/home/ubuntu/rtw88/main.c:1623:2: error: implicit declaration of function ‘tasklet_setup’; did you mean ‘twsk_net_set’? [-Werror=implicit-function-declaration]
  tasklet_setup(&rtwdev->tx_tasklet, rtw_tx_tasklet);
  ^~~~~~~~~~~~~
  twsk_net_set
cc1: some warnings being treated as errors
scripts/Makefile.build:273: recipe for target '/home/ubuntu/rtw88/main.o' failed
make[2]: *** [/home/ubuntu/rtw88/main.o] Error 1
Makefile:1731: recipe for target '/home/ubuntu/rtw88' failed
make[1]: *** [/home/ubuntu/rtw88] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-42-generic'
Makefile:79: recipe for target 'all' failed
make: *** [all] Error 2
```

I installed these beforehand:

```
sudo apt-get install make gcc git build-essential linux-headers-$(uname -r) dkms
```

gokhansolak, 2020-09-26:
When I checkout to [98e63c9](https://github.com/lwfinger/rtw88/commit/98e63c92e1f52d7cf6235c5069ac8a085a4830dc) it **make** successfully.

lwfinger, 2020-09-26:
I failed to notice that kernel 5.9, which I use, changed the tasklet setup. The code is now testing for that case. Please pull again.

closed 2020-09-26 by lwfinger

gokhansolak, 2020-09-26:
I can confirm that it can *make* now.