lwfinger/rtw88 #24: Fix build on Debian Buster

Floessie, 2020-09-30:
Hello,

here's a little patch to fix the build on Debian Buster (kernel 4.19):

```patch
diff --git a/main.h b/main.h
index c396c5e..6707abd 100644
--- a/main.h
+++ b/main.h
@@ -30,6 +30,15 @@
 #include <linux/etherdevice.h>
 #endif
 
+#if LINUX_VERSION_CODE < KERNEL_VERSION(5, 4, 0)
+#define fallthrough do {} while (0)
+#endif
+
+#if LINUX_VERSION_CODE < KERNEL_VERSION(5, 1, 0)
+#define static_assert(expr, ...) __static_assert(expr, ##__VA_ARGS__, #expr)
+#define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
+#endif
+
 #if LINUX_VERSION_CODE < KERNEL_VERSION(4, 7, 0)
 #define NUM_NL80211_BANDS IEEE80211_NUM_BANDS
 #endif
```

HTH,
Flössie

lwfinger, 2020-10-01:
Thanks. A variant of this is now applied.

closed 2020-10-01 by lwfinger

Floessie, 2020-10-01:
Hi Larry,

I'm sorry, but you missed the second part of the patch. Here's what I get now:

```
user@buster:~/rtw88$ make
make -C /lib/modules/4.19.0-11-amd64/build M=/home/user/rtw88 modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-11-amd64'
  CC [M]  /home/user/rtw88/main.o
  CC [M]  /home/user/rtw88/mac80211.o
  CC [M]  /home/user/rtw88/util.o
  CC [M]  /home/user/rtw88/debug.o
  CC [M]  /home/user/rtw88/tx.o
  CC [M]  /home/user/rtw88/rx.o
  CC [M]  /home/user/rtw88/mac.o
  CC [M]  /home/user/rtw88/phy.o
  CC [M]  /home/user/rtw88/coex.o
  CC [M]  /home/user/rtw88/efuse.o
  CC [M]  /home/user/rtw88/fw.o
  CC [M]  /home/user/rtw88/ps.o
  CC [M]  /home/user/rtw88/sec.o
  CC [M]  /home/user/rtw88/wow.o
  CC [M]  /home/user/rtw88/bf.o
  CC [M]  /home/user/rtw88/regd.o
  LD [M]  /home/user/rtw88/rtw_core.o
  CC [M]  /home/user/rtw88/rtw8822b.o
In file included from /usr/src/linux-headers-4.19.0-11-common/include/linux/list.h:9,
                 from /usr/src/linux-headers-4.19.0-11-common/include/linux/module.h:9,
                 from /home/user/rtw88/rtw8822b.c:5:
/usr/src/linux-headers-4.19.0-11-common/include/linux/kernel.h:72:25: error: expected declaration specifiers or '...' before '(' token
 #define ARRAY_SIZE(arr) (sizeof(arr) / sizeof((arr)[0]) + __must_be_array(arr))
                         ^
/home/user/rtw88/rtw8822b.c:2279:15: note: in expansion of macro 'ARRAY_SIZE'
 static_assert(ARRAY_SIZE(rf_para_tx_8822b) == ARRAY_SIZE(rf_para_rx_8822b));
               ^~~~~~~~~~
make[4]: *** [/usr/src/linux-headers-4.19.0-11-common/scripts/Makefile.build:309: /home/user/rtw88/rtw8822b.o] Error 1
make[3]: *** [/usr/src/linux-headers-4.19.0-11-common/Makefile:1545: _module_/home/user/rtw88] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-11-amd64'
make: *** [Makefile:79: all] Error 2
```

Best,
Flössie

lwfinger, 2020-10-01:
Sorry. Does a new pull now work?

Floessie, 2020-10-02:
Yes. Thanks! :+1: