lwfinger/rtw88 #25: Thank You!

bfree-github, 2020-10-19:
There doesn't seem to be a way to reach you, so I'm filing an issue :^)

I spent an entire day trying to get ubuntu to work with my realtek card, and after building about a dozen driver projects, I was about to give up.  Then I stumbled on a one-line mention of your rtw88 project and decided to give it a last try.

In just a couple of minutes I checked out your code, built it, and voila - everything works!  I wanted to thank you for creating/maintaining this lifesaver - you rock!

I consider this issue closed.

aphelionz, 2020-10-19:
This was my experience as well setting up my laptop a few months back :) thank you!

lwfinger, 2020-10-19:
You are both welcome.

closed 2020-10-19 by lwfinger