lwfinger/rtw88 #26: Not working on new Linux kernel

halveen, 2020-10-26:
there's a new linux kernel 5.8.0-2parrot1-amd64 released recently but this drivers don't work with it'
the drivers worked well for me on kernel 5.7.0-2parrot2-amd64

lwfinger, 2020-10-26:
What do you mean it does not work? Is there a build error, or is some error logged in the output of 'dmesg | less'? If the former, then you must do a git pull.

halveen, 2020-10-27:
It does not show my WiFi driver at all, I just see Ethernet and vpn connection I tried reinstalling the drivers but that did not fix it
I will check what 'dmesg | less' output is.

lwfinger, 2020-10-27:
ALWAYS do that dmesg check before you report an issue.

halveen, 2020-10-27:
what should I check for when i run dmesg

lwfinger, 2020-10-27:
Error messages!!!

halveen, 2020-10-27:
I uninstalled the kernel, re-installing it now if I still have the problem I will let you know.
Thanks for your help

halveen, 2020-10-27:
Uninstalled and re-installed the kernel problem solved, thank you.

closed 2020-10-27 by halveen