lwfinger/rtw88 #27: Not working on new kernel

Twijg76, 2020-12-07:
I have used this repo for my rtl8822be chip for a long time with few issues wjile updating my kernel every few days.
Now it doesn't seem to work for at least this version, maybe also the previous 1 or 2. I am running openSUSE with the 5.9.11 kernel.
When I do dmesg, I see following errors:

Trough the output, a couple of times the following error:
```
[   48.042555] failed to read DBI register, addr=0x0719
[   40.865802] WARNING: CPU: 5 PID: 63 at /home/twijg/gitprogrammas/rtw88/pci.c:1173 rtw_dbi_read8.constprop.0+0xa0/0xb0 [rtw_pci]

```
and at the end this a few times:
```
[ 3377.001805] rtw_8822be 0000:01:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[ 3377.001812] rtw_8822be 0000:01:00.0: mac power on failed
[ 3377.001813] rtw_8822be 0000:01:00.0: failed to power on mac
[ 3791.004408] rtw_8822be 0000:01:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[ 3791.004414] rtw_8822be 0000:01:00.0: mac power on failed
[ 3791.004415] rtw_8822be 0000:01:00.0: failed to power on mac
```

lwfinger, 2020-12-07:
For kernels as new as 5.9.11, I would expect you to use the in-kernel driver, rather than this repo. What happens when you do that? It is possible that my repo diverges from the kernel version.

Twijg76, 2020-12-08:
It does seem to be working at the moment (updated to 5.9.12 and *not* ran rtw88). I'll keep an eye on it. Thanks a lot for making my WiFi work for so long when it didn't with standard kernel code.

closed 2020-12-08 by Twijg76