lwfinger/rtw88 #28: Issue when upgrading kernel

seblint, 2020-12-08:
Hi,

First of all, thank you very much for your work, your driver is the only one working on ubuntu 20.04.1 for my Realtek 8822be card, it's a life saver.
I know you don't want to change the code for Ubuntu but i don't think the issue i stumble upon is ubuntu related.

Anyway, here's the thing. On a fresh install, i installed your driver as you instructed and it worked like a charm (after setting the disable_aspm option to Y for rtw_pci, this might be ubuntu related).

But, after the first kernel upgrade, even though i installed the drivers again as you instructed, they wouldn't load on boot or with modprobe because rtwpci was in use. After a lot of head scratching, we figured out that on install the distro's drivers (rtwpci and rtw88) were not removed from /lib/modules/$(KVER)/kernel/drivers/net/wireless/realtek/rtw88/ and backed up in ~/rtw88/backup_drivers.tar because that backup_drivers.tar file already existed from the previous install on the previous kernel.

I looked into the makefile and it appears to me that this eventuality is not taken into account, i think that a solution would be to replace in the makefile :

```
install: all
ifeq (,$(wildcard ./backup_drivers.tar))
	@echo Making backups
	@tar cPf backup_drivers.tar $(MODDESTDIR)
	@modprobe -rf rtwpci
	@modprobe -rf rtw88
	@rm -f $(MODDESTDIR)/rtw88*.ko*
	@rm -f $(MODDESTDIR)/rtwpci.ko*
endif
```

by

```
install: all
ifeq (,$(wildcard ./backup_drivers.tar))
	@echo Making backups
	@tar cPf backup_drivers.tar $(MODDESTDIR)
	@modprobe -rf rtwpci
	@modprobe -rf rtw88
	@rm -f $(MODDESTDIR)/rtw88*.ko*
	@rm -f $(MODDESTDIR)/rtwpci.ko*
else
	@echo Making backups
	@tar rPf backup_drivers.tar $(MODDESTDIR)
	@modprobe -rf rtwpci
	@modprobe -rf rtw88
	@rm -f $(MODDESTDIR)/rtw88*.ko*
	@rm -f $(MODDESTDIR)/rtwpci.ko*
endif
```

This is not a huge issue but i'm worried that it would create problems if one wants to use dkms to install your drivers.
Again, thank you very much for your work and i'm sorry if i'm making you waste your time with this.

lwfinger, 2020-12-08:
I have no idea what you have done. The in-kernel driver is rtw88_pci, and the one produced by my repo is rtw_pci. I think that rtwpci comes from Ubuntu. If you want to use this repo, you need to blacklist rtwpci and rtw88, not modify the Makefile in this repo.

Yes, I believe this issue is Ubuntu related.

seblint, 2020-12-09:
From what i gathered in my research, it seems that rtwpci comes from Realtek and it's causing problems to people who are not using Ubuntu. 
But the Makefile in this repo is designed to remove it from the drivers directory (/lib/modules/$(KVER)/kernel/drivers/net/wireless/realtek/rtw88/ or MODDESTDIR in the Makefile) so that one doesn't have to blacklist rtwpci.
The issue is that it only does so on first install of your driver, when reinstalling it after a kernel update, because backup_drivers.tar already exists, rtwpci is not removed from the new drivers directory and your driver stops working.

I agree that blacklisting rtwpci (and rtw88) is a perfectly fine solution (and it's the one i used before looking into the Makefile), but because it's not necessary on first install, people like myself or those who help on Linux forums, can struggle to find it.
Thus, what i'm suggesting is a little change in the Makefile so that rtwpci and rtw88 are removed from the drivers directory on every install of your driver, whether backup_drivers.tar exists or not.

(I'm sorry if i'm not making myself clear, english is not my first language and i'm not a trained computer engineer).

lwfinger, 2020-12-09:
OK, rtwpci may have come from Realtek, but it was NOT Realtek that put it on your computer. IT WAS YOUR DISTRO! I do not have that file on my computer as openSUSE does not provide such driver files!

I will modify the Makefile to delete those files even though blacklisting is a better solution. I will NOT recreate the backup file.

goverp, 2020-12-12:
Larry, sorry to but in and be the bearer of bad news, but that "fix" broke my installation for rtl8723de.  It's a Gentoo setiup, kernel 5.9.14, none of the RTL stuff in the kernel enabled.
in ~/rtw88 (after the latest "git pull"), "make" worked fine, but, err, "make install" fails with "modprobe: FATAL: Module rtwpci not found."  I'd not hit this message before; backing up the old drivers has never been relevant, and as it's my own config, and nothing to do with Ubuntu, there's no rtwpci or rtw88 to be removed.

I solved it by hacking the Makefile back to its previous version; I guess others may trip over this.  If deleting the unwanted modules remains in place, it needs to accept "module not found" errors.  I don't speak Makefile, so I've not attempted a patch.

Thanks for your patience, and persistence in supporting this set of drivers!

lwfinger, 2020-12-12:
Blame me. I accepted a patch from some user without proper testing. Do a new 'git pull' and it will work.

closed 2020-12-12 by lwfinger