lwfinger/rtw88 #29: RTL8822BE 5GHz AP mode not supported?

mid-kid, 2020-12-10:
Before the r8822be staging driver got removed with linux 5.2, I could use 5GHz AP mode with this adapter just fine. Now, `hostapd` fails with the following message:
```
Channel 36 (primary) not allowed for AP mode, flags: 0x853 NO-IR
wlan0: IEEE 802.11 Configured channel (36) not found from the channel list of current mode (2) IEEE 802.11a
wlan0: IEEE 802.11 Hardware does not support configured channel
Could not select hw_mode and channel. (-3)
```

Is 5GHz AP mode not supported anymore with this driver?

mid-kid, 2020-12-10:
Out of curiosity, I've applied this patch (I have no clue where `reg_rule->flags` is being set):
```
--- a/regd.c
+++ b/regd.c
@@ -283,7 +283,7 @@ static void rtw_regd_apply_beaconing_flags(struct wiphy *wiphy,
 
                        ch->flags &= ~IEEE80211_CHAN_DISABLED;
 
-                       if (!(reg_rule->flags & NL80211_RRF_NO_IR))
+                       //if (!(reg_rule->flags & NL80211_RRF_NO_IR))
                                ch->flags &= ~IEEE80211_CHAN_NO_IR;
                }
        }
```
And hostapd works like it used to, now, with devices connecting to it just fine. Of course, this isn't an acceptable solution, but it convinces me that it's not a hardware limitation.

lwfinger, 2020-12-10:
Was it just the low-band of 5G that failed? How about channels 120, or 157? Do they also fail with the unaltered code?

IEEE80211_CHAN_NO_IR means "do not initiate radiation, this includes  sending probe requests or beaconing."

Is channel 36 allowed in your regulatory zone?

mid-kid, 2020-12-11:
Running `iw list` tells me the "No IR" thing for all 5GHz frequencies, not a single channel is allowed according to it. I'm in western europe, and my main router transmits over 5GHz just fine, though I'm not informed about regulations in my area.
How are the regulatory zones determined in the kernel? Is it based on some region setting, the hardware or something else? Maybe my kernel's misconfigured? I'm running `sys-kernel/gentoo-kernel-5.9.11`, which basically inherits [fedora's stock config](https://src.fedoraproject.org/rpms/kernel/raw/94a4277f8827d1b2c911deabe56e7d929dc93146/f/kernel-x86_64-fedora.config) with [minor tweaks](https://raw.githubusercontent.com/mgorny/gentoo-kernel-config/master/base.config).

lwfinger, 2020-12-11:
There is a regulatory daemon that supplies that information based on the region code supplied by the driver. Your card's EFUSE may be supplying a code other than EU. Check dmesg for any hints.

mid-kid, 2020-12-12:
`dmesg` didn't really give any hints aside from `cfg80211` loading some certificates. What did, however, was checking if there was any package with "regul" in the description, which led me to wireless-regdb and crda, which I'm sure you know (but I didn't) are the two packages involved in informing the kernel about these regulations.
Checking the archlinux package, it seems they call [a script](https://github.com/archlinux/svntogit-packages/blob/packages/crda/trunk/set-wireless-regdom) from udev when the module is loaded, to use `iw reg set <country code>`. Running this with the appropriate country code for my region fixed this issue without requiring any patches, but doesn't persist across reboots. On a fresh boot, `iw reg get` returns `country 00: DFS-UNSET`, both globally and for my device. To get a list of regions, the command `regdbdump /usr/lib/crda/regulatory.bin` can be used.

I wonder why Gentoo doesn't provide a method of setting the regulatory region, as neither any of the init scripts nor `netifrc` seem to have any option for this, but for now I'll just set it from an /etc/local.d script, and bring this issue up to the Gentoo devs.

Anyway, thanks for the hints.

closed 2020-12-12 by mid-kid

hamad12a, 2023-01-12:
@mid-kid How to execute that code? Is it a bash code or what?

lwfinger, 2023-01-12:
No, that is not a script. He modified and rebuilt his kernel. Given that you asked this question, I would suggest that you not try this method.

hamad12a, 2023-01-12:
@lwfinger Which kernel? (excuse my  curiosity; I'm enthusiastic)

lwfinger, 2023-01-12:
Have you installed the crda package? It contains the rules for your region. I have no idea what kernel he used. In addition, that patch violates the wireless rules for his jurisdiction. If I did that, the FCC police might be visiting me. There are rules for the 5 GHz band to keep the wifi users from interfereing with radar, etc.

hamad12a, 2023-01-12:
I'm aware of the consequences. Would you mind if you tell me how to reverse the code and make it work, at least the main steps to get there? Where do I learn about reversing driver codes? Appreciated.

---- lwfinger wrote ----

Have you installed the crda package? It contains the rules for your region. I have no idea what kernel he used. In addition, that patch violates the wireless rules for his jurisdiction. If I did that, the FCC police might be visiting me. There are rules for the 5 GHz band to keep the wifi users from interfereing with radar, etc.

lwfinger, 2023-01-12:
I want no trouble with the law.

mid-kid, 2023-01-12:
The kernel doesn't matter. To bypass the restrictions you can either look for that line in any recent linux kernel and comment it out, or build a kernel with `CONFIG_CFG80211_REQUIRE_SIGNED_REGDB=n` and make your own wireless-regdb regulatory.bin.

But you seem misguided. See my comment above about how `iw reg set <country>` worked to fix the allowed channels in my country, and try that first.

hamad12a, 2023-01-12:
@mid-kid Actually I tried but I think it didn't work for me!
Here is the footage.
I issued  the command `sudo iw reg get` and the output is

```
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0 (self-managed)
country DZ: DFS-UNSET
	(2402 - 2437 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
	(2422 - 2462 @ 40), (6, 22), (N/A), AUTO-BW, NO-80MHZ, NO-160MHZ
	(2447 - 2482 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-80MHZ, NO-160MHZ
	(5170 - 5190 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5190 - 5210 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5210 - 5230 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5230 - 5250 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5250 - 5270 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5270 - 5290 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5290 - 5310 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5310 - 5330 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5490 - 5510 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5510 - 5530 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5530 - 5550 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5550 - 5570 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5570 - 5590 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5590 - 5610 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5610 - 5630 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5630 - 5650 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5650 - 5670 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5670 - 5690 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5690 - 5710 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5710 - 5730 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5735 - 5755 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5755 - 5775 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5775 - 5795 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5795 - 5815 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5815 - 5835 @ 20), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-HT40PLUS, NO-80MHZ, NO-160MHZ, PASSIVE-SCAN
```
Now, the command `sudo iw reg set CA` and the previous command return

```
global
country CA: DFS-FCC
	(2402 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
	(5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR

phy#0 (self-managed)
country DZ: DFS-UNSET
	(2402 - 2437 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
	(2422 - 2462 @ 40), (6, 22), (N/A), AUTO-BW, NO-80MHZ, NO-160MHZ
	(2447 - 2482 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-80MHZ, NO-160MHZ
	(5170 - 5190 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5190 - 5210 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
[... 62 lines cut ...]
Creating a virtual WiFi interface... ap1 created.

ERROR: Your adapter can not transmit to channel 36, frequency band 5GHz.

Doing cleanup.. done
Config dir: /tmp/create_ap.wlp1s0.conf.bOQDi2SJ
PID: 291436
Network Manager found, set ap1 as unmanaged device... DONE
wlp1s0 is already associated with channel 6 (2437 MHz), fallback to channel 6
Creating a virtual WiFi interface... ap1 created.
RTNETLINK answers: Device or resource busy

ERROR: Maybe your WiFi adapter does not fully support virtual interfaces.
       Try again with --no-virt.

Doing cleanup.. done
Config dir: /tmp/create_ap.new1.conf.4qxIYdGG
PID: 295050
Network Manager found, set ap0 as unmanaged device... DONE
Creating a virtual WiFi interface... ap0 created.

ERROR: Your adapter can not transmit to channel 140, frequency band 5GHz.

Doing cleanup.. done
```

hamad12a, 2023-01-12:
> To bypass the restrictions you can either look for that line in any recent linux kernel and comment it out, or build a kernel with CONFIG_CFG80211_REQUIRE_SIGNED_REGDB=n and make your own wireless-regdb regulatory.db.

@mid-kid Could you provide more information on which file I have to edit?
My kernel version using `uname -r` is
`6.0.12-76060006-generic`
The source code directory embeds multiple file there `/usr/src/linux-headers-6.0.12-76060006-generic/kernel/` 
```
bounds.s         debug/           futex/           Kconfig.freezer  kcsan/           module/          sched/           
.bounds.s.cmd    dma/             gcov/            Kconfig.hz       livepatch/       power/           time/            
bpf/             entry/           gen_kheaders.sh  Kconfig.locks    locking/         printk/          trace/           
cgroup/          events/          irq/             Kconfig.preempt  Makefile         rcu/        
```

mid-kid, 2023-01-12:
Use `iw list` to figure out if your adapter supports the frequency and channel at all.

hamad12a, 2023-01-12:
> iw list
```
iw list
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 20
	max scan IEs length: 422 bytes
	max # sched scan SSIDs: 20
	max # match sets: 11
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports RSN-IBSS.
	Device supports AP-side u-APSD.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * P2P-client
		 * P2P-GO
		 * P2P-device
	Band 1:
		Capabilities: 0x11ef
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 4 usec (0x05)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps (short preamble supported)
			* 5.5 Mbps (short preamble supported)
			* 11.0 Mbps (short preamble supported)
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (22.0 dBm)
			* 2417 MHz [2] (22.0 dBm)
			* 2422 MHz [3] (22.0 dBm)
			* 2427 MHz [4] (22.0 dBm)
			* 2432 MHz [5] (22.0 dBm)
			* 2437 MHz [6] (22.0 dBm)
			* 2442 MHz [7] (22.0 dBm)
			* 2447 MHz [8] (22.0 dBm)
			* 2452 MHz [9] (22.0 dBm)
			* 2457 MHz [10] (22.0 dBm)
[... 184 lines cut ...]
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ VHT_IBSS ]: VHT-IBSS
		* [ RRM ]: RRM
		* [ SCAN_START_TIME ]: scan start timestamp
		* [ BSS_PARENT_TSF ]: BSS last beacon/probe TSF
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ FILS_MAX_CHANNEL_TIME ]: FILS max channel attribute override with dwell time
		* [ ACCEPT_BCAST_PROBE_RESP ]: accepts broadcast probe response
		* [ OCE_PROBE_REQ_HIGH_TX_RATE ]: probe request TX at high rate (at least 5.5Mbps)
		* [ OCE_PROBE_REQ_DEFERRAL_SUPPRESSION ]: probe request tx deferral and suppression
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ EXT_KEY_ID ]: Extended Key ID support
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

mid-kid, 2023-01-12:
Please refrain from pasting entire logs, use a pasting service if you need to share this much text.
And do your own diligence of trying to read it at all first.

hamad12a, 2023-01-12:
Sorry, I am a newbie and I'm not much aware of what all of the output means therefore I have to provide it all. Using pastebin is better to not condensate the discussion.

mid-kid, 2023-01-12:
Please google on how to build a kernel from source. The kernel-headers are just the headers, not the full source.
Also disabling the config option would just allow you to load a custom regdb, so you'd have to build that too.
You're in way over your head, and I'd advise leaving this be.

lwfinger, 2023-01-13:
I agree rhat generating a kernel would be beyond his abilities, at least at the moment.

That said, there is a problem with the regulatory information in the rtw_8852be. I also have an Intel 7260, and for a US regulatory domain, it shows the first 4 5GHz channels without the No IR annotation, but the 8852be does. I will contact Realtek, and see if we are missing something.

dubhater, 2023-01-13:
I don't know anything about this, but don't you need to set the COUNTRY field in create_ap_5.conf ?

mid-kid, 2023-01-13:
That's the same as using `iw reg set` globally. Wasn't aware of that option either, though.

hamad12a, 2023-01-13:
> I agree rhat generating a kernel would be beyond his abilities, at least at the moment.
>
> That said, there is a problem with the regulatory information in the rtw_8852be. I also have an Intel 7260, [...]

Why would it be so? If brief steps were provided I would re-generate the kernel. Anyway, it's in my to-do list.

mid-kid, 2023-01-13:
There's no brief steps and it's heavily distribution- and setup-dependent.

hamad12a, 2023-01-13:
It's okay. Why don't you describe the "heavily distribution- and setup-dependent" instead of stating it?

mid-kid, 2023-01-14:
Because it's not brief and you'll have to google.

lwfinger, 2023-01-16:
If you use 'sudo iw reg set US' (or whatever country you want), the PASSIVE-SCAN is no longer set, and 'iw list' no longer shows the "No IR" flags. Using "EU" fails, but individual countries such as "DE" or "FR" work.

I thought I did that last week, but I must have done it wrong.