lwfinger/rtw88 #30: RHEL8 rtw88 for 8821ce wifi card not working

Zaryob, 2020-12-14:
hi i am using rhel8.
The kernel I use is 4.18.0-240.1.1.el8_3.x86_64 from the kernel-lt package.

I also tried the kernel-ml (5.10 mainline kernel) package from the epel 8 repository, but I could not use it because I got the kernel panic.

I followed these steps. I initially built the source:
```
$ make
make -C /lib/modules/4.18.0-240.1.1.el8_3.x86_64/build M=/home/zaryob/rtw88 modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-240.1.1.el8_3.x86_64'
  CC [M]  /home/zaryob/rtw88/main.o
  CC [M]  /home/zaryob/rtw88/mac80211.o
  CC [M]  /home/zaryob/rtw88/util.o
  CC [M]  /home/zaryob/rtw88/debug.o
  CC [M]  /home/zaryob/rtw88/tx.o
  CC [M]  /home/zaryob/rtw88/rx.o
  CC [M]  /home/zaryob/rtw88/mac.o
  CC [M]  /home/zaryob/rtw88/phy.o
  CC [M]  /home/zaryob/rtw88/coex.o
  CC [M]  /home/zaryob/rtw88/efuse.o
  CC [M]  /home/zaryob/rtw88/fw.o
  CC [M]  /home/zaryob/rtw88/ps.o
  CC [M]  /home/zaryob/rtw88/sec.o
  CC [M]  /home/zaryob/rtw88/wow.o
  CC [M]  /home/zaryob/rtw88/bf.o
  CC [M]  /home/zaryob/rtw88/regd.o
  LD [M]  /home/zaryob/rtw88/rtw_core.o
  CC [M]  /home/zaryob/rtw88/rtw8822b.o
  CC [M]  /home/zaryob/rtw88/rtw8822b_table.o
  LD [M]  /home/zaryob/rtw88/rtw_8822b.o
  CC [M]  /home/zaryob/rtw88/rtw8822be.o
  LD [M]  /home/zaryob/rtw88/rtw_8822be.o
  CC [M]  /home/zaryob/rtw88/rtw8822c.o
  CC [M]  /home/zaryob/rtw88/rtw8822c_table.o
  LD [M]  /home/zaryob/rtw88/rtw_8822c.o
  CC [M]  /home/zaryob/rtw88/rtw8822ce.o
  LD [M]  /home/zaryob/rtw88/rtw_8822ce.o
  CC [M]  /home/zaryob/rtw88/rtw8723d.o
  CC [M]  /home/zaryob/rtw88/rtw8723d_table.o
  LD [M]  /home/zaryob/rtw88/rtw_8723d.o
  CC [M]  /home/zaryob/rtw88/rtw8723de.o
  LD [M]  /home/zaryob/rtw88/rtw_8723de.o
  CC [M]  /home/zaryob/rtw88/rtw8821c.o
  CC [M]  /home/zaryob/rtw88/rtw8821c_table.o
  LD [M]  /home/zaryob/rtw88/rtw_8821c.o
  CC [M]  /home/zaryob/rtw88/rtw8821ce.o
  LD [M]  /home/zaryob/rtw88/rtw_8821ce.o
  CC [M]  /home/zaryob/rtw88/pci.o
  LD [M]  /home/zaryob/rtw88/rtw_pci.o
  Building modules, stage 2.
  MODPOST 10 modules
  CC      /home/zaryob/rtw88/rtw_8723d.mod.o
  LD [M]  /home/zaryob/rtw88/rtw_8723d.ko
  CC      /home/zaryob/rtw88/rtw_8723de.mod.o
  LD [M]  /home/zaryob/rtw88/rtw_8723de.ko
  CC      /home/zaryob/rtw88/rtw_8821c.mod.o
  LD [M]  /home/zaryob/rtw88/rtw_8821c.ko
  CC      /home/zaryob/rtw88/rtw_8821ce.mod.o
  LD [M]  /home/zaryob/rtw88/rtw_8821ce.ko
  CC      /home/zaryob/rtw88/rtw_8822b.mod.o
  LD [M]  /home/zaryob/rtw88/rtw_8822b.ko
  CC      /home/zaryob/rtw88/rtw_8822be.mod.o
  LD [M]  /home/zaryob/rtw88/rtw_8822be.ko
  CC      /home/zaryob/rtw88/rtw_8822c.mod.o
  LD [M]  /home/zaryob/rtw88/rtw_8822c.ko
  CC      /home/zaryob/rtw88/rtw_8822ce.mod.o
  LD [M]  /home/zaryob/rtw88/rtw_8822ce.ko
  CC      /home/zaryob/rtw88/rtw_core.mod.o
  LD [M]  /home/zaryob/rtw88/rtw_core.ko
  CC      /home/zaryob/rtw88/rtw_pci.mod.o
  LD [M]  /home/zaryob/rtw88/rtw_pci.ko
make[1]: Leaving directory '/usr/src/kernels/4.18.0-240.1.1.el8_3.x86_64'
```

then I installed, 
```
make install
make -C /lib/modules/4.18.0-240.1.1.el8_3.x86_64/build M=/home/zaryob/rtw88 modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-240.1.1.el8_3.x86_64'
  Building modules, stage 2.
  MODPOST 10 modules
make[1]: Leaving directory '/usr/src/kernels/4.18.0-240.1.1.el8_3.x86_64'
Install rtw88 SUCCESS
```

This is modinfo output of 8821ce module
```
modinfo rtw_8821ce
filename:       /lib/modules/4.18.0-240.1.1.el8_3.x86_64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8821ce.ko.xz
license:        Dual BSD/GPL
description:    Realtek 802.11ac wireless 8821ce driver
author:         Realtek Corporation
rhelversion:    8.3
srcversion:     2D16A2C76B6B4216356C119
alias:          pci:v000010ECd0000C821sv*sd*bc*sc*i*
depends:        rtw_pci,rtw_8821c
name:           rtw_8821ce
vermagic:       4.18.0-240.1.1.el8_3.x86_64 SMP mod_unload modversions 
```

then I enabled it.
```
modprobe rtw_8821ce
```

this is dmesg output.
```
dmesg|grep 8821
[    9.251318] Bluetooth: hci0: RTL: examining hci_ver=08 hci_rev=000c lmp_ver=08 lmp_subver=8821
[    9.252309] Bluetooth: hci0: RTL: loading rtl_bt/rtl8821c_fw.bin
[    9.254182] Bluetooth: hci0: RTL: loading rtl_bt/rtl8821c_config.bin
[  292.834270] rtw_8821ce 0000:03:00.0: Firmware version 24.5.0, H2C version 12
[  292.851989] rtw_8821ce 0000:03:00.0: rfe 2 isn't supported
[  292.852083] rtw_8821ce 0000:03:00.0: failed to setup chip efuse info
[  292.852084] rtw_8821ce 0000:03:00.0: failed to setup chip information
```

I dont make any additional step for /etc/modprobe.d file etc.

Zaryob, 2020-12-14:
[image]
This is also latest dmesg output for 4.18 kernel

Zaryob, 2020-12-14:
Update: I tried `modprobe rtw_8821ce` on [image] dmesg output was the same as top.

I tried to start rtl8821ce card on opensuse. It was worked with rtw_8821ce

lwfinger, 2020-12-14:
Did you notice the line "[  292.851989] rtw_8821ce 0000:03:00.0: rfe 2 isn't supported?" As of this morning, Realtek has not submitted the fixes for rfe2 with 8821ce.

For future reference, this problem has been covered in several issues in this repo!

closed 2020-12-14 by lwfinger

alium, 2021-01-28:
question: your driver works not for 8821ce, or are 2 (or more) different revisions of 8821ce, when one revision works, and another not? because i have same error: rfe 2 isn't supported.

lwfinger, 2021-01-29:
Did you do any research on this problem? When you do, you will learn the current state.

alium, 2021-01-29:
> Did you do any research on this problem? When you do, you will learn the current state.

thank you for the very valuable answer

lwfinger, 2021-02-02:
rfe2 code is now available. You also need to reinstall the firmware.