lwfinger/rtw88 #31: make: *** [Makefile:84: install] Error 1

KewinOgink, 2020-12-18:
Hi,

I have a lenovo ideapad C340 and since the update to ubuntu 20.04 have had no wifi.
I follow the guide step by step, but at `sudo make install` I get the problem: `make: *** [Makefile:84: install] Error 1` 

Maybe this is of any help: 
output for `lspci -v`:
```
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter (rev ff) (prog-if ff)
	!!! Unknown header type 7f
	Kernel modules: rtw88_8822be, rtw_8822be
```

Any idea what I can do to get the driver properly installed?
Best,

Kewin

lwfinger, 2020-12-18:
Your lspci output says you already have two drivers installed. They are probably conflicting with each other. Driver rtw88_8822be comes from the in-kernel driver. Driver rtw_8822be comes from this repo. If you want to use the one from this repo (not recommended), you will need to blacklist the in-kernel drivers.

I had modified the Makefile because a user wanted some changes. That was a mistake, but I have fixed it. do a 'git pull' and this repo shoud install correctly.

KewinOgink, 2020-12-20:
Thank you very much for you answer. 
I don't understand why it is not recommended to use the rtw_8822be driver, because how else would I be able to fix the wifi issue?
Either way, I removed the repository, but still the kernel modules are there. I tried unloading them with `sudo modprobe -r rtw_8822be`, but that didn't seem to do anything. Please, do you have any idea how I can reset my wifi drivers?

lwfinger, 2020-12-21:
Your lsmod command is badly formatted, but clearly the rtw88_* drivers are not there; however, if they were not on your system, then lspci command would not see them.

The Linux kernel will load ALL drivers that it finds on the system for a given piece of hardware. In your case, that will not work. When rtw88_* and rtw_* drivers both exist, they will interfere with each other, and neither will work. If you want to use rtw_* (this repo), you will need to blacklist rtw88_8822be, rtw88_8822b, rtw88_pci, and rtw88_core. If you want to use the ones built into the kernel, uninstall the drivers from this repo.

If you are satisfied that you are loading ONLY the drivers you want, and the driver still does not work, then check the output of dmesg for clues.

KewinOgink, 2020-12-22:
Dear lwfinger,

I run into 2 problems:
-whether I do or do not blacklist rtw88_8822be, rtw88_8822b, rtw88_pci, and rtw88_core, the lspci -v output keeps saying the same thing as in the first post. I do not know how I can remove either module. I have been googling around a lot, but the only thing I saw, `sudo modprobe -r` does not seem to work. 
-I have (again) followed all the steps in your guide, but again get this same makefile issue as I have stated before, that prevents me from doing `sudo make install`.

Therefore, I am basically still stuck with the initial problem. 
I don't really understand what is going on. I just want my wifi, I don't care with which drivers.

I have no idea if this is of any help, but here is the `dmesg | grep rtw` output:
`sudo dmesg | grep rtw
[sudo] password for kewin: 
[    3.440443] Modules linked in: rtw_8822be(OE+) snd_seq fjes(-) rtw_8822b(OE) crct10dif_pclmul ghash_clmulni_intel aesni_intel crypto_simd cryptd snd_seq_device glue_helper snd_timer rtw_pci(OE) rapl rtw_core(OE) intel_cstate hid_sensor_accel_3d snd input_leds hid_sensor_trigger drm_kms_helper mac80211(OE) industrialio_triggered_buffer kfifo_buf cec mei_me rc_core hid_sensor_iio_common i2c_algo_bit cfg80211(OE) fb_sys_fops syscopyarea compat(OE) sysfillrect serio_raw efi_pstore wmi_bmof intel_wmi_thunderbolt soundcore libarc4 industrialio hid_multitouch(+) joydev sysimgblt processor_thermal_device(+) mei intel_rapl_common intel_pch_thermal intel_soc_dts_iosf ideapad_laptop(+) int3403_thermal sparse_keymap int340x_thermal_zone acpi_pad int3400_thermal acpi_thermal_rel mac_hid sch_fq_codel parport_pc ppdev lp drm parport ip_tables x_tables autofs4 hid_sensor_custom wacom usbhid hid_sensor_hub hid_generic rtsx_pci_sdmmc nvme crc32_pclmul xhci_pci xhci_pci_renesas ahci i2c_i801 intel_lpss_pci`

`[    3.440491]  rtw_pci_probe+0x49/0x8f0 [rtw_pci]
[    3.440519]  rtw_8822be_driver_init+0x23/0x1000 [rtw_8822be]
[    3.440549] rtw_8822be 0000:02:00.0: failed to allocate hw
[    3.440556] rtw_8822be: probe of 0000:02:00.0 failed with error -12`

lwfinger, 2020-12-23:
With regard to your 'sudo make install' problem, did you do a 'git pull?' I fixed that problem and updated the repo.

The problem indicated by [ 3.440491] rtw_pci_probe+0x49/0x8f0 [rtw_pci] [ 3.440519] rtw_8822be_driver_init+0x23/0x1000 [rtw_8822be] [ 3.440549] rtw_8822be 0000:02:00.0: failed to allocate hw [ 3.440556] rtw_8822be: probe of 0000:02:00.0 failed with error -12 is serious. It indicates no memory to allocate a the software struct for hw. I have never seen this problem.  Is your system severely memory limited? Post the output of 'free."

KewinOgink, 2020-12-23:
Yes I have done the git pull. 
I have a dual boot with windows and Linux. \
By the way thank you for helping out. I've been googling around for several weeks and I appreciate your help.

```
free -m
              total        used        free      shared  buff/cache   available
Mem:           7678        3119        1720         718        2838        3550
Swap:          2047           0        2047

```

KewinOgink, 2020-12-26:
[dmesg.txt](https://github.com/lwfinger/rtw88/files/5743089/dmesg.txt)
Here you go

mrizkimaulidan, 2020-12-29:
i have the exact same problem as you, i'am using the newest kali linux 2020.4 with 5.9.0 kernel and rtl8723de wifi module

Priya-Pandey1, 2020-12-29:
I am using the same for ubuntu 20.04 and rtl8723de module and the error is same while running make . Is there any thing for bluetooth cause my bluetooth is also not working in ubuntu 20.04

lwfinger, 2020-12-29:
Make shoulld now work. I know nothing of Bluetooth.

closed 2020-12-29 by lwfinger

KewinOgink, 2020-12-30:
Hi it is fixed, and lspci -v now shows only the rtw_8822be kernel module, but still the wifi tab is not there at the settings. I guess I will have to reinstall ubuntu.

Either way thank you for your help.