lwfinger/rtw88 #33: rtw_8822ce can't wake up from suspend

nshigit, 2020-12-27:
cat /etc/os-release 
NAME="Linux Mint"
VERSION="20 (Ulyana)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 20"
VERSION_ID="20"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=ulyana
UBUNTU_CODENAME=focal

inxi -Fxz
System:    Kernel: 5.8.18-050818-generic x86_64 bits: 64 compiler: N/A Desktop: Xfce 4.14.2 
           Distro: Linux Mint 20 Ulyana base: Ubuntu 20.04 focal 
Machine:   Type: Laptop System: HUAWEI product: BOHL-WXX9 v: M1020 serial: <filter> 
           Mobo: HUAWEI model: BOHL-WXX9-PCB v: M1020 serial: <filter> UEFI: HUAWEI v: 1.05 
           date: 07/20/2020 
Battery:   ID-1: BAT1 charge: 39.8 Wh condition: 39.8/41.6 Wh (96%) model: Desay HB4692J5ECW-31 
           status: Unknown 
           Device-1: hidpp_battery_0 model: Logitech Wireless Mouse M325 
           charge: 50% (should be ignored) status: N/A 
CPU:       Topology: 6-Core model: AMD Ryzen 5 4500U with Radeon Graphics bits: 64 type: MCP 
           arch: Zen rev: 1 L2 cache: 3072 KiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm bogomips: 28446 
           Speed: 1772 MHz min/max: 1400/2375 MHz Core speeds (MHz): 1: 1771 2: 3984 3: 1773 
           4: 3983 5: 3983 6: 1772 
Graphics:  Device-1: AMD Renoir vendor: QUANTA driver: amdgpu v: kernel bus ID: 03:00.0 
           Display: x11 server: X.Org 1.20.8 driver: amdgpu,ati unloaded: fbdev,modesetting,vesa 
           resolution: 1920x1080~60Hz 
           OpenGL: renderer: AMD RENOIR (DRM 3.38.0 5.8.18-050818-generic LLVM 11.0.0) 
           v: 4.6 Mesa 21.0.0-devel (git-a22b85c 2020-12-23 focal-oibaf-ppa) direct render: Yes 
Audio:     Device-1: AMD vendor: QUANTA driver: snd_hda_intel v: kernel bus ID: 03:00.1 
           Device-2: AMD Raven/Raven2/FireFlight/Renoir Audio Processor vendor: QUANTA 
           driver: snd_rn_pci_acp3x v: kernel bus ID: 03:00.5 
           Device-3: AMD Family 17h HD Audio vendor: QUANTA driver: snd_hda_intel v: kernel 
           bus ID: 03:00.6 
           Sound Server: ALSA v: k5.8.18-050818-generic 
Network:   Device-1: Realtek RTL8822CE 802.11ac PCIe Wireless Network Adapter vendor: & Tele RSH 
           driver: rtw_8822ce v: N/A port: 2000 bus ID: 01:00.0 
           IF: wlp1s0 state: up mac: <filter> 
           IF-ID-1: br-2b39dfd66955 state: down mac: <filter> 
           IF-ID-2: br-6c49fb830b23 state: down mac: <filter> 
           IF-ID-3: docker0 state: down mac: <filter> 
Drives:    Local Storage: total: 476.94 GiB used: 151.32 GiB (31.7%) 
           ID-1: /dev/nvme0n1 vendor: Samsung model: MZVLB512HBJQ-00000 size: 476.94 GiB 
Partition: ID-1: / size: 467.96 GiB used: 151.32 GiB (32.3%) fs: ext4 dev: /dev/nvme0n1p2 
Sensors:   System Temperatures: cpu: 75.6 C mobo: N/A gpu: amdgpu temp: 56 C 
           Fan Speeds (RPM): N/A 
Info:      Processes: 278 Uptime: 13h 33m Memory: 15.08 GiB used: 6.99 GiB (46.3%) Init: systemd 
           runlevel: 5 Compilers: gcc: 9.3.0 Shell: zsh v: 5.8 inxi: 3.0.38 

After wake up from suspend, the network driver can't be loaded. the dmesg output is:
[  306.980490] rtw_8822ce 0000:01:00.0: Firmware version 7.3.0, H2C version 14
[  306.980500] rtw_8822ce 0000:01:00.0: can't change power state from D3hot to D0 (config space inaccessible)
[  306.980542] rtw_8822ce 0000:01:00.0: Firmware version 7.3.0, H2C version 14
[  306.982485] rtw_8822ce 0000:01:00.0: mac power on failed
[  306.982487] rtw_8822ce 0000:01:00.0: failed to power on mac
[  306.982488] rtw_8822ce 0000:01:00.0: failed to setup chip efuse info
[  306.982489] rtw_8822ce 0000:01:00.0: failed to setup chip information
[  306.983141] rtw_8822ce: probe of 0000:01:00.0 failed with error -114

lwfinger, 2020-12-29:
The critical line is 
[ 306.980500] rtw_8822ce 0000:01:00.0: can't change power state from D3hot to D0 (config space inaccessible)

What happens with the rtw88_* drivers built into kernel 5.8?

I have no idea why the config space is inaccessible. Do you double boot Windows? If so, turn off the Windows rapid boot option.

There are a couple of parameters for rtw_pci that might help. The first is disable_aspm. Create a file named /etc/modprobe.d/rtw_pci.conf containing a single line that says "options rtw_pci disable_aspm=1" (without the quotation marks). Then reboot and test suspension.

If that fails, edit the file above and replace "aspm" with "msi". Again reboot and test. If it still fails, change the contents of the file to be "options rtw_pci disable_aspm=1 disable_msi-1". Reboot and test. If it still fails, then I cannot help.

am813nt, 2020-12-30:
Neither "*disable_aspm=1*" nor "*disable_msi=1*" work. But "*disable_aspm=1 disable_msi=1*" works when *pci=noaer* is set in grub.

am813nt, 2020-12-30:
Neither of these work. It either doesn't work with pcie_port_pm=off and pcie_aspm=off in grub.
If I do after suspend
```
modprobe -r rtwpci && modprobe rtwpci
```
then there's message in the log:

```
rtw_8822ce 0000:01:00.0: Firmware version 7.3.0, H2C version 14
rtw_8822ce 0000:01:00.0: Refused to change power state, currently in D3
```
And the only way I know to make wifi work again is to reboot. 
The same behavior is with kernels 5.4 and 5.8 (vanilla Ubuntu 20.04 and Ubuntu 20.10).

lwfinger, 2020-12-30:
You cannot do it with those grub parameters. Why don't you try what I suggested?

am813nt, 2020-12-30:
I tried. Neither of them work.

lwfinger, 2020-12-30:
Did you try both together. One person has reported that worked. If you have, there is nothing I can do. You need to report it as stated in README,md.

am813nt, 2020-12-30:
Yes, I tried both parameters by one and together.
It looks like rtwpci module in Ubuntu doesn't support disable_aspm parameter:

```
$ modinfo rtwpci
filename:       /lib/modules/5.4.0-58-generic/kernel/drivers/net/wireless/realtek/rtw88/rtwpci.ko
license:        Dual BSD/GPL
description:    Realtek 802.11ac wireless PCI driver
author:         Realtek Corporation
srcversion:     139C107846E8BC1036278DE
alias:          pci:v000010ECd0000D723sv*sd*bc*sc*i*
alias:          pci:v000010ECd0000C82Fsv*sd*bc*sc*i*
alias:          pci:v000010ECd0000C822sv*sd*bc*sc*i*
alias:          pci:v000010ECd0000B822sv*sd*bc*sc*i*
depends:        mac80211,rtw88
retpoline:      Y
intree:         Y
name:           rtwpci
vermagic:       5.4.0-58-generic SMP mod_unload 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        4F:99:95:CB:A3:36:65:3E:1E:5A:30:73:58:12:C3:10:81:02:EA:FF
sig_hashalgo:   sha512
signature:      6A:89:2D:DF:D1:66:69:1A:60:1B:4D:4D:DE:02:50:C4:90:9C:33:95:
                B4:84:4B:95:57:C6:9F:3A:C0:2F:B5:40:99:DC:FA:FE:B9:D3:22:61:
                1F:F0:3D:8E:40:35:85:7A:A9:24:63:91:65:34:52:DE:08:C4:10:57:
                6C:9D:16:7E:68:54:3A:2B:48:C5:5D:51:56:F9:50:A2:0E:E9:78:98:
                87:F7:EA:50:EB:3C:39:AA:47:61:8B:62:3B:E8:AB:E0:9F:B8:FF:64:
                BB:37:37:C8:01:22:3B:35:A6:18:F0:DA:D1:96:42:7E:B1:25:E8:FB:
                BF:A4:88:4C:1E:B0:23:A1:9F:9A:0C:CB:AD:6D:B9:6A:AE:69:3D:3A:
                73:CB:16:2A:15:35:32:F2:99:16:88:88:B4:C8:92:C8:74:5C:BC:D7:
                E7:46:31:A3:A5:C5:D8:97:AA:8D:9D:3F:FA:61:27:11:5B:8F:13:DA:
                EB:6E:FD:4F:20:B4:FC:9C:B9:63:CD:B9:50:0B:41:5A:22:F0:57:3B:
                61:B2:7C:C2:CA:33:70:19:0D:E1:74:E4:BE:7E:8F:03:68:D3:14:4A:
                BD:50:81:C6:2F:E4:BB:C7:CC:E5:EC:D6:1D:B4:C3:A6:D5:C4:14:72:
                2C:DC:68:25:40:5B:DA:5C:40:D5:06:B2:D0:59:9E:3A:AB:59:EE:F5:
                4C:A2:43:67:E5:C6:74:A7:61:1F:39:5F:11:BD:11:77:AA:A5:FC:C7:
                EF:50:7E:AF:0D:B5:86:6A:9E:DF:72:48:7B:EA:EF:9F:48:D4:BD:BC:
                A3:D1:AF:31:3D:E5:ED:F5:FC:6A:63:54:A9:7F:65:A2:BE:0D:CB:D2:
                A4:2D:AE:5B:96:DE:89:17:0D:7D:F5:9D:A5:DD:E8:4A:BD:06:9B:5D:
                CF:79:F2:DB:26:F9:9B:77:0D:EB:FC:02:4B:3E:CE:68:49:BA:98:0F:
                4F:03:0E:99:50:BE:4B:90:DD:ED:E5:2A:90:FE:B5:DE:64:05:5A:7A:
                BC:63:03:91:01:84:F3:87:5B:0D:6E:73:43:CA:4F:EA:2F:9D:B2:AD:
                3C:A5:56:96:2A:A4:3C:33:17:37:2D:6D:B7:7E:F1:F5:80:DA:27:70:
                25:D3:D6:BE:0A:3E:68:63:47:FA:D4:ED:5F:BA:CF:68:9D:5D:C2:2A:
                F5:26:08:43:7B:72:BF:82:51:93:5F:DD:D1:DE:87:BA:E6:A5:3A:7A:
                68:80:78:B2:98:00:8A:46:30:F5:23:01:BD:69:88:DC:1F:27:5D:7C:
                B4:DE:8F:2A:29:3B:AF:91:ED:93:70:B3:F1:3E:28:9F:CC:44:8A:A0:
                0E:37:58:48:74:FA:18:A5:B8:CE:D8:B0
parm:           disable_msi:Set Y to disable MSI interrupt support (bool)
```
```
$ ls -1 /sys/module/rtwpci/parameters/
disable_msi
```

lwfinger, 2020-12-31:
If you are running rtwpci, then you are NOT using the routines from this repo. Everything here is named rtw_*.

closed 2020-12-31 by lwfinger

Eronex, 2021-09-03:
Hello!
I get same problem on Fedora 34: rtw_8822ce can't wake up from suspend.

First dmesg output is before adding options to driver:

```
[eronex@fedora ~]$ cat rtw_before_ASPMandMSIdisable.txt 
[    3.633700] rtw_8822ce 0000:01:00.0: enabling device (0000 -> 0003)
[    3.648920] rtw_8822ce 0000:01:00.0: Firmware version 9.9.4, H2C version 15
[    3.650260] rtw_8822ce 0000:01:00.0: Firmware version 9.9.10, H2C version 15
[    3.827926] rtw_8822ce 0000:01:00.0 wlp1s0: renamed from wlan0
[    5.900537] rtw_8822ce 0000:01:00.0: start vif 80:30:49:37:c6:ab on port 0
[    5.902134] rtw_8822ce 0000:01:00.0: stop vif 80:30:49:37:c6:ab on port 0
[    6.106584] rtw_8822ce 0000:01:00.0: start vif 12:07:66:5e:bb:20 on port 0
[    8.226816] rtw_8822ce 0000:01:00.0: stop vif 12:07:66:5e:bb:20 on port 0
[    8.434557] rtw_8822ce 0000:01:00.0: start vif 80:30:49:37:c6:ab on port 0
[   10.926056] rtw_8822ce 0000:01:00.0: sta 70:3a:51:10:ef:4b joined with macid 0
[   36.860358] rtw_8822ce 0000:01:00.0: sta 70:3a:51:10:ef:4b with macid 0 left
[   36.916389] rtw_8822ce 0000:01:00.0: stop vif 80:30:49:37:c6:ab on port 0
[   37.123532] rtw_8822ce 0000:01:00.0: start vif 7a:78:de:dc:d9:0b on port 0
[   37.334662] rtw_8822ce 0000:01:00.0: stop vif 7a:78:de:dc:d9:0b on port 0
[   37.542502] rtw_8822ce 0000:01:00.0: start vif 80:30:49:37:c6:ab on port 0
[   38.031783] rtw_8822ce 0000:01:00.0: stop vif 80:30:49:37:c6:ab on port 0
[   38.303071] rtw_8822ce 0000:01:00.0: can't change power state from D3hot to D0 (config space inaccessible)
[   40.372467] rtw_8822ce 0000:01:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[   40.372478] rtw_8822ce 0000:01:00.0: mac power on failed
[   40.372480] rtw_8822ce 0000:01:00.0: failed to power on mac
[   40.372601] Modules linked in: uinput rfcomm snd_seq_dummy snd_hrtimer xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_nat_tftp nf_conntrack_tftp bridge stp llc nft_objref nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat ip6table_nat ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_raw iptable_security ip_set nf_tables nfnetlink ip6table_filter ip6_tables iptable_filter bnep sunrpc vfat fat intel_rapl_msr intel_rapl_common edac_mce_amd kvm_amd kvm irqbypass rapl rtw88_8822ce rtw88_8822c snd_hda_codec_realtek pcspkr snd_hda_codec_generic huawei_wmi sparse_keymap wmi_bmof joydev ledtrig_audio snd_hda_codec_hdmi rtw88_pci rtw88_core snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec uvcvideo btusb mac80211 btrtl snd_hda_core videobuf2_vmalloc btbcm videobuf2_memops
[   40.373376] Modules linked in: uinput rfcomm snd_seq_dummy snd_hrtimer xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_nat_tftp nf_conntrack_tftp bridge stp llc nft_objref nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat ip6table_nat ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_raw iptable_security ip_set nf_tables nfnetlink ip6table_filter ip6_tables iptable_filter bnep sunrpc vfat fat intel_rapl_msr intel_rapl_common edac_mce_amd kvm_amd kvm irqbypass rapl rtw88_8822ce rtw88_8822c snd_hda_codec_realtek pcspkr snd_hda_codec_generic huawei_wmi sparse_keymap wmi_bmof joydev ledtrig_audio snd_hda_codec_hdmi rtw88_pci rtw88_core snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec uvcvideo btusb mac80211 btrtl snd_hda_core videobuf2_vmalloc btbcm videobuf2_memops
[   40.374102] Modules linked in: uinput rfcomm snd_seq_dummy snd_hrtimer xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_nat_tftp nf_conntrack_tftp bridge stp llc nft_objref nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat ip6table_nat ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_raw iptable_security ip_set nf_tables nfnetlink ip6table_filter ip6_tables iptable_filter bnep sunrpc vfat fat intel_rapl_msr intel_rapl_common edac_mce_amd kvm_amd kvm irqbypass rapl rtw88_8822ce rtw88_8822c snd_hda_codec_realtek pcspkr snd_hda_codec_generic huawei_wmi sparse_keymap wmi_bmof joydev ledtrig_audio snd_hda_codec_hdmi rtw88_pci rtw88_core snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec uvcvideo btusb mac80211 btrtl snd_hda_core videobuf2_vmalloc btbcm videobuf2_memops
[   42.403876] rtw_8822ce 0000:01:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[   42.403885] rtw_8822ce 0000:01:00.0: mac power on failed
[   42.403887] rtw_8822ce 0000:01:00.0: failed to power on mac
[   44.410579] rtw_8822ce 0000:01:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[   44.410587] rtw_8822ce 0000:01:00.0: mac power on failed
[   44.410589] rtw_8822ce 0000:01:00.0: failed to power on mac
[   46.411053] rtw_8822ce 0000:01:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[   46.411064] rtw_8822ce 0000:01:00.0: mac power on failed
[   46.411067] rtw_8822ce 0000:01:00.0: failed to power on mac
[   59.006475] rtw_8822ce 0000:01:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[... cut ...]
[  156.013686] rtw_8822ce 0000:01:00.0: mac power on failed
[  156.013697] rtw_8822ce 0000:01:00.0: failed to power on mac
[  168.012142] rtw_8822ce 0000:01:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[  168.012153] rtw_8822ce 0000:01:00.0: mac power on failed
[  168.012156] rtw_8822ce 0000:01:00.0: failed to power on mac
[  170.012662] rtw_8822ce 0000:01:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[  170.012673] rtw_8822ce 0000:01:00.0: mac power on failed
[  170.012675] rtw_8822ce 0000:01:00.0: failed to power on mac
```

lwfinger, 2021-09-03:
It appears that your only option is to set up a script to unload the driver before suspend or hibernate, and reload after resume.
As root create a file  /usr/lib/systemd/system-sleep/rtw88.sleep with the contents as follows:
#!/bin/sh
if [ "${1}" == "pre" ]; then
  modprobe -r rtw_8822ce
  echo "we are suspending at $(date)..." > /tmp/systemd_suspend_test
elif [ "${1}" == "post" ]; then
  modprobe rtw_8822ce
fi

Finally, make that file executable:
sudo chmod a+x /usr/lib/systemd/system-sleep/rtw88.sleep

Note that this problem likely arises because of faulty coding in your BIOS. My system works perfectly with hibernale of suspend with no problems.

Eronex, 2021-09-05:
> It appears that your only option is to set up a script to unload the driver before suspend or hibernate, and reload after resume.
> As root create a file /usr/lib/systemd/system-sleep/rtw88.sleep with [...]
Very big thankyou! It worked! :-) But in my case it is necessary "rtw_8822ce"  to replace with "rtw88_8822ce".
> Note that this problem likely arises because of faulty coding in your BIOS. My system works perfectly with hibernale of suspend with no problems.
This bag reported on Bugzilla from different laptops: [https://bugzilla.redhat.com/show_bug.cgi?id=1971223](url). It reference on code in `net/mac80211/util.c`. I believe in kernel developers, that they shall fixed this error :)

lwfinger, 2021-09-05:
Your logs do not show that WARNING at net/mac80211/util.c:2343, thus that bug does not match your issue.

Eronex, 2021-09-05:
Here yes, I was show dmesg in message. Later, after rebooting, the system itself prepared an bugreport that can be viewed by reference to Bugzilla. There is a mention, at what stage a failure occurred.

kavehtehrani, 2023-10-07:
> It appears that your only option is to set up a script to unload the driver before suspend or hibernate, and reload after resume. As root create a file /usr/lib/systemd/system-sleep/rtw88.sleep with [...]

Amazing! I was looking forever why this was happening. I have a Legion Pro 5 with Ubuntu 23.04 installed and this issue really floored me.