lwfinger/rtw88 #35: ntl8821ce driver did not work in debian10 testing

hadwinbook, 2021-01-08:
Hi
After install the rtw88 and reboot the system.I get the error.

Here is my system message:thanks for help
```bash
#cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux bullseye/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
```bash
#uname -a
Linux debian 5.9.0-5-amd64 #1 SMP Debian 5.9.15-1 (2020-12-17) x86_64 GNU/Linux
````
```bash
#lspci -v
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 08)
Subsystem: Lenovo Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers
Flags: bus master, fast devsel, latency 0, IOMMU group 0
Capabilities: <access denied>
Kernel driver in use: skl_uncore

00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07) (prog-if 00 [VGA controller])
Subsystem: Lenovo UHD Graphics 620
Flags: bus master, fast devsel, latency 0, IRQ 140, IOMMU group 1
Memory at f1000000 (64-bit, non-prefetchable) [size=16M]
Memory at d0000000 (64-bit, prefetchable) [size=256M]
I/O ports at e000 [size=64]
Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
Capabilities: <access denied>
Kernel driver in use: i915
Kernel modules: i915

00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
Subsystem: Lenovo Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
Flags: fast devsel, IRQ 255, IOMMU group 2
Memory at f252a000 (64-bit, non-prefetchable) [disabled] [size=4K]
Capabilities: <access denied>

00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21) (prog-if 30 [XHCI])
Subsystem: Lenovo Sunrise Point-LP USB 3.0 xHCI Controller
Flags: bus master, medium devsel, latency 0, IRQ 130, IOMMU group 3
Memory at f2500000 (64-bit, non-prefetchable) [size=64K]
Capabilities: <access denied>
Kernel driver in use: xhci_hcd
Kernel modules: xhci_pci

00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
Subsystem: Lenovo Sunrise Point-LP Thermal subsystem
Flags: fast devsel, IRQ 18, IOMMU group 3
Memory at f252b000 (64-bit, non-prefetchable) [size=4K]
Capabilities: <access denied>
Kernel driver in use: intel_pch_thermal
Kernel modules: intel_pch_thermal

00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
Subsystem: Lenovo Sunrise Point-LP CSME HECI
Flags: bus master, fast devsel, latency 0, IRQ 142, IOMMU group 4
Memory at f252c000 (64-bit, non-prefetchable) [size=4K]
Capabilities: <access denied>
Kernel driver in use: mei_me
Kernel modules: mei_me

00:17.0 SATA controller: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] (rev 21) (prog-if 01 [AHCI 1.0])
Subsystem: Lenovo Sunrise Point-LP SATA Controller [AHCI mode]
Flags: bus master, 66MHz, medium devsel, latency 0, IRQ 131, IOMMU group 5
Memory at f2528000 (32-bit, non-prefetchable) [size=8K]
[... 342 lines cut ...]
[   49.510927] pcieport 0000:00:1d.2: AER:   TLP Header: 00000000 00000000 00000000 00000000
[   49.510939] rtw_8821ce 0000:05:00.0: AER: can't recover (no error_detected callback)
[   49.510948] pcieport 0000:00:1d.2: AER: device recovery failed
[   50.549714] pcieport 0000:00:1d.2: AER: Multiple Uncorrected (Non-Fatal) error received: 0000:00:1d.2
[   50.549742] pcieport 0000:00:1d.2: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Completer ID)
[   50.549753] pcieport 0000:00:1d.2:   device [8086:9d1a] error status/mask=00008000/00010000
[   50.549760] pcieport 0000:00:1d.2:    [15] CmpltAbrt              (First)
[   50.549767] pcieport 0000:00:1d.2: AER:   TLP Header: 00000000 00000000 00000000 00000000
[   50.549780] rtw_8821ce 0000:05:00.0: AER: can't recover (no error_detected callback)
[   50.549881] pcieport 0000:00:1d.2: AER: device recovery failed
[   53.237763] pcieport 0000:00:1d.2: AER: Multiple Uncorrected (Non-Fatal) error received: 0000:00:1d.2
[   53.237815] pcieport 0000:00:1d.2: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Completer ID)
[   53.237826] pcieport 0000:00:1d.2:   device [8086:9d1a] error status/mask=00008000/00010000
[   53.237847] pcieport 0000:00:1d.2:    [15] CmpltAbrt              (First)
[   53.237854] pcieport 0000:00:1d.2: AER:   TLP Header: 00000000 00000000 00000000 00000000
[   53.237869] rtw_8821ce 0000:05:00.0: AER: can't recover (no error_detected callback)
[   53.237957] pcieport 0000:00:1d.2: AER: device recovery failed
[   53.237964] pcieport 0000:00:1d.2: AER: Multiple Uncorrected (Non-Fatal) error received: 0000:00:1d.2
[   53.238002] pcieport 0000:00:1d.2: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Completer ID)
[   53.238010] pcieport 0000:00:1d.2:   device [8086:9d1a] error status/mask=00008000/00010000
[   53.238016] pcieport 0000:00:1d.2:    [15] CmpltAbrt              (First)
[   53.238022] pcieport 0000:00:1d.2: AER:   TLP Header: 00000000 00000000 00000000 00000000
[   53.238035] rtw_8821ce 0000:05:00.0: AER: can't recover (no error_detected callback)
[   53.238062] pcieport 0000:00:1d.2: AER: device recovery failed
```

lwfinger, 2021-01-08:
Where does that driver 8821ce come from? It must be blacklisted.

hadwinbook, 2021-01-09:
> Where does that driver 8821ce come from? It must be blacklisted.

I get the 8821ce  driver  from tomaspinho/rtl8821ce.The 8821ce driver can work,but  unstable.
I tried to uninstall 8821ce driver and install rtw88 only,the error message was same as above.

sergiomb2, 2021-01-15:
I tested this code in October and my RTL8821CE also didn't worked also code of kernel 4.9 didn't worked ( https://bugzilla.kernel.org/show_bug.cgi?id=210779 ) 

@hadwinbook  please, can you send the pci ID of your RTL8821ce ?  `lspci -s 01:00.0 -nk`

Thank you

hadwinbook, 2021-01-16:
> I tested this code in October and my RTL8821CE also didn't worked also code of kernel 4.9 didn't worked ( https://bugzilla.kernel.org/show_bug.cgi?id=210779 )
> 
> @hadwinbook please, can you send the pci ID of your RTL8821ce ? `lspci -s 01:00.0 -nk`
> 
> Thank you

Here:
```bash
$ lspci -s 05:00.0 -nk
05:00.0 0280: 10ec:c821
	Subsystem: 17aa:c024
	Kernel driver in use: rtl8821ce
	Kernel modules: 8821ce
```

hadwinbook, 2021-02-05:
@sergiomb2
After install the latest version (Add firmware for rfe2 on 8821c). My wifi works fine. 
You can try it.

sergiomb2, 2021-02-05:
> @sergiomb2
> After install the latest version (Add firmware for rfe2 on 8821c). My wifi works fine.
> You can try it.

what is the date of the firmaware ? 
and about vanilla drive https://bugzilla.kernel.org/show_bug.cgi?id=210779 ?

lwfinger, 2021-02-05:
Rfe4 is still not supported. The firmware file is dated 02 Feb 2021, but you would have seen that if you had done a 'git pull'.

sergiomb2, 2021-02-05:
Thank you , I wonder why drive from tomaspinho/rtl8821ce works ... 

https://github.com/tomaspinho/rtl8821ce#wi-fi-not-working-for-kernel--59

hadwinbook, 2021-02-10:
Thank you for update the ref2 support.But It seems unstable,my system always crash. When the system crash,I can't do anything,only force shutdown the system.
After start the system.I got an error in messages log.
```
Feb 10 23:37:04 wow kernel: [    8.397536] rtw_8821ce 0000:05:00.0: start vif e2:36:ef:38:ac:9b on port 0
Feb 10 23:37:08 wow kernel: [   12.895358] rtw_8821ce 0000:05:00.0: stop vif e2:36:ef:38:ac:9b on port 0
Feb 10 23:37:09 wow kernel: [   13.417659] rtw_8821ce 0000:05:00.0: start vif e8:d0:fc:ea:b6:01 on port 0
Feb 10 23:37:13 wow kernel: [   18.112023] rtw_8821ce 0000:05:00.0: sta a0:9b:12:89:50:bf joined with macid 0
Feb 10 23:42:30 wow kernel: [  336.841313] rtw_8821ce 0000:05:00.0: failed to send h2c command
```
If you need more message,please tell me.

lwfinger, 2021-02-10:
That tells me nothing about the crash.

hadwinbook, 2021-02-11:
I config the rsyslog :*.*  /var/log/messages
The system crash,check the messages between crash time and the next boot time .There is no info about the crash.
The crash time around 17:29:07-17:31:00.
17:41:28 is the time  I boot system.   
```
Feb 11 17:28:27 wow rtkit-daemon[1175]: Supervising 8 threads of 4 processes of 1 users.
Feb 11 17:28:28 wow rtkit-daemon[1175]: Supervising 8 threads of 4 processes of 1 users.
Feb 11 17:28:28 wow rtkit-daemon[1175]: Supervising 8 threads of 4 processes of 1 users.
Feb 11 17:29:07 wow gnome-shell[1700]: Can't update stage views actor StWidget is on because it needs an allocation.
Feb 11 17:41:28 wow systemd[1]: Finished Monitoring of LVM2 mirrors, snapshots etc. using dmeventd or progress polling.
Feb 11 17:41:28 wow kernel: [    0.000000] Linux version 5.10.0-3-amd64 (debian-kernel@lists.debian.org) (gcc-10 (Debian 10.2.1-6) 10.2.
1 20210110, GNU ld (GNU Binutils for Debian) 2.35.1) #1 SMP Debian 5.10.13-1 (2021-02-06)
Feb 11 17:41:28 wow systemd[1]: Reached target Local File Systems (Pre).
Feb 11 17:41:28 wow kernel: [    0.000000] Command line: BOOT_IMAGE=/vmlinuz-5.10.0-3-amd64 root=UUID=70b93827-5812-45d7-8256-23a549b6d9
2f ro quiet
```
I uninstall the rtw88 driver,everything is fine,no more crash .How can I debug the info about the crash.

lwfinger, 2021-02-11:
First of all, with the driver from this repo, run the command 'lsmod | grep rtw' and post the answer.

To capture the crash, the pros use a network console, but that would be beyond your skills. The best we can hope is to open a terminal that occupies the fulls screen. The vertical part is most important. then run the command 'dmesg -wt'. It is important to get this command running before the crash and that this window be visible when it happens.

hadwinbook, 2021-02-12:
> First of all, with the driver from this repo, run the command 'lsmod | grep rtw' and post the answer.
>
> To capture the crash, the pros use a network console, but that would be beyond your skills. The [...]

Here is my try :
```
# lsmod | grep rtw
rtw_8821ce             16384  0
rtw_8821c              90112  1 rtw_8821ce
rtw_pci                28672  1 rtw_8821ce
rtw_core              212992  2 rtw_8821c,rtw_pci
mac80211              983040  2 rtw_core,rtw_pci
cfg80211              974848  2 rtw_core,mac80211

# lspci -s 05:00.0 -v
05:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821CE 802.11ac PCIe Wireless Network Adapter
        Subsystem: Lenovo RTL8821CE 802.11ac PCIe Wireless Network Adapter
        Flags: bus master, fast devsel, latency 0, IRQ 144, IOMMU group 15
        I/O ports at b000 [size=256]
        Memory at f2100000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: <access denied>
        Kernel driver in use: rtw_8821ce
        Kernel modules: rtw_8821ce
```

I use the `dmesg -wt | tee  -a dmesg.log` with the fullscreen,and wait the carsh happen.

[dmesg.log](https://github.com/lwfinger/rtw88/files/5970083/dmesg.log)

lwfinger, 2021-02-12:
You are running the WRONG firmware!!!! Mine says "rtw_8822ce 0000:02:00.0: Firmware version 9.9.5, H2C version 15" - yours does not match that. You need to do a 'git pull' and another 'sudo make install'.

hadwinbook, 2021-02-14:
> You are running the WRONG firmware!!!! Mine says "rtw_8822ce 0000:02:00.0: Firmware version 9.9.5, H2C version 15" - yours does not match that. You need to do a 'git pull' and another 'sudo make install'.

I used the latest repo include "rtw_8822ce 0000:02:00.0: Firmware version 9.9.5, H2C version 15" firmware.
Everything is same as before, rtw_8822ce firmware not load,system load the rtw8821c_fw.bin firmware.the crash continue.
I am a little curious why the rtl8821ce need to load rtw_8822ce firmware.

lwfinger, 2021-02-14:
I got confused. It should not.

This problem is one of incompatibility in your BIOS to handle this device and its coding. I thought I asked before if you had the latest BIOS installed, but I cannot find that section now.

You need to create an options file /etc/modprobe.d/50-rtw_pci.conf with the following line:
options rtw_pci disable_msi=y disable_aspm=y
Then reboot.

If you still have the problem after those options are enabled, then you will need to report this problem as README.md states. A couple of rules and suggestions: That mailing list will not accept html mail. Be sure you are sending plain text. In addition, control the size of any attachments. The mail will go to several hundred recipients. Better for you to isolate the critical parts of your logs and include them in-line.

sergiomb2, 2021-02-14:
I tried today , no luck 

uname -a 
Linux ideapad.local 5.10.15-100.fc32.x86_64 #1 SMP Wed Feb 10 17:52:05 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

```
cd ~/rtw88
git pull
make
sudo make install

shutdown -h now 

[Sun Feb 14 18:27:45 2021] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[Sun Feb 14 18:27:45 2021] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[Sun Feb 14 18:27:45 2021] rtw_8821ce 0000:07:00.0: Firmware version 24.8.0, H2C version 12
[Sun Feb 14 18:27:45 2021] rtw_8821ce 0000:07:00.0 wlp7s0: renamed from wlan0
[Sun Feb 14 18:27:46 2021] rtw_8821ce 0000:07:00.0: start vif 28:39:26:7b:1f:d5 on port 0
[Sun Feb 14 18:27:46 2021] rtw_8821ce 0000:07:00.0: stop vif 28:39:26:7b:1f:d5 on port 0
[Sun Feb 14 18:27:46 2021] rtw_8821ce 0000:07:00.0: start vif 22:69:a2:01:af:d1 on port 0
[Sun Feb 14 18:27:46 2021] rtw_8821ce 0000:07:00.0: stop vif 22:69:a2:01:af:d1 on port 0
[Sun Feb 14 18:27:47 2021] rtw_8821ce 0000:07:00.0: start vif 22:69:a2:01:af:d1 on port 0

wlp7s0: disconnected
        "Realtek RTL8821CE"
        wifi (rtw_8821ce), 22:69:A2:01:AF:D1, hw, mtu 1500
```

hadwinw, 2021-02-16:
After set the options as you said .System no longer crashes.Thank you.

closed 2021-02-16 by hadwinw

sergiomb2, 2021-04-23:
Since  kernel 5.11.12 (on fedora) more or less the wifi started to work out of the box .

https://bugzilla.kernel.org/show_bug.cgi?id=210779