lwfinger/rtw88 #37: modprobe: ERROR: could not insert 'rtw_8822be': Exec format error

KewinOgink, 2021-01-14:
Hi here I am again.

I still have a Lenovo-IdeaPad-C340-14IWL, with a dual boot windows 10 and ubuntu.
I had ubuntu 20.10, but couldn't get the wifi working.
I reinstalled 20.10 clean and the same problem occurs.
I installed 18.04 and the same problem is there.

Sometimes wifi works and sometimes it does not. No wifi networks found / failed to activate wireless connection etc.

I notice that sometimes my laptop freezes every 10 seconds or so for a second, not just the mouse but everything. It might be related but could not find what to do about it.

Either way, I have spend hours and hours on this problem and it is very frustrating. Here is where I am at now. I don't know a lot about ubuntu, so please any help is really welcome.

`lspci` ouput
`02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter (rev ff)
`

If understand [this](https://www.maketecheasier.com/fix-wi-fi-not-working-ubuntu/) correctly means that I need a proper rtl_8822be driver. That is what I download here in this rtw88 repo.

`lsmod | grep rtw` output:
`rtwpci                 24576  0
rtw88                 610304  1 rtwpci
mac80211              851968  2 rtwpci,rtw88
cfg80211              712704  2 mac80211,rtw88
`

I did not blacklist anything, because I don't see conflicting drivers.

I sudo do 
`sudo modprobe rtw_8822be` and it says 
`modprobe: ERROR: could not insert 'rtw_8822be': Exec format error`

please, how can I simply have wifi working reliably? That is one of the essential things of having a laptop

lwfinger, 2021-01-14:
In the output of the dmesg command, what is the reason for the modprobe error?

Unfortunately, Lenovo has a habit of releasing laptops with new Realtek chips that have no Linux drivers. That is the reason this repo is provided. Note that whining about your problem does NOT make me want to help you.

KewinOgink, 2021-01-14:
My apologies, I appreciate your help.

[dmesg.txt](https://github.com/lwfinger/rtw88/files/5816605/dmesg.txt)

lwfinger, 2021-01-14:
Your driver fails at
------------[ cut here ]------------
failed to read DBI register, addr=0x0719
WARNING: CPU: 2 PID: 768 at /home/kewin/rtw88/pci.c:1173 rtw_dbi_read8.constprop.29+0xaa/0xc0 [rtw_pci]

I would not have expected that failure to have generated the modprobe message.

I have updated to source to match the current kernel version. Do a 'git pull' and try again.

KewinOgink, 2021-01-14:
```
git pull
make
sudo make install
sudo modprobe -r rtw_8822be
sudo modprobe rtw_8822be
```
restarted
same problem remains.
 `lsmod | grep rtw`:
```
rtw_8822be             16384  0
rtw_8822b             225280  1 rtw_8822be
rtw_pci                24576  1 rtw_8822be
rtw_core              180224  2 rtw_8822b,rtw_pci
mac80211              851968  2 rtw_core,rtw_pci
cfg80211              712704  2 rtw_core,mac80211
```
`dmesg -t`:
[dmesg.txt](https://github.com/lwfinger/rtw88/files/5817262/dmesg.txt)

lwfinger, 2021-01-14:
I just sent the warning part of yout log to my friend at Realtek. We will see if he can help.

lwfinger, 2021-01-15:
He suggests that you try the module parameter disable_aspm=1.

KewinOgink, 2021-01-16:
I'm not sure if I did it correctly. I followed this to do
```sudo nano ../../etc/modprobe.d/fix_wifi.conf```
and in that file I put `disable_aspm=1`.

However now at the settings I cannot even turn on wifi, as it says no wifi adapter is found

lwfinger, 2021-01-16:
No, that is not correct. You need to delete that file. Create file /etc/modprobe.d/50-rtw_pci.conf with the following line
`options rtw_pci disable_aspm=1`

KewinOgink, 2021-01-17:
I did a 
```
cd ~/rtw88
git pull
make
sudo make install
```

and then did as you said and rebooted, but still the same problem.
In the first minute or so it shows the networks and connects to wifi, but fails afterwards
`dmesg -t > dmesg.txt`
[dmesg.txt](https://github.com/lwfinger/rtw88/files/5825799/dmesg.txt)

lwfinger, 2021-01-17:
That improved things. The error now is

rtw_8822be 0000:02:00.0: Refused to change power state, currently in D3
rtw_8822be 0000:02:00.0: mac power on failed
rtw_8822be 0000:02:00.0: failed to power on mac
rtw_8822be 0000:02:00.0: failed to setup chip efuse info
rtw_8822be 0000:02:00.0: failed to setup chip information
rtw_8822be: probe of 0000:02:00.0 failed with error -114

Try adding disable_aspm=y to the options file.

KewinOgink, 2021-01-17:
I can barely believe it, but it seems to work! 
I have restarted multiple times and also switched windows-ubuntu, and switched off and on wifi and I keep having wifi!

I don't know who you are or why you helped me, but thank you very much!

lwfinger, 2021-01-18:
I have been involved with Realtek drivers since 2008. I have no detailed information about the wireless hardware, but I have a good understanding of the interaction with Linux.

Your problem was a bad interaction between the PCI driver and your hardware.

KewinOgink, 2021-01-19:
Seems I was a little too soon. It worked for about a day, but the problem returned. I have seen in dmesg -t the following output. I am sure this is just another option to set, just don't know which one.
``` 
---[ end trace 67a9d738e0680461 ]---
rtw_pci 0000:02:00.0: failed to read ASPM, ret=-5
rtw_pci 0000:02:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
rtw_pci 0000:02:00.0: mac power on failed
rtw_pci 0000:02:00.0: failed to power on mac
rtw_pci 0000:02:00.0: leave idle state failed
rtw_pci 0000:02:00.0: failed to leave ips state
rtw_pci 0000:02:00.0: failed to leave idle state
wlp2s0: send auth to 00:0f:94:46:55:e8 (try 1/3)
wlp2s0: send auth to 00:0f:94:46:55:e8 (try 2/3)
wlp2s0: send auth to 00:0f:94:46:55:e8 (try 3/3)
wlp2s0: authentication with 00:0f:94:46:55:e8 timed out
rtw_pci 0000:02:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
------------[ cut here ]------------
```

[dmesg.txt](https://github.com/lwfinger/rtw88/files/5834853/dmesg.txt)

KewinOgink, 2021-01-25:
It stopped working again after a couple of complete restarts. Not sure why, with the next restart it might work again. 
When this happens I noticed that my entire laptop (or at least sound and mouse) freezes for about 2 seconds every 30 seconds or so. 
`lspci -v`:
```
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter (rev ff) (prog-if ff)
	!!! Unknown header type 7f
	Kernel driver in use: rtw_pci
	Kernel modules: rtwpci

```

```
lsmod | grep rtw
rtwpci                 24576  0
rtw88                 610304  1 rtwpci
mac80211              851968  2 rtwpci,rtw88
cfg80211              712704  2 mac80211,rtw88
```

[dmesg.txt](https://github.com/lwfinger/rtw88/files/5865254/dmesg.txt)

lwfinger, 2021-01-25:
The only other module parameter available is disable_msi. Put "disable_msi=y' in your options file.

The last time I bought a laptop, the finalists were Toshiba and Lenovo. I'm glad I picked the Toshiba.

KewinOgink, 2021-01-30:
Hi, I have tried this option too and my config file looks like this now:
```
options rtw_pci disable_aspm=1 
options rtw_pci disable_aspm=y
options rtw_pci disable_msi=y
```
Sometimes on a reboot it works, and sometimes it doesn't, just like with only having the previous options. 

[dmesg.txt](https://github.com/lwfinger/rtw88/files/5897617/dmesg.txt)

Do you think that if I switch to another linux version, it might work? maybe Ubuntu 20.04, or maybe mint?

lwfinger, 2021-01-30:
Those options should be on a single line: "options rtw_pci disable_aspm=y disable_msi=y"

You have a mixed-up setup. The driver should be rtw_pci, but the error is coming from rtwpci! Where does that one come from? Please run the command "lsmod | grep rtw" and post the result.

KewinOgink, 2021-01-31:
`lsmod | grep rtw` shows rtwpci and not rtw_pci. I am not sure where it comes from, a `history | grep rtwpci`didn't give any results

```
$ lsmod | grep rtw
rtwpci                 24576  0
rtw88                 610304  1 rtwpci
mac80211              851968  2 rtwpci,rtw88
cfg80211              712704  2 mac80211,rtw88
```

lwfinger, 2021-01-31:
These do NOT come from this repo and they are likely interfering. I think they come from Ubuntu. Please blacklist rtwpci and rtw88.

KewinOgink, 2021-02-04:
I blacklisted the drivers as you said. I did it on 2 separate lines:
`blacklist rtwpci`
`blacklist rtw88`

Now `lsmod | grep rtw` does not give any output anymore. 

[dmesg.txt](https://github.com/lwfinger/rtw88/files/5924396/dmesg.txt)
[lspci.txt](https://github.com/lwfinger/rtw88/files/5924397/lspci.txt)

lwfinger, 2021-02-04:
Now install the driver from this repo!

KewinOgink, 2021-02-05:
I didn't know which one to install, so I just tried rtw_8822b and it seems to have worked! I will monitor whether it is stable across multiple restarts and switches from and to windows boot. Promising result!

lwfinger, 2021-02-05:
My repo matches the latest in the kernel with pending patches. I would it to be better than any other driver.

KewinOgink, 2021-02-15:
I don't want to update it to ubuntu 20.04 just in case, but still working, so case closed!

Many thanks!

closed 2021-02-15 by KewinOgink

manishpatel005, 2021-08-13:
I have rtl8822be driver and had installed Ubuntu 18.04 with 5.4.0-42-generic kernel and was facing the same problem. Following all the steps, my wifi adapter got removed. I had to use `echo "options r8822be aspm=0"  | sudo tee /etc/modprobe.d/r8822be.conf` and reboot. 
This thread is by far the most trusted source- Thanks lwfinger and KewinOgink

sachinkumarsingh092, 2023-09-06:
`echo "options r8822be aspm=0"  | sudo tee /etc/modprobe.d/r8822be.conf` also worked for me.

lwfinger, 2023-09-06:
Interesting. There is no driver r8822be, thus this options file does nothing. The drivers that use this module parameter are rtw88_pci for the in-kernel driver, and rtw_pci for the one from this repo. It is working without any options.