lwfinger/rtw88 #39: Will Wifi-Direct(P2P mode) be supported?

hongjia, 2021-01-26:
I am using the rtw88 modules from kernel 5.10:

```
 $ lspci

...
04:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter
...

$ lsmod|grep rtw

rtw88_8822be           16384  0
rtw88_8822b           229376  1 rtw88_8822be
rtw88_pci              28672  1 rtw88_8822be
rtw88_core            159744  2 rtw88_pci,rtw88_8822b
mac80211              983040  2 rtw88_pci,rtw88_core
cfg80211              974848  3 r8712u,rtw88_core,mac80211

```

But P2P mode is not supported:

```
# iw list
Wiphy phy0
...
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
	Band 1:
...
```

Is there any plan to add P2P mode support in the drivers?

I see the closed issuse #38 is related to this. I am also trying to enable miracast (gnome-wireless-displays) on my laptop but it fails to work due to lacking support.

lwfinger, 2021-01-26:
This approaches a "Don't ask me" kind of question. The linux-wireless mailing list mentioned in READ.md will get to the driver's authors. I DID NOT WRITE THIS DRIVER!

Have you tried P2P using NetworkManager. That should work for any mac80211-based wireless driver.

closed 2021-01-26 by lwfinger

isasmendiagus, 2022-08-17:
@hongjia Hi dear Hong,
I just sent an email to the linux-wireless mailing list asking if there are any intentions to add P2P support to this driver.
Hope we will have some response soon, I'll let you know any news

Thanks

zachanast18, 2022-11-30:
Do we have any update on this?

hongjia, 2022-12-27:
> @hongjia Hi dear Hong, I just sent an email to the linux-wireless mailing list asking if there are any intentions to add P2P support to this driver. Hope we will have some response soon, I'll let you know any news
> 
> Thanks

Thank you very much.