lwfinger/rtw88 #40: Kernel version is 4.11 which is below 4.19

lpbittoo, 2021-02-15:
[root@localhost Downloads]# cd rtw88-master/
[root@localhost rtw88-master]# make
make -C /lib/modules/4.11.12-100.fc24.x86_64/build M=/home/jaiprakash/Downloads/rtw88-master modules
make[1]: Entering directory '/usr/src/kernels/4.11.12-100.fc24.x86_64'
Makefile:923: "Cannot use CONFIG_STACK_VALIDATION, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  CC [M]  /home/jaiprakash/Downloads/rtw88-master/main.o
/home/jaiprakash/Downloads/rtw88-master/main.c:25:65: error: expected ‘)’ before ‘bool’
 module_param_named(disable_lps_deep, rtw_disable_lps_deep_mode, bool, 0644);
                                                                 ^~~~
/home/jaiprakash/Downloads/rtw88-master/main.c:26:48: error: expected ‘)’ before ‘bool’
 module_param_named(support_bf, rtw_bf_support, bool, 0644);
                                                ^~~~
/home/jaiprakash/Downloads/rtw88-master/main.c:27:48: error: expected ‘)’ before ‘uint’
 module_param_named(debug_mask, rtw_debug_mask, uint, 0644);
                                                ^~~~
/home/jaiprakash/Downloads/rtw88-master/main.c:29:36: error: expected ‘)’ before string constant
 MODULE_PARM_DESC(disable_lps_deep, "Set Y to disable Deep PS");
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jaiprakash/Downloads/rtw88-master/main.c:30:30: error: expected ‘)’ before string constant
 MODULE_PARM_DESC(support_bf, "Set Y to enable beamformee support");
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jaiprakash/Downloads/rtw88-master/main.c:31:30: error: expected ‘)’ before string constant
 MODULE_PARM_DESC(debug_mask, "Debugging mask");
                              ^~~~~~~~~~~~~~~~
/home/jaiprakash/Downloads/rtw88-master/main.c:1842:15: error: expected declaration specifiers or ‘...’ before string constant
 MODULE_AUTHOR("Realtek Corporation");
               ^~~~~~~~~~~~~~~~~~~~~
/home/jaiprakash/Downloads/rtw88-master/main.c:1843:20: error: expected declaration specifiers or ‘...’ before string constant
 MODULE_DESCRIPTION("Realtek 802.11ac wireless core module");
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jaiprakash/Downloads/rtw88-master/main.c:1844:16: error: expected declaration specifiers or ‘...’ before string constant
 MODULE_LICENSE("Dual BSD/GPL");
                ^~~~~~~~~~~~~~
scripts/Makefile.build:294: recipe for target '/home/jaiprakash/Downloads/rtw88-master/main.o' failed
make[2]: *** [/home/jaiprakash/Downloads/rtw88-master/main.o] Error 1
Makefile:1496: recipe for target '_module_/home/jaiprakash/Downloads/rtw88-master' failed
make[1]: *** [_module_/home/jaiprakash/Downloads/rtw88-master] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.11.12-100.fc24.x86_64'
Makefile:79: recipe for target 'all' failed
make: *** [all] Error 2

lpbittoo, 2021-02-15:
Now i am getting this....................................
tw_tx_write_data_rsvd_page_get’:
/home/jaiprakash/Downloads/rtw88-master/tx.c:457:2: error: implicit declaration of function ‘skb_put_data’ [-Werror=implicit-function-declaration]
  skb_put_data(skb, buf, size);
  ^~~~~~~~~~~~
cc1: some warnings being treated as errors

lwfinger, 2021-04-11:
I will not be able to find all the API changes as far back as 4.11. Is it possible for you to use a newer kernel?

closed 2021-04-11 by lwfinger