lwfinger/rtw88 #42: Wifi connection dissapear after few minutes of working

CristianSw, 2021-03-02:
kernel 5.11.2-arch1-1
Network controller: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter (rev ff)

**Steps to reproduce:**
1. Boot into system
2. Connect to wifi
3. Wait few minutes 

**Expected result:**
Wifi connection is stable.
**Actual result:**
Wifi connection dissapear after few minutes, restarting systtem wifi option dissapear from menu or make system able to connect but issue is reproducible.
**dmesg:** https://pastebin.com/BDdUC79j

rbrick, 2021-03-20:
I’m having similar issues with the same adapter. I connect to my 5 GHz network, and it constantly disconnects & re-prompts for network password. 2.4 GHz is perfectly stable for me. 

I apologize, I don’t have any logs, was testing in a live environment, wanting to move away from Windows but found I had issues with WiFi drivers.

lwfinger, 2021-03-21:
Stability issues are beyond me. You need to follow the instructions in README.md to cummunicate with the driver authors.

closed 2021-03-21 by lwfinger

jonatanaxe, 2021-04-28:
Try to disable the wifi-power-on and then turn the notebook off and on

`sudo sed -i 's/3/2/' /etc/NetworkManager/conf.d/default-wifi-powersave-on.conf 
`

https://gist.github.com/jonatanaxe/a59b01bfe5b2d00fa267cacaf962dbe9