lwfinger/rtw88 #45: Bluetooth devices not being loaded

marioscience, 2021-03-07:
Hello, 

Bluetooth on my debian machine stopped working a couple of days ago. I have been trying to find the root of the problem but haven't been able to. So now I resort to ask for help.

My motherboard is

Asus ROG Strix Z370-I Gaming Mini ITX LGA1151 Motherboard
which has the  Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter. This is the device that provides bluetooth connection as well. 

Error log: 
- bluetoothctl: No default controller available
- hcitool dev: Devices: <empty>. hcitool device is empty.
- systemctl start hciuart: Failed to start hciuart.service: Unit hciuart.service not found.
- ls -Al /dev/ttyAMA0: No such file or directory
- sudo rfkill list all: Only 'Wireless LAN' shows up. No bluetooth.
- hciconfig scan: Can't get device info: No such device
- sudo bluetoothd -n -d: bunch of 'Key file does not have key' errors. Then 'Name already in use' and 'Unable to get on D-Bus' 
- bt-adapter -l: No adapters found
- sudo cat /var/log/syslog: [haven't really examined closely]

Other notes:
- systemctl status bluetooth: active(running) [Looks OK. NO ERROR]
- dmesg | grep -i blue: No apparent errors
- lspci: wifi card rtl8822be is in use
- lsmod: modules for rtw_8822be and r8822be are loaded
- systemctl | grep -i blue: loaded active   running Bluetooth service

I will include all the output from running the commands above:
`$bluetoothctl           
Agent registered
[bluetooth]# power on
No default controller available
[bluetooth]# 
`
`$hcitool dev 
Devices:
`
`$systemctl start hciuart 
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to start 'hciuart.service'.
Authenticating as: <username>
Password: 
==== AUTHENTICATION COMPLETE ===
Failed to start hciuart.service: Unit hciuart.service not found.
`

`$ls -Al /dev/ttyAMA0
ls: cannot access '/dev/ttyAMA0': No such file or directory
`

`$sudo rfkill list all 
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
`

`$hciconfig scan  
Can't get device info: No such device
`

`$ sudo bluetoothd -n -d
bluetoothd[13175]: Bluetooth daemon 5.50
bluetoothd[13175]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
bluetoothd[13175]: src/main.c:parse_config() Key file does not have key “DiscoverableTimeout” in group “General”
bluetoothd[13175]: src/main.c:parse_config() Key file does not have key “PairableTimeout” in group “General”
bluetoothd[13175]: src/main.c:parse_config() Key file does not have key “Privacy” in group “General”
bluetoothd[13175]: src/main.c:parse_config() Key file does not have key “Name” in group “General”
bluetoothd[13175]: src/main.c:parse_config() Key file does not have key “Class” in group “General”
bluetoothd[13175]: src/main.c:parse_config() Key file does not have key “DeviceID” in group “General”
bluetoothd[13175]: src/main.c:parse_config() Key file does not have key “ReverseServiceDiscovery” in group “General”
bluetoothd[13175]: src/main.c:parse_config() Key file does not have key “MinEncKeySize” in group “GATT”
D-Bus setup failed: Name already in use
bluetoothd[13175]: Unable to get on D-Bus
`

`$bt-adapter -l 
No adapters found
`

`$lspci   
00:00.0 Host bridge: Intel Corporation 8th Gen Core Processor Host Bridge/DRAM Registers (rev 07)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 07)
00:14.0 USB controller: Intel Corporation 200 Series/Z370 Chipset Family USB 3.0 xHCI Controller
00:16.0 Communication controller: Intel Corporation 200 Series PCH CSME HECI #1
00:17.0 SATA controller: Intel Corporation 200 Series PCH SATA controller [AHCI mode]
00:1b.0 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #17 (rev f0)
00:1b.3 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #20 (rev f0)
00:1c.0 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #1 (rev f0)
00:1c.4 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #5 (rev f0)
00:1d.0 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #9 (rev f0)
00:1f.0 ISA bridge: Intel Corporation Z370 Chipset LPC/eSPI Controller
00:1f.2 Memory controller: Intel Corporation 200 Series/Z370 Chipset Family Power Management Controller
00:1f.3 Audio device: Intel Corporation 200 Series PCH HD Audio
00:1f.4 SMBus: Intel Corporation 200 Series/Z370 Chipset Family SMBus Controller
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (2) I219-V
01:00.0 VGA compatible controller: NVIDIA Corporation GP104 [GeForce GTX 1070] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GP104 High Definition Audio Controller (rev a1)
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter
05:00.0 USB controller: ASMedia Technology Inc. ASM2142 USB 3.1 Host Controller
`

`$lsmod | grep -i rtw   
rtw_8822be             16384  0
rtw_8822b             229376  1 rtw_8822be
rtw_pci                28672  1 rtw_8822be
rtw_core              200704  2 rtw_8822b,rtw_pci
mac80211              835584  5 rtl_pci,rtw_core,rtlwifi,rtw_pci,r8822be
cfg80211              774144  4 rtw_core,rtlwifi,mac80211,r8822be
`

I might have to black list some of those above, but they look the same as someone who had their bluetooth working properly, so I don't know if the issue is here. 

I've been hit with issues that I have solved on my own before with my general knowledge of how this driver works, but this issue in particular got me fearing two things:
     - Either there is much more I still don't understand about the inner workings of this driver.
     - My physical bluetooth card is damaged and this is a hardware issue. (I prefer to be an idiot that doesn't know how this system works rather than my motherboard having a component fried)

Anyway, please help shed some light into what might be causing this issue. I would really appreciate it.

dubhater, 2021-03-18:
Does lsusb list your bluetooth device? And nevermind grep, read all of dmesg. Maybe put it on pastebin.

marioscience, 2021-03-19:
$lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:081b [WEBCAM]
Bus 001 Device 002: ID 1ea7:0066 SHARKOON Technologies GmbH [Mediatrack Edge Mini Keyboard]
Bus 001 Device 004: ID 0b05:1872 ASUSTek Computer, Inc. <- this is the brand of the motherboard, could be the bluetooth device?
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

This is the dmesg output: https://pastebin.com/uRPGdMwc

Thanks for taking a look.

marioscience, 2021-03-19:
This is why I am fearing the worst. Everything looks fine in my eyes. As if it should work, but it doesn't so it might be a hardware issue. I am going to get windows running in a virtual machine to see if it will get the the bluetooth going that way.

dubhater, 2021-03-19:
" 0b05:1872 ASUSTek Computer, Inc. " is some LED controller.

It *looks* like you don't have a Bluetooth adapter at all. Is it disabled in the BIOS? Also, I see you're not using the rtw88 driver. I'm not sure why you're posting here?

dubhater, 2021-03-19:
Or maybe you were using rtw88 but then you updated the kernel or something and forgot to recompile?

marioscience, 2021-03-19:
I did update my kernel. Then went to recompile and didn't find the original git repo that I used to install the drivers. I found this one which includes my device so I decided to clone this repo, build and install. I did however try to reinstall the repo version I originally used but no luck. Can't remember the name of the repo but the link I had for it gives me a 404 in github. 

That makes sense, funny that the RGB lights have their own drivers and are working but bluetooth/wifi is this whole ordeal. I am really thankful that lwfinger put together these driver repos. 

So, how would I go about getting a clean slate to clone/build/install this driver? I tried sudo rmod rtl8822be and in the rtw88 repo make/install but maybe that is not enough?

lwfinger, 2021-03-19:
Why do you feel you need a clean slate? Doing "git pull; make; sudo make install" while in the download directory (rtw88>) will reinstall this driver.

Without a BT interface showing in the lsusb listing, you will NEVER get BlueTooth.

This rtw88 repo is ONLY for wifi. The kernel drivers btusb and btrtl in directory drivers/bluetooth are used to drive the Realtek BT devices.

marioscience, 2021-03-19:
I felt the need for that since I thought no device visible = driver not loaded properly. Does lsusb just checks the hardware component directly? 

Ah, I was going to look for the name of the older driver I had but I deleted it because I thought this one had worked (since I had wifi). 

So any ideas or recommendations? Am I condemned to using windows until the end of days or do I have to buy a new computer to keep using Linux? I seriously don't know what to do, I haven't had time to run the VM Windows experiment yet but will be doing that tomorrow. Will actually install in a separate partition since that way windows can access hardware directly. 

THANK YOU PROPRIETARY DRIVER VENDORS. THIS IS TOTALLY GREAT. YOU ARE ESSENTIALLY KEEPING OWNERSHIP OF THE HARDWARE I PAID FOR. WELL PLAYED.

marioscience, 2021-03-19:
The image of Linus giving them the mighty finger comes to mind, they have it well deserved. *sigh*

dubhater, 2021-03-20:
Did you check the BIOS? Some laptops let you disable the Bluetooth there and then the operating system shouldn't see it at all.

To easily check if it's a distro problem somehow (too old kernel? I don't know) you can put the live Opensuse Tumbleweed image on a USB stick and boot that. It has a recent kernel with rtw88 included.

marioscience, 2021-03-20:
Yeah, the bluetooth controller is enabled in the BIOS settings. I will try the Opensuse Tumbleweed image. Hopefully my motherboard is not broken. I can live without bluetooth I think, but it would still be a hard blow.

marioscience, 2021-03-20:
I was seeing a comment about how it is unlikely the bluetooth module is broken since the wifi is in the same card as the bluetooth. Since wifi is working the card is probably good. I don't know enough about how the actual card hardware is constructed but I hope they are correct. We'll see.

lwfinger, 2021-03-20:
Yes, lsusb interrogates the hardware without use of any driver. Its scan method is exactly the one used by the kernel to detect if a particular device is available. To reiterate, if no BT device in the lsusb scan, no BT driver will be loaded.

marioscience, 2021-03-20:
I booted into an openSUSE live usb. The bluetooth settings said 'no bluetooth device detected', lsusb gave the same output as above with no bluetooth in sight. I gotta just accept reality for what it is and declare this bluetooth component dead. 

Closing this issue, thank you guys for taking a look and helping out.

closed 2021-03-20 by marioscience

marioscience, 2021-03-21:
Aha! So, took apart my computer took out the battery in the motherboard and reset the CMOS. Bluetooth is back up. 

For anyone facing this issue this solved it for me: https://rog.asus.com/forum/showthread.php?108642-Bluetooth-suddenly-stopped-working

Again, I really appreciate the help and the awesome driver to support Linux computing. You guys are awesome.