lwfinger/rtw88 #47: How to remove

seamarhay, 2021-03-11:
Ran the make command for this and it worked like a charm. However, I then learned that I do not have this device. How do I remove the files once installed / made (make)?

lwfinger, 2021-03-11:
Are you really that worried about a few MB of disk space? If you do not have the device, then those files are just like ALL the drivers for other devices you do not have. They just sit there on the disk. When you update your kernel and the old one is deleted, they will be gone.

Did you consider 'sudo make uninstall'?

closed 2021-03-11 by lwfinger

seamarhay, 2021-03-11:
Thanks for your help. I'm not worried about the MBs, soI'll just allow the
next kernel update delete the unused files. Just trying to keep the machine
clean. I did try the sudo make uninstall. Again thank you!

castillofrancodamian, 2024-05-30:
> Are you really that worried about a few MB of disk space? If you do not have the device, then those files are just like ALL the drivers for other devices you do not have. They just sit there on the [...]

Using this command, I restarted the PC and I still see networks in network manager.