lwfinger/rtw88 #49: Compilation fails on kernels with fsleep

David96, 2021-03-21:
The check for fsleep in commit accc527 doesn't work - on newer kernels the build fails because fsleep gets redefined. I tried finding out more about the way you checked for fsleep being defined but didn't find any information about whether this is supposed to work. An alternative way would apparently be using the __weak__ function attribute, but I couldn't get that to work either. For now I just removed the definition in main.h as a workaround for me.

David96, 2021-03-21:
Did some more digging, the weak function attribute doesn't work because fsleep is defined in a header file and therefore ends up in the same .c file as our redefinition. The weak attribute works at linker level and therefore has no effect if both definitions are in the same file. Another way of checking for the fsleep function would be with some Makefile magic, something like this:
```
NOFSLEEP:= $(shell echo "#include <delay.h>\nint main() { fsleep(0);}" | $(CC) $(CFLAGS) -o /dev/null -Werror -xc - >/dev/null 2>/dev/null && echo 0 || echo 1)
ifeq "$(NOFSLEEP)" "1"
        EXTRA_CFLAGS += -DNO_FSLEEP
endif
```
Now this doesn't work as is because I couldn't figure out in which variable the kernel build system keeps the include directories… CFLAGS is empty so the compilation always fails for some missing header files.

lwfinger, 2021-03-21:
The fix is to test the kernel version. I tried the quick and dirty way and broke it yesterday.

closed 2021-03-21 by lwfinger