lwfinger/rtw88 #51: Not working on rtl8822be

CristianSw, 2021-04-01:
kernel 5.11.11-arch1-1
Network controller: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter (rev ff)

Steps to reproduce: 
1. Build current version 
2. Restart computer
3. Wait to connect to wifi
Expected result :
Wifi work fine, find and connet to wifi networks, working stable.

Actual result:
Wifi try to connect to wifi network and crash, system dont see wifi adapter.

Dmesg for debugging
[dmesg.txt](https://github.com/lwfinger/rtw88/files/6244627/dmesg.txt)

lwfinger, 2021-04-01:
Your dmesg.txt starts after a critical piece of information. Please post the entire output of dmesg.

CristianSw, 2021-04-02:
[dmesg.txt](https://github.com/lwfinger/rtw88/files/6249615/dmesg.txt)

lwfinger, 2021-04-02:
You are still cutting off the part I need!! If you want me to help, I need to know what condition the kernel found.

CristianSw, 2021-04-03:
Copied from /var/log/dmesg
[dmesg_full.txt](https://github.com/lwfinger/rtw88/files/6252347/dmesg_full.txt)

CristianSw, 2021-04-03:
Copied from /var/log/dmesg
[dmesg_full.txt](https://github.com/lwfinger/rtw88/files/6252374/dmesg_full.txt)

output from dmesg | more command
[dmesg_more.txt](https://github.com/lwfinger/rtw88/files/6252376/dmesg_more.txt)

lwfinger, 2021-04-03:
In this case, you did not have the error. I need a full dmesg WITH the error!

The one thing I noticed is that your BIOS is severely broken as follows:

[    0.217934] ACPI BIOS Error (bug): Failure creating named object [\_SB.PCI0.XHC.RHUB.TPLD], AE_ALREADY_EXISTS (20201113/dswload2-326)
[    0.217944] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20201113/psobject-220)
[    0.217947] ACPI: Skipping parse of AML opcode: Method (0x0014)
[    0.218013] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS01], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218019] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218022] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218055] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS02], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218060] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218063] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218096] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS03], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218100] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218102] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218136] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS04], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218140] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218143] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218175] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS05], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218180] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218182] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218215] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS06], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218219] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218222] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218255] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS07], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218259] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218261] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218294] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS08], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218298] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218301] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218333] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS09], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218338] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218340] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218373] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS10], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218377] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218379] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218412] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS11], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218416] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218419] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218452] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS12], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218456] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218458] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218491] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.HS13], AE_NOT_FOUND (20201113/dswload2-162)
[    0.218498] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.218501] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.218538] ACPI BIOS Error (bug): Failure creating named object [\_SB.PCI0.XHC.RHUB.HS14._UPC], AE_ALREADY_EXISTS (20201113/dswload2-326)
[    0.218544] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20201113/psobject-220)
[    0.218546] ACPI: Skipping parse of AML opcode: Method (0x0014)
[    0.218549] ACPI BIOS Error (bug): Failure creating named object [\_SB.PCI0.XHC.RHUB.HS14._PLD], AE_ALREADY_EXISTS (20201113/dswload2-326)
[    0.218554] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20201113/psobject-220)
[    0.218556] ACPI: Skipping parse of AML opcode: Method (0x0014)
[    0.220676] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.USR1], AE_NOT_FOUND (20201113/dswload2-162)
[    0.220681] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.220683] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.220688] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.USR2], AE_NOT_FOUND (20201113/dswload2-162)
[    0.220692] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20201113/psobject-220)
[    0.220694] ACPI: Skipping parse of AML opcode: Scope (0x0010)
[    0.220731] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.XHC.RHUB.SS01], AE_NOT_FOUND (20201113/dswload2-162)
[... cut ...]
[    0.225772] ACPI BIOS Error (bug): Failure creating named object [\_SB.PCI0.PEG0.PEGP._ON], AE_ALREADY_EXISTS (20201113/dswload2-326)
[    0.225779] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20201113/psobject-220)
[    0.225782] ACPI: Skipping parse of AML opcode: Method (0x0014)
[    0.225785] ACPI BIOS Error (bug): Failure creating named object [\_SB.PCI0.PEG0.PEGP._OFF], AE_ALREADY_EXISTS (20201113/dswload2-326)
[    0.225790] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20201113/psobject-220)

Is there an upgrade available?

CristianSw, 2021-04-06:
Updated bios firmware, reinstalled arch with lts kernel. At the moment work fine. Seems that problem was in bios. Test it few days and issue can be closed.

CristianSw, 2021-04-07:
I think i catch message that you need, row 1014
[myDmesgLog.txt](https://github.com/lwfinger/rtw88/files/6269781/myDmesgLog.txt)

lwfinger, 2021-04-07:
It is line 1046: [    4.252323] rtw_8822be 0000:07:00.0: can't change power state from D3hot to D0 (config space inaccessible)

You are still getting a lot of BIOS errors such as 
[    0.245580] ACPI BIOS Error (bug): Failure creating named object [\_SB.PCI0.XHC.RHUB.TPLD], AE_ALREADY_EXISTS (20200925/dswload2-326)
[    0.245587] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20200925/psobject-220)
[    0.245590] ACPI: Skipping parse of AML opcode: Method (0x0014)

That error in Line 1046 should be fixed by adding two module options to the load. Enter the following lines:
echo -e "options rtw_pci disable_msi=y disable_aspm=y" | sudo tee /etc/modprobe.d/50-rtw_pci.conf
sudo modprobe -rv rtw_8822be
sudo modprobe -v rtw_pci

The last on of these should show those options on the line with rtw_pci.

CristianSw, 2021-04-07:
Find more detailed log that is journal.
[journal.txt](https://github.com/lwfinger/rtw88/files/6273285/journal.txt)
Hope that this one help.

CristianSw, 2021-04-07:
Catch another journal when after reboot wifi worked and after some time connection failed.
[journalWhenConnectionFailed.txt](https://github.com/lwfinger/rtw88/files/6273884/journalWhenConnectionFailed.txt)

Nfekih, 2021-04-18:
I am having the same issue. Since September 2020. Didn't find a workaround. 
kernel is 5.8.0-50-generic

magic0whi, 2021-04-20:
Same issue here, 
[dmesg.log](https://github.com/lwfinger/rtw88/files/6341117/dmesg.log)

lwfinger, 2021-04-20:
@magic0whi - You DO NOT have the same problem. If you look at the dmesg that you posted, you are getting lots of warnings and bugs logged. The first warning, often the most instructive, is for a "write RF mode table fail". Have you set the options "disable_msi=y" and "disable_aspm=y" for rtw88_pci?

By the way, you are using the kernel version of the driver. Why are you posting here?

CristianSw, 2021-04-21:
@lwfinger i added 
echo -e "options rtw_pci disable_msi=y disable_aspm=y" | sudo tee /etc/modprobe.d/50-rtw_pci.conf
sudo modprobe -rv rtw_8822be
sudo modprobe -v rtw_pci
and situation is the same, commands are show on line with rtw_pci, but wifi is not working.
Tested on the last version, on restart wifi is active but is not find any active wifi network or it cant connect to network.
[journalFailedToCOnnectToWifi.txt](https://github.com/lwfinger/rtw88/files/6350769/journalFailedToCOnnectToWifi.txt)

lwfinger, 2021-04-21:
After removing rtw_8822be, you should use 'sudo modprobe -v rtw_8822be' to reload all the drivers. That will let you know if it works without rebooting.

Sending the journal is not as good as sending the dmesg output. Run the command 'dmesg > dmesg.txt' and post dmesg.txt. On some distros, you may need to use 'sudo dmesg > dmesg.txt'.

CristianSw, 2021-04-23:
@lwfinger   here is dmesg file
[dmesg.txt](https://github.com/lwfinger/rtw88/files/6363851/dmesg.txt)

lwfinger, 2021-04-24:
Your problem is here:
rtw_8822be 0000:07:00.0: can't change power state from D3hot to D0 (config space inaccessible)

Searching for that message yields a few instances. The clearest explanation is that the BIOS has bugs. Can you update yours?

CristianSw, 2021-04-25:
@lwfinger  i updated to the last version of BIOS

lwfinger, 2021-04-25:
I have taken this as far as I can. You need to follow the instructions in the last paragraph of README.md.

closed 2021-04-25 by lwfinger

jonatanaxe, 2021-04-28:
Try to disable the wifi-power-on and then turn the notebook off and on

```
sudo sed -i 's/3/2/' /etc/NetworkManager/conf.d/default-wifi-powersave-on.conf 
```

https://gist.github.com/jonatanaxe/a59b01bfe5b2d00fa267cacaf962dbe9