lwfinger/rtw88 #52: 8723de Slow wifi speed, high ping and packet loss

antonysamuel, 2021-04-04:
in my hp laptop with rtl8723de chip WiFi speed is unstable and too slow. ping to modem is too high and also packet loss is there.
I'm getting full speed on windows and also via external WiFi adaptor. Tried manjaro , popOS, OpenSuse ,Garuda Linux with kernel 5.8 , 5.9 , 5.10 & 5.11 same issue is in all  WiFi worked otb in all distros but this issue exist.
I've installed this driver and same issue facing here too

Results of my actual WiFi speed and speed I'm getting is attached below.
Please do help me out to get it working.

[image]
[image]
[image]
[image]

lwfinger, 2021-04-04:
As I have written many times, I am not the author of this driver, and I have no knowledge of the internals of any of the Realtek chips. My role is to provide the current, up to date, kernel source code for users of older kernels, and to fix build problems for older kernels. Any problems with stability and/or performance of the chips should be taken up with the actual authors, who can be reached using the procedure described in the last paragraph of README.md.

closed 2021-04-04 by lwfinger

munsoft004, 2021-05-23:
> in my hp laptop with rtl8723de chip WiFi speed is unstable and too slow. ping to modem is too high and also packet loss is there.
> I'm getting full speed on windows and also via external WiFi [...]

I also have identical issue on my HP laptop with `RTL8723DE`. Tried on Fedora and Arch Linux. On Arch tried both default and this git still can't find any fix.

OS: Arch Linux
Kernel: linux-5.12.5

lwfinger, 2021-05-23:
I have no knowledge of the internals for any of the Realtek chips, thus my ability to fix problems with stability or performance are limited. My skills are such that I can fix build problems, and some things that issue BUG logs. The actual place to submit the stability and performance problems are outlined in the last paragraph of README.md.

There is an alternative driver available at https://github.com/smlinux/rtl8723de. Obviously, I do not control that repo, nor can I ensure that it will build on all kernels, but it might help you. That driver is derived from the same source as the Windows driver.