lwfinger/rtw88 #53: Debian 10

c0br4x0r, 2021-04-07:
After installing Debian 10, install the firmware to be able to use Wifi, from rtw88.git.
I was able to use the Wifi of my HP Proobok.
However, after a while, the following error begins to be generated and after that, my notebook is crach.
I attach features of my OS and errors before crash.

Thank you very much for your help!

root@mjunhytgb:~# cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

root@mjunhytgb:~# uname -a
Linux mjunhytgb 4.19.0-16-amd64 #1 SMP Debian 4.19.181-1 (2021-03-19) x86_64 GNU/Linux

syslog

Apr  7 00:17:02 mjunhytgb kernel: [19031.285444] rtw_8821ce 0000:02:00.0: firmware failed to leave lps state
Apr  7 00:17:22 mjunhytgb kernel: [19051.281404] rtw_8821ce 0000:02:00.0: firmware failed to leave lps state
Apr  7 00:22:45 mjunhytgb kernel: [19374.591525] broken atomic modeset userspace detected, disabling atomic

lwfinger, 2021-04-07:
This problem is one that I have not seen before. Try running the following commands to see if that helps:

echo -e "options rtw_pci disable_msi=y disable_aspm=y" | sudo tee /etc/modprobe.d/50-rtw_pci.conf
sudo modprobe -rv rtw_8822be
sudo modprobe -v rtw_8822be

c0br4x0r, 2021-04-08:
Hello, thank you very much for your response.
I made some BIOS changes, disable bluetooth, wake on lan and power management. 
I'll add the changes you tell me and let you know what's going on.

Thanks a lot.

dubhater, 2021-05-03:
@c0br4x0r Is your wifi card of the RFE 2 type?

Linux 5.12 brought support for RFE 2 cards, so I tried it out, and now my system freezes pretty often when I use speedtest.net. Also with lighter wifi traffic, like when I open Firefox and it loads whatever page I had open, but speedtest.net is a more reliable trigger. There's another user with the same problem here: https://www.reddit.com/r/archlinux/comments/n0zoo5/driver_issue_rtw88_8821ce/ and I bet the "crash" in https://github.com/lwfinger/rtw88/issues/35 was the same thing.

No useful messages came through netconsole. (I used the kernel parameter loglevel=7.)

disable_aspm=1 did not help.

Now I'm trying out disable_msi=1.

dubhater, 2021-05-04:
disable_msi=1 seemed promising, but it just took longer to freeze. I tried all combinations of the two parameters and nothing helps.

lwfinger, 2021-05-04:
Since you cannot provide any log information concerning the freeze, I am unable to help with it.

The usual place to report such problems is at linux-wireless@vger.kernel.org as stated in the last paragraph of README.md.

closed 2021-06-07 by lwfinger