lwfinger/rtw88 #54: wifi doesnt work 5.10.0-kali7-amd64 with RTL8723DE

747parker, 2021-05-01:
I'm using 5.10.0-kali7-amd64 with RTL8723DE controller; following the steps wifi doesnt work. I've been watching and reading lot of info about but nothing has work.

No, no error during or after installing. I'm running in dual boot with Windows 10, could be this the cause of this malfunctioning?

more info:
Linux kali 5.10.0-kali7-amd64 #1 SMP Debian 5.10.28-1kali1 (2021-04-12) x86_64 GNU/Linux
Realtek Semiconductor Co., Ltd. RTL8723DE 802.11b/g/n PCIe Adapter
└─$ iwconfig                                                                 
lo        no wireless extensions.
eth0      no wireless extensions.

I dont know what to do now, Ive follow lot of instructions but nothing has work. I need your help.

lwfinger, 2021-05-01:
Windows might be a problem. Make sure that the Windows quick boot is turned off. That might cause the interface t be left in a strange condition.

Are there any pertinent messages in the output of the 'dmesg' command?

747parker, 2021-05-01:
This are the relevant messages from dmesg. 

[dmesg.txt](https://github.com/lwfinger/rtw88/files/6410326/dmesg.txt)

lwfinger, 2021-05-01:
It appears that both rtw_core and rtw88_core are loading. If you are using this repo, you must blacklist the kernel version.

747parker, 2021-05-02:
Could you please tell me how to do it? Its not so clear for me.

lwfinger, 2021-05-02:
sudo edit /etc/modprobe.d/blacklist.conf

At the bottom of the file, add lines that say
blacklist rtw88_8723de
blacklist rtw88_8723d
blacklist rtw88_core
blacklist rtw88_pci

747parker, 2021-05-02:
Now it works. Thank you I've been fighting with this issue for many days. 
[image]