lwfinger/rtw88 #55: modprobe error in Pop!_OS 21.04 beta

lbkNhubert, 2021-06-02:
Followed steps in readme.

zzz@zzz:~$ sudo modprobe rtw_8822ce
modprobe: ERROR: could not insert 'rtw_8822ce': Exec format error

zzz@zzz:~/Downloads/rtw88$ uname -a
Linux pop-os 5.11.0-7614-generic #15~1618626693~21.04~ecb25cd-Ubuntu SMP Thu Apr 22 15:59:53 UTC  x86_64 x86_64 x86_64 GNU/Linux

zzz@zzz:~/Downloads/rtw88$ sudo dmesg | grep rtw
[    4.766252] rtw_core: disagrees about version of symbol module_layout
[  340.548457] rtw_core: disagrees about version of symbol module_layout

Guessing that the current code is not compatible with the kernel version.

lwfinger, 2021-06-02:
The kernel source that you compiled against does not match your running kernel. The code for rtw88 is fine.

lbkNhubert, 2021-06-02:
Thanks for the reply. I ran make clean, then make, and sudo make install, and the driver is working, The make command had entries of the form: Skipping BTF generation for /mypath/rtw88/rtw_8822ce.ko due to unavailability of vmlinux. I'll see if I can make sure that they kernel source is up to date and retry the above. Thank you again for the help and for the quick response.

closed 2021-06-07 by lwfinger