lwfinger/rtw88 #56: rtl8822ce drivers don't work after suspend in Ubuntu 20.04.

zee-bit, 2021-06-04:
logs after system wake-up:
```console
$ sudo dmesg | grep rtw
```
```console
[34438.830775] rtw_8822ce 0000:03:00.0: can't change power state from D3hot to D0 (config space inaccessible)
[34442.011457] rtw_8822ce 0000:03:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[34442.011464] rtw_8822ce 0000:03:00.0: mac power on failed
[34442.011465] rtw_8822ce 0000:03:00.0: failed to power on mac
[34444.022399] rtw_8822ce 0000:03:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[34444.022409] rtw_8822ce 0000:03:00.0: mac power on failed
[34444.022411] rtw_8822ce 0000:03:00.0: failed to power on mac
```
```console
$ sudo lshw -C network
```
```console
...
*-generic DISABLED
       description: Wireless interface
       product: RTL8822CE 802.11ac PCIe Wireless Network Adapter
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 1
       bus info: pci@0000:03:00.0
       logical name: wlp3s0
       version: ff
       serial: 80:30:49:3e:40:23
       width: 32 bits
       clock: 66MHz
       capabilities: bus_master vga_palette cap_list ethernet physical wireless
       configuration: broadcast=yes driver=rtw_8822ce driverversion=5.10.41-xanmod1 firmware=N/A latency=255 link=no maxlatency=255 mingnt=255 multicast=yes wireless=IEEE 802.11
       resources: irq:78 ioport:d000(size=256) memory:fc800000-fc80ffff
...
```

lwfinger, 2021-06-07:
On my system, rtw_8888ce survives both sleep and hibernation. Most systems with power state changes are fixed by adding the options "disable_msi=y", "disable_aspm=y", or both.

kelebek333, 2021-06-11:
You can create a file related suspend action. Run following command.

```
sudo mkdir -p /lib/systemd/system-sleep
sudo touch /lib/systemd/system-sleep/rtw_8821ce
sudo gio open /lib/systemd/system-sleep/rtw_8821ce
```

Paste following lines to text editor and save it.

```
#!/bin/sh

# rtw_8821ce suspend

case $1 in
    pre)  rmmod rtw_8821ce rtw_8821c rtw_pci rtw_core ;;
    post) modprobe rtw_8821ce  ;;
esac
```

So, the driver module is unloaded when suspend and reloaded again when wakes up.

lwfinger, 2021-06-11:
This should work. Note that the pre) line can be simplified to pre) modprobe -r rtw_8821ce. Just as modprobe rtw_8821ce loads all the required dependencies, the modprobe -r will remove them.

zee-bit, 2021-06-23:
Thanks, @kelebek333 and @lwfinger for the prompt response.
Actually, this was more of a kernel issue for me. I haven't faced this after I downgraded my kernel to `5.10.34-xanmod`. Everything works fine for me now. The driver now works even after suspend or hibernate. :)

closed 2021-06-23 by zee-bit