lwfinger/rtw88 #57: dkms support?

brainchild0, 2021-06-06:
Does this code base have support for `dkms`, which is not mentioned in the documentation? (Many driver repositories have a `dkms-install.sh` item at the top level.)

lwfinger, 2021-06-07:
The repo provides dkms.conf. It is up to you to install it. I do not use dkms as I want better control of drivers.

closed 2021-06-07 by lwfinger

brainchild0, 2021-06-07:
Maybe  a brief note in the documentation for how to install using DKMS, for users so inclined but without the experience. A note of one line along with a representation of the command or commands to execute may be completely adequate. It is not obvious to everyone, as some have yet to  use the tool in this way.

KaKi87, 2023-09-19:
Hello, did you find out what the command is ? Thanks

lwfinger, 2023-09-19:
Once again - I do not support dkms. All supporting dkms files have been submitted by others. If someone wants to add a section on dkms to the README, I will publish it.

For me, 'make' and 'sudo make install' whenever your kernel changes is not a big deal.

KaKi87, 2023-09-19:
> Once again - I do not support dkms.

I know, that's why I was asking @brainchild0.

> For me, 'make' and 'sudo make install' whenever your kernel changes is not a big deal.

I'm setting up a scrap laptop for an elderly person, so I can't ask of them to do that.

Meanwhile, they learned to perform updates using the GUI so they wouldn't depend on me, therefore if I can't workaround this issue then they're still going to depend on me because their wifi will break every once in a while.

brianwitte, 2023-09-20:
@KaKi87,

First of all, thanks for following up about this and letting us know more about your use case.  Although Larry and I do not currently support dkms directly right now, I am going to make an effort to improve the documentation (as suggested by @brainchild0) around it at the very least.

Second, I see how you need something easy for the elderly person who is using the machine so let me think through it a bit with you (this is obviously outside the scope of this repo, but you are doing a good thing and I should help you if I can) so below are my initial thoughts.

1. Most Linux distributions and desktop environments should support `.desktop` entry like documented here -> https://wiki.archlinux.org/title/desktop_entries. Perhaps, you can create one that can execute a script containing the commands for building and reinstalling the driver. I am not sure how permissions would work with this or how they would be prompted for their password, but something like this should be straightforward with a little research. 

2. There are lightweight GUI tools like [Gnome's Zenity](https://help.gnome.org/users/zenity/stable/) that may be an option. It may look something like:

```
#!/bin/bash
PASSWORD=$(zenity --password --title="Authentication required")
echo $PASSWORD | sudo -S sh -c "cd /path/to/driver && make clean && make && make install"
```

I am not sure if either of those are viable for you, but they seem like they may work well under the right circumstances.