lwfinger/rtw88 #58: fsleep is showing error ( file rtw8822c.c )

OyePuru, 2021-06-08:
I cloned it as said in readme.md while making it it show me an error

Implict declaration of function and showing ssleep as suggestion 

Line no 1480 

Could me tell how can i install this driver or else where should i do from 

I even edited them to ssleep but after installing and rebooting it didn't showed me any wifidriver 

Then i uninstalled and again did the same 

I am quite new to linux envoirment could you help me with this :)

lwfinger, 2021-06-08:
What kernel are you running. Run the 'uname -r' command and post it here.

I fixed the missing fsleep situation some time ago, but perhaps I got it wrong.

OyePuru, 2021-06-09:
5.7.0-2parrot2-amd64

OyePuru, 2021-06-09:
/home/oyepuru/rtw88/rtw8822c.c: In function ‘rtw8822c_txgapk_read_offset’:
/home/oyepuru/rtw88/rtw8822c.c:1480:2: error: implicit declaration of function ‘fsleep’; did you mean ‘ssleep’? [-Werror=implicit-function-declaration]
 1480 |  fsleep(1000);
      |  ^~~~~~
      |  ssleep

While making the file i am facing this

closed 2021-06-29 by OyePuru

michaldorsett, 2022-02-24:
I just cloned the latest, and I still am receiving that same error. Where was this fixed? Doing a blame shows that part of the code has not changed in 10 months.

lwfinger, 2022-02-24:
According to 'git blame main.h', the condition for using the internal fsleep was changed 2022/02/05, or 19 days ago.

Perhaps it would help if you were to specify what distro and what kernel you are using! Obviously, your distro has backported the definition of fsleep to a kernel older than 5.8. As noted in README.md, this is a case that I do not handle unless you provide the test to exclude your distro. That is where the RHEL part came from.