lwfinger/rtw88 #59: Slow WIFI speeds on RTL8723DE

sortedcord, 2021-06-24:
I am currently using a vanilla arch installation with networkmanager as my network manager and the WIFI speeds are terribly slow. On Windows10 on the same machine I get around 60-70mbps while on Linux it doesn't even reach 20mbps. The network also drops frequently. This didn't use to happen before but when I upgraded to a newer Linux kernel, I have been facing such issues ( I don't remember exactly what kernel version it is but https://github.com/smlinux/rtl8723de drivers were working fine for me. And as far as I know they are not supported anymore). I tried vanilla Debian, Endevour, Artix, Mint, Manjaro and got similar results. Any Idea why this is happening and if anyone else is facing the same issue

lwfinger, 2021-06-25:
If you found a driver that worked for you, but it is not supported, then you will likely need to learn enough about kernel programming  to support it yourself. As you do not give any specifics regarding what kernel you are now using, I'm not sure what we can do for you. Should you even have posted on the repo?

closed 2021-06-25 by lwfinger

kelebek333, 2021-06-30:
>  ( I don't remember exactly what kernel version it is but https://github.com/smlinux/rtl8723de drivers were working fine for me. And as far as I know they are not supported anymore). 

This driver can build up to Linux 5.10.x on any Linux disto. So you can use this drive for a long time with Linux 5.10 LTS or 5.4 LTS kernels. 

You can install this drive with the following commands in a Ubuntu based distribution (eg Linux Mint). I have added patch for newer kernel version to the driver of the PPA version. It also incluede a blacklist file for preventing conflict with rtw88 modules.

```
sudo add-apt-repository ppa:kelebek333/kablosuz
sudo apt-get update
sudo apt install rtl8723de-dkms
```

BoskyWSMFN, 2021-06-30:
> I have added patch to the driver of the PPA version for newer kernel version

Just checked with 5.11 - no go.
[make.log](https://github.com/lwfinger/rtw88/files/6743100/make.log)

kelebek333, 2021-06-30:
Did you install headers package? I am using kernel 5.13 and driver successfully built.

```
dkms status rtl8723de
rtl8723de, 5.1.1.8~ppa, 5.12.13-xanmod1, x86_64: installed
rtl8723de, 5.1.1.8~ppa, 5.13-051300-generic, x86_64: installed
```

BoskyWSMFN, 2021-06-30:
> Did you install headers package for kernel version?

Yes. Removed header for 5.11.0-16 and 5.11.0-18.
```
Building for 5.11.0-16-generic, 5.11.0-18-generic and 5.11.0-22-generic
Module build for kernel 5.11.0-16-generic was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 5.11.0-18-generic was skipped since the
kernel headers for this kernel does not seem to be installed.
Building initial module for 5.11.0-22-generic
```

kelebek333, 2021-06-30:
I haven't updated PPA repository for hirsute base. Although I added the updated package for other bases, I did not add it for the hirsute base. This is how I was able to spot it. Now, You can try to install with the updated package.

BoskyWSMFN, 2021-06-30:
> I haven't updated PPA repository for hirsute base. Although I added the updated package for other bases (xenial, bionic, focal, groovy etc.), I did not add it for the hirsute base. This is how I was able to spot it. Now, You can try to install with the updated package.

Oh, right! I totally forgotten I had made dist upgrade recently and to put uname output here. Sorry for it.
Package with your patch (for focal) installed successfully. Thank you!

kelebek333, 2021-06-30:
> EDIT
> And Network Manager and something else failed after reboot. =) Can't even load kernel.
> Don't worry, I was about to rollback anyway.

I don't think the problem is driver related.  Did you try to disable secureboot?

If you still want to remove rtl8723de driver, boot the system in recovery mode. Login with root and run following command for write permission.

`mount -o remount,rw /`

Then you can uninstall the package with the following command. 

`sudo apt purge rtl8723de-dkms`

BoskyWSMFN, 2021-07-01:
> I don't think the problem is driver related. Did you try to disable secureboot?

You're right, the problem isn't driver related. Yes, I disabled secured boot before installing.
The problem is in the base. After dist-upgrade something was broken (network, wireless, sound etc were affected) and I can't figure out what exactly. I am sure clean install or roll back will help.