lwfinger/rtw88 #61: firmware failed to leave lps state

pooh2pooh, 2021-06-30:
Hi! The following errors constantly appear in the log. The WiFi connection is stable at the same time.

```
июн 30 19:39:31 not-a-virus kernel: rtw_8822ce 0000:01:00.0: failed to send h2c command
июн 30 19:39:31 not-a-virus kernel: rtw_8822ce 0000:01:00.0: failed to send h2c command
июн 30 19:39:31 not-a-virus kernel: rtw_8822ce 0000:01:00.0: failed to send h2c command
июн 30 19:39:31 not-a-virus kernel: rtw_8822ce 0000:01:00.0: failed to send h2c command
июн 30 19:39:31 not-a-virus kernel: rtw_8822ce 0000:01:00.0: failed to send h2c command
июн 30 19:39:31 not-a-virus kernel: rtw_8822ce 0000:01:00.0: failed to send h2c command
июн 30 19:39:31 not-a-virus kernel: rtw_8822ce 0000:01:00.0: firmware failed to leave lps state
```

Fak3, 2022-01-14:
I can confirm this bug on my lenovo laptop:
```
04:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter
        Subsystem: Lenovo Device c123
        Flags: bus master, fast devsel, latency 0, IRQ 79, IOMMU group 12
        I/O ports at 2000 [size=256]
        Memory at c3600000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-c8-22-01
        Capabilities: [158] Latency Tolerance Reporting
        Capabilities: [160] L1 PM Substates
        Kernel driver in use: rtw_8822ce
        Kernel modules: rtw88_8822ce
```

i tried to add `options rtw88_pci disable_aspm=1` to my /etc/modprobe.d/local.conf but still have my system journal and dmesg flooded with errors
```
[    3.516129] rtw_8822ce 0000:04:00.0: enabling device (0000 -> 0003)
[    3.523097] rtw_8822ce 0000:04:00.0: Firmware version 9.9.4, H2C version 15
[    3.530951] rtw_8822ce 0000:04:00.0: Firmware version 9.9.10, H2C version 15
[    3.999011] rtw_8822ce 0000:04:00.0 wlp4s0: renamed from wlan0
[    5.324088] rtw_8822ce 0000:04:00.0: start vif dc:e9:94:7e:5f:85 on port 0
[    5.325431] rtw_8822ce 0000:04:00.0: stop vif dc:e9:94:7e:5f:85 on port 0
[    5.561473] rtw_8822ce 0000:04:00.0: start vif ee:c7:7c:0e:6b:3c on port 0
[    9.847260] rtw_8822ce 0000:04:00.0: stop vif ee:c7:7c:0e:6b:3c on port 0
[   10.085443] rtw_8822ce 0000:04:00.0: start vif dc:e9:94:7e:5f:85 on port 0
[   14.738024] rtw_8822ce 0000:04:00.0: sta 10:fe:ed:51:2a:94 joined with macid 0
[   49.030634] rtw_8822ce 0000:04:00.0: firmware failed to leave lps state
[   49.033828] rtw_8822ce 0000:04:00.0: failed to send h2c command
[   49.036994] rtw_8822ce 0000:04:00.0: failed to send h2c command
```

any solution?

Fak3, 2022-01-15:
Update: the message flood only happens when laptop charger is not connected.
1) I connected the charger. The "firmware failed to leave lps state" message stopped to appear, but "failed to send h2c command" message still continuously flooded my dmesg
2) I put laptop to sleep, then wake up. Message flood stopped.
3) I disconnected the charger, and after few minutes message started to come in again.

Fak3, 2022-01-15:
FWIW I have reported issue here as well: https://bugzilla.kernel.org/show_bug.cgi?id=215496

lwfinger, 2022-01-15:
Surprisingly, almost no one pays any attention to the kernel bugzilla, which is why the README.md tells you how to file bug information.

If it happens only when the charger is not connected, then I suspect that your laptop has an internal voltage problem. The voltage supplied to the power regulators will be somewhat higher when plugged into power than when operating on the battery. Can you contact your laptop supplier? Perhaps there is a recall, or a BIOS update to handle this issue.

pooh2pooh, 2022-01-15:
This is more like incorrect management of the device's power saving.

I managed to fix it. But I don't remember exactly how. If I remember, I'll tell you about it.

Fak3, 2022-01-18:
@pooh2pooh do you use bluetooth devices by any chance?

pooh2pooh, 2022-01-20:
> @pooh2pooh do you use bluetooth devices by any chance?

yes. bluetooth mouse and headphones

rilwansmith, 2022-01-27:
> This is more like incorrect management of the device's power saving.
> 
> I managed to fix it. But I don't remember exactly how. If I remember, I'll tell you about it.
> 
> I contacted the kernel developers, they pointed in the right direction.

I'm interested in the fix because this same problem is occurring on KDE NEON HP laptop

lwfinger, 2022-01-27:
My best guess would be to set "disable_lps_deep=y" option for rtw88_core.

rilwansmith, 2022-01-28:
I solved this problem by permanently disabling the power_save feature also known as power management for the wireless interface of my KDE NEON HP Laptop.
Use a text editor to open the file /etc/NetworkManager/conf.d/default-wifi-powersave-on.conf.
Replace the content of the file with the below text.

```
[connection]
wifi.powersave = 2
```
Then reboot your computer.

Fak3, 2022-01-28:
@rilwansmith do you use Bluetooth mouse by any chance? Trying to figure out if it is relevant.

rilwansmith, 2022-01-28:
No

> @rilwansmith do you use Bluetooth mouse by any chance? Trying to figure out if it is relevant.

No.

lwfinger, 2022-02-28:
Like all module options, you need to create (as root) a file /etc/modules.d/50-rtw88.conf. In this file, add a single line that says "options rtw88_core disable_lps_deep=y" and reboot.

Beyley, 2022-03-01:
> Like all module options, you need to create (as root) a file /etc/modules.d/50-rtw88.conf. In this file, add a single line that says "options rtw88_core disable_lps_deep=y" and reboot.

I tried that and sadly i get the exact same issue, no change on dmesg either

lwfinger, 2022-03-01:
Actually, the above line is for the in-kernel driver. To use this one, the line should be 'options rtw_core disable_lps_deep=y'

You can check that you got it correctly by the following:

sudo modprobe -rv rtw_8821ce
sudo modprobe -v rtw_8821ce

You should see the option on the rtw_core line for the second command.

Beyley, 2022-03-01:
> Actually, the above line is for the in-kernel driver. To use this one, the line should be 'options rtw_core disable_lps_deep=y'
>
> You can check that you got it correctly by the following:
>
> sudo [...]

I see the option on the second command, yet still the same issue and no change to dmesg output

lwfinger, 2022-03-01:
Which driiver are you using - in-kernel or from repo?

Beyley, 2022-03-01:
> Which driiver are you using - in-kernel or from repo?

Ive tried both (exact same issue), currently the one in the repo is the one in use

lwfinger, 2022-03-01:
There are two rtw_pci options to try, namely "disable_msi=y" and "disable_aspm=y". add a second line to your options file that says

options rtw_pci disable_msi=y disable_aspm=y

Beyley, 2022-03-01:
same problem, heres the [dmesg log](https://cdn.discordapp.com/attachments/762039175683244072/948296624893997156/afterrtw_pci.log) grepped to `rtw`

FOSSBOSS, 2023-05-08:
what is lps anyway?
is that "low power"?
I see this message when im out in the yard with my laptop, but not when I'm indoors near the wifi router. might it be signal power related?

lwfinger, 2023-05-09:
I think lps means low power state. Your observations indicate that it is related to the signal strength.

What device do you have? Repo rtw88 now drives 12 different ones - each with their pecularities.

FavoritoHJS, 2023-12-25:
bump, have a RTL8822BE adapter with this issue - after a few minutes a `rtw_8822be 0000:06:00.0: firmware failed to leave lps state` appears in the journal every minute and a half.
tried setting kernel options and disabling powersave on networkmanager, neither worked... on 2,4GHz.
on 5GHz i had 0 journal spam for multiple hours. Interesting...
ps: issue might be related to #82

tortured1, 2024-02-18:
I'm setting up a new HP laptop with Fedora 39.  The rtw88_882ce driver built in to the 6.5.6-300.fc39.x86_64 kernel works just fine.  The problem comes in when I update the laptop, and the kernel gets updated to 6.7.4-200.fc39, and I install and use Cinnamon as my desktop.

Under Cinnamon, I get a bunch of different errors with the rtw_882ce driver like "device [10ec:c822] error status/mask=00004000/004000000]" and then the "firmware failed to leave lps state" and "failed to send h2c command" errors.

To recap all of the suggestions made in this thread...

I tried creating /etc/NetworkManager/conf.d/default-wifi-powersave-on.conf and putting this in the file:

```
[connection]
wifi.powersave = 2
```

I even tried adding the "wifi.powersave=2" into the connection entry in /etc/NetworkManager/system-connections/mynetwork.nmconnection file.  No go.

My *plan* was to create /etc/modules.d/50-rtw88.conf, and put the following in there, HOWEVER, /etc/modules.d doesn't exist in Fedora 39.

```
options rtw88_core disable_lps_deep=y
options rtw88_pci disable_msi=y disable_aspm=y
options rtw_core disable_lps_deep=y
options rtw_pci disable_msi=y disable_aspm=y
```

There seemed to be two different versions of things, one with the 88 and one without, I figured I'd put them both in there.  I'm unsure of where to actually put these on my system, though.

It didn't make any difference if the laptop was plugged in or not.  I could watch the connection bounce up and down. 

The really strange part is that the network connection is fine if I go back to GNOME, and I don't get any of those errors. 

What desktop are other people having this problem using?

lwfinger, 2024-02-18:
rtw88_882ce is not a valid driver name. I guess it is an 8822ce and the driver is rtw88_8822ce.

The difference in the names in the options file is that one set is for the kernel drivers, and one is for this repo.

I just upgraded a Fedora 38 virtual machine to Fedora 39. The directory  /etc/modprobe.d/ is still there. You are putting the  options file in the wrong place.

I generally use either the KDE desktop, or xFce if I want a lighter weight desktop. The difference between desktops is that the user-space code to support wifi devices is different. If you use NetworkManager in each, there should not be much difference.

tortured1, 2024-02-18:
Obsv, that was a transcription error, as I couldn't copy/paste the errors without a network connection.  Absolutely a catch-22 situation, yes.

As far as I can tell, both desktops are using the same NetworkManager.  It's very odd.

As you pointed out, there is an /etc/*modprobe*.d, but the advice given earlier was to put it into a file in /etc/*modules*.d, which doesn't exist.  Putting the options into `/etc/modprobe.d/rtw_8822ce.conf` (or local.conf, which I did first, but decided to give it a more meaningful name) did work.

Also, FWIW, I had left out the disable_msi=y the first time through, and things didn't work, but after I added it, it did.  I didn't try other permutations.

Felipe-Freire, 2024-08-06:
> I solved this problem by permanently disabling the power_save feature also known as power management for the wireless interface of my KDE NEON HP Laptop. Use a text editor to open the file /etc/NetworkManager/conf.d/default-wifi-powersave-on.conf Replace the content of the file with the below text.
> 
> ```
> [connection]
> wifi.powersave = 2
> ```
> 
> Then reboot your computer. Problem solved.

This worked for me.

The problem appeared when connecting a bluetooth headset. I monitored the warnings with:
```
sudo dmesg | grep rtw_8821ce
```
When I restarted the machine, the problem was no longer verified.

griffi-gh, 2024-08-14:
my workaround for nixos:
```
  networking.networkmanager.wifi.powersave = false;
  boot.extraModprobeConfig = ''
    options rtw88_core disable_lps_deep=y
    options rtw88_pci disable_msi=y disable_aspm=y
    options rtw_core disable_lps_deep=y
    options rtw_pci disable_msi=y disable_aspm=y
  '';
```

M-Try, 2024-09-03:
same issue here! RealTek RTW 8822ce using the driver `rtw_8822ce` on a Lenovo G4 V15, running Arch, `iwd`, `wpa_supplicant` and `systemd-resolved`. Previously used NetworkManager.
After a long time of adding random configs I have now gotten myself into a state where the WiFi module just *will not connect* to any WiFi network (stuck on "connecting" according to `iwctl`). I have copied the config from @griffi-gh, this however does not fix the issue. Blacklisting the `rtw_8822ce` driver just disables wireless networking (obviously).

The WiFi card is *known* good. Windows can use it flawlessly. IIRC the Arch Installer also handled it correctly. Notably, the Arch Installer it seems also uses `iw` as a driver according to `lspci`.

This is where I'm stuck. Same errors as everyone else, `failed to send h2c command` and `firmware failed to leave lps state`. Any other ideas?

M-Try, 2024-09-04:
> [...]. IIRC the Arch Installer also handled it correctly. Notably, the Arch Installer it seems also uses `iw` as a driver according to `lspci`. 

I misread. The Arch Installer additionally uses `lw` as a driver, not `iw`, which is not a driver. `lw` is a Broadcom driver. Im unsure if this makes any difference in this case at all.
After double checking I can confirm: Yes, at the very least on my machine, the Arch installer *works with networking* and uses the rtw88 drivers. Just like my main install, the installer uses `iwd` and `systemd-resolved`. The installer in question is the installer version from April of 2024. No "`firmware failed to leave lps state`" or "`failed to send h2c command`". There are no suspicious messages in the kernel log at all; It's reliable, even after suspending and subsequently waking the system.

dubhater, 2024-09-04:
@M-Try You must have installed something that is causing problems. Maybe tlp?

lahwaacz, 2024-09-04:
If you have an installer from April and an updated system, there are definitely different versions of relevant software like the `linux` and `linux-firmware` packages. Can you check with the latest installer if it works reliably or not?

micsthepick, 2024-09-04:
```
[1247143.139280] rtw_8822be 0000:03:00.0: firmware failed to leave lps state
```
Also running arch.
any further relavant info required, please  let me know

micsthepick, 2024-09-04:
```
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter
	Subsystem: ASUSTeK Computer Inc. Device 8746
	Flags: bus master, fast devsel, latency 0, IRQ 154
	I/O ports at d000 [size=256]
	Memory at f7300000 (64-bit, non-prefetchable) [size=64K]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, IntMsgNum 0
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-b8-22-01
	Capabilities: [158] Latency Tolerance Reporting
	Capabilities: [160] L1 PM Substates
	Kernel driver in use: rtw_8822be
```

user7z, 2024-11-13:
my systemd hangs , it freezes when i saw journal i saw this : 
Nov 13 23:04:52 lp rtkit-daemon[3555]: Supervising 8 threads of 5 processes of 1 users.
Nov 13 23:04:52 lp rtkit-daemon[3555]: Supervising 8 threads of 5 processes of 1 users.
Nov 13 23:04:53 lp kwin_wayland[3528]: kwin_libinput: Libinput: client bug: timer event9 tap: scheduled expiry is in the past (-68ms), your system is too slow
Nov 13 23:04:53 lp kwin_wayland[3528]: kwin_libinput: Libinput: client bug: timer event9 hold: scheduled expiry is in the past (-208ms), your system is too slow
Nov 13 23:04:55 lp systemd[3429]: Started Htop - Process Viewer.
░░ Subject: A start job for unit UNIT has finished successfully
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit UNIT has finished successfully.
░░ 
░░ The job identifier is 742.
Nov 13 23:04:55 lp plasmashell[3672]: The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
Nov 13 23:04:57 lp wpa_supplicant[3290]: wlo1: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-52 noise=9999 txrate=520000
Nov 13 23:05:03 lp kernel: rtw_8822ce 0000:01:00.0: firmware failed to leave lps state
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:06 lp kwin_wayland_wrapper[3605]: Key repeat discarded, Wayland compositor doesn't seem to be processing events fast enough!
Nov 13 23:05:09 lp kernel: rtw_8822ce 0000:01:00.0: firmware failed to leave lps state
Nov 13 23:05:11 lp kernel: rtw_8822ce 0000:01:00.0: firmware failed to leave lps state

what hangs my laptop is a systemd reflection , my laptops has freezed for a 4 seconds , what about this rtw_8822ce thing

curvi, 2025-04-04:
I'm sorry, if it's off topic a bit, but i had both, the "failed to leave lps" (~every second) and "failed to send h2c command" (~5 times/second) on a rtw8821ce module I just bought in a mini-pc. I tried everything, and disabling lps got rid of this one, but not the h2c errors.
The solution [was here](https://forum.openmediavault.org/index.php?thread/49571-rtw-8822ce-failed-to-send-h2c-command/) and is as simple as removing and re-inserting the wifi card. As I said, disabling the low power state with the file in /etc/modprobe.d/50-rtw_8821.conf got rid of the "failed to leave lps" errors, but was apparently not the cause for my errors. Thanks for the tips anyway!

anthony-lutjen, 2025-05-03:
I too kept getting `rtw_8821ce 0000:01:00.0: failed to send h2c command` too. Couldn't connect to NTP and after fixing config and restarting systemd-timesyncd,  it went away. FWIW the interface did go down and reconnected.

Still getting `rtw_8821ce 0000:01:00.0: firmware failed to leave lps state` before disabling lps