lwfinger/rtw88 #64: Missing file rtl8822cu_config.bin

gcem, 2021-08-11:
Hi! I have a computer with RTL8822CE. I have installed the latest drivers (87f32b6) using

```
make
sudo make install
```

Now, WiFi works perfectly. But I can't use Bluetooth.

When I checked `dmesg`, I have seen the following:

```
[    3.166314] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000c lmp_ver=0a lmp_subver=8822
[    3.168309] Bluetooth: hci0: RTL: rom_version status=0 version=3
[    3.168311] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822cu_fw.bin
[    3.168328] bluetooth hci0: firmware: failed to load rtl_bt/rtl8822cu_fw.bin (-2)
[    3.168353] bluetooth hci0: Direct firmware load for rtl_bt/rtl8822cu_fw.bin failed with error -2
[    3.168355] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8822cu_fw.bin not found
```

The `rtl8822cu_fw.bin` file was in `/lib/firmware/rtw88/`. I moved it to `/lib/firmware/rtl_bt/` as hinted by the error. This solved the errors above but now I get another error:

```
[  284.579185] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000c lmp_ver=0a lmp_subver=8822
[  284.581169] Bluetooth: hci0: RTL: rom_version status=0 version=3
[  284.581175] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822cu_fw.bin
[  284.588938] bluetooth hci0: firmware: direct-loading firmware rtl_bt/rtl8822cu_fw.bin
[  284.588976] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822cu_config.bin
[  284.589020] bluetooth hci0: firmware: failed to load rtl_bt/rtl8822cu_config.bin (-2)
[  284.589025] bluetooth hci0: Direct firmware load for rtl_bt/rtl8822cu_config.bin failed with error -2
```

It doesn't look like there is a `rtl8822cu_config.bin` in the repository. Do I need to create it in some other way? Thanks a lot!

If it helps, my kernel version is 5.10.0-8-amd64, I am on Debian Sid.

kelebek333, 2021-08-11:
You can try to install that package. http://ftp.br.debian.org/debian/pool/non-free/f/firmware-nonfree/firmware-realtek_20210315-3_all.deb

gcem, 2021-08-11:
Thank you @kelebek333. I have also found the file here:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtl_bt/

I moved the file to my `rtl_bt` and now everything is working fine. But maybe installing the package would be easier, so I hope it helps others with the same issue.

I wonder if the file needs to be added to this repository too.

closed 2021-08-11 by lwfinger