lwfinger/rtw88 #65: Sudo modprobe rtw_8723de Modprobe: ERROR: could not insert rtw_8723de operation not permitted

chirag900, 2021-09-03:
I am not able to use wifi on my ubuntu 20.04 laptop. 
I followed these instructions:

1. sudo apt-get update
sudo apt-get install make gcc linux-headers-$(uname -r) build-essential git

2.git clone https://github.com/lwfinger/rtw88.git
cd rtw88
make
sudo make install
Install rtw88 success

3. Sudo modprobe -r rtw_8723de

4. Sudo modprobe rtw_8723de 
Modprobe: ERROR: could not insert rtw_8723de operation not permitted.

Could you please help me with this.

Thank you so much.

lwfinger, 2021-09-04:
You should look at the dmesg output to see what has gone wrong. My suspicion is that you have secure boot enabled in the BIOS, but the module is not signed. Note that I do not sign my modules. There are some descriptions of methods to perform that operation, or you can disable secure boot in the BIOS.

sribooshan, 2021-09-10:
@lwfinger I disabled secure boot in BIOS and executed ```sudo modprobe rtw_8822be```
There was no output but suddenly, my system started detecting Wi-Fi networks near me and I was able to connect with my home Wi-Fi. After probably a minute, I tried to open a browser but no website loaded.  And then in the settings, all nearby Wi-Fi networks disappeared from the list. How to solve this issue? I badly need Wi-Fi on Ubuntu.

I use Ubuntu 20.04.3 LTS, kernel version: 5.11.0-27-generic
Wi-Fi hardware: RealTek 8822BE

Please help me.

kelebek333, 2021-09-10:
Try to run following commands for disabling power management of network manager.

`echo -e '[connection]\nwifi.powersave = 2' | sudo tee /etc/NetworkManager/conf.d/default-wifi-powersave-on.conf`
`sudo systemctl restart NetworkManager`

sribooshan, 2021-09-10:
@kelebek333  Thank you for your help. But, it does not work. Wi-Fi is not found even in the Settings window.

kelebek333, 2021-09-10:
Install netcat-openbsd package and post outputs of following commands.

```
sudo dmesg  | nc termbin.com 9999
NetworkManager --print-config
lsmod | grep rtw
dkms status
```

kelebek333, 2021-09-10:
Install **netcat-openbsd** and **inxi**  packages and post outputs of following commands.

```
sudo dmesg  | nc termbin.com 9999
journalctl -b -u NetworkManager --no-pager -l  | nc termbin.com 9999
NetworkManager --print-config
inxi -Nnxxz
lsmod | grep rtw
dkms status
mokutil --sb-state​
```

sribooshan, 2021-09-10:
@kelebek333 I installed the packages using my USB Ethernet and removed it while executing the commands you mentioned.
```
booshan@booshbuntu:~$ sudo dmesg  | nc termbin.com 9999
nc: getaddrinfo for host "termbin.com" port 9999: Temporary failure in name resolution
booshan@booshbuntu:~$ journalctl -b -u NetworkManager --no-pager -l  | nc termbin.com 9999
nc: getaddrinfo for host "termbin.com" port 9999: Temporary failure in name resolution
booshan@booshbuntu:~$ NetworkManager --print-config
# NetworkManager configuration: /etc/NetworkManager/NetworkManager.conf (lib: 10-dns-resolved.conf, 20-connectivity-ubuntu.conf, no-mac-addr-change.conf) (run: 10-globally-managed-devices.conf) (etc: default-wifi-powersave-on.conf)

[main]
# rc-manager=symlink
# auth-polkit=true
# dhcp=internal
dns=systemd-resolved
plugins=ifupdown,keyfile

[connectivity]
uri=http://connectivity-check.ubuntu.com/

[ifupdown]
managed=false

[logging]
# backend=journal
# audit=true

[device]
wifi.scan-rand-mac-address=no

[device-mac-addr-change-wifi]
match-device=driver:rtl8723bs,driver:rtl8189es,driver:r8188eu,driver:8188eu,driver:eagle_sdio,driver:wl
wifi.scan-rand-mac-address=no

[connection]
wifi.powersave=2

# no-auto-default file "/var/lib/NetworkManager/no-auto-default.state"
booshan@booshbuntu:~$ inxi -Nnxxz
Network:   Device-1: Realtek RTL8822BE 802.11a/b/g/n/ac WiFi adapter driver: N/A port: 5000 bus ID: 07:00.0 chip ID: 10ec:b822 
           Device-2: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: Lenovo driver: r8169 v: kernel port: 3000 
           bus ID: 08:00.0 chip ID: 10ec:8168 
           IF: enp8s0 state: down mac: <filter> 
booshan@booshbuntu:~$ lsmod | grep rtw
rtw_8822be             16384  0
rtw_8822b             225280  1 rtw_8822be
rtw_pci                24576  1 rtw_8822be
rtw_core              196608  2 rtw_8822b,rtw_pci
mac80211             1024000  2 rtw_core,rtw_pci
cfg80211              888832  2 rtw_core,mac80211
booshan@booshbuntu:~$ dkms status
8812au, 5.6.4.2_35491.20191025, 5.11.0-27-generic, x86_64: installed
nvidia, 460.91.03, 5.11.0-27-generic, x86_64: installed
booshan@booshbuntu:~$ mokutil --sb-state​
mokutil: unrecognized option '--sb-state​'
Usage:
  mokutil OPTIONS [ARGS...]

Options:
  --help				Show help
  --list-enrolled			List the enrolled keys
  --list-new				List the keys to be enrolled
  --list-delete				List the keys to be deleted
  --import <der file...>		Import keys
  --delete <der file...>		Delete specific keys
  --revoke-import			Revoke the import request
  --revoke-delete			Revoke the delete request
  --export				Export keys to files
  --password				Set MOK password
  --clear-password			Clear MOK password
  --disable-validation			Disable signature validation
  --enable-validation			Enable signature validation
  --sb-state				Show SecureBoot State
  --test-key <der file>			Test if the key is enrolled or not
  --reset				Reset MOK list
  --generate-hash[=password]		Generate the password hash
  --ignore-db				Ignore DB for validation
  --use-db				Use DB for validation
  --import-hash <hash>			Import a hash into MOK or MOKX
  --delete-hash <hash>			Delete a hash in MOK or MOKX
  --set-verbosity <true/false>		Set the verbosity bit for shim
  --pk					List the keys in PK
  --kek					List the keys in KEK
  --db					List the keys in db
  --dbx					List the keys in dbx
  --timeout <-1,0..0x7fff>		Set the timeout for MOK prompt

Supplimentary Options:
  --hash-file <hash file>		Use the specific password hash
  --root-pw				Use the root password
  --simple-hash				Use the old password hash method
  --mokx				Manipulate the MOK blacklist
```

I also executed the first two commands again after connecting the system to internet using my phone USB. It resulted in these outputs:
```
booshan@booshbuntu:~$ sudo dmesg  | nc termbin.com 9999
https://termbin.com/5mpdm
booshan@booshbuntu:~$ journalctl -b -u NetworkManager --no-pager -l  | nc termbin.com 9999
https://termbin.com/pw3p9
```

kelebek333, 2021-09-10:
> [ 3452.027234] rtw_8822be 0000:07:00.0: failed to read ASPM, ret=-5
> [ 3454.027352] rtw_8822be 0000:07:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
> [ 3454.027367] rtw_8822be 0000:07:00.0: mac power on failed
> [ 3454.027370] rtw_8822be 0000:07:00.0: failed to power on mac
> [ 3454.027372] rtw_8822be 0000:07:00.0: leave idle state failed
> [ 3454.027549] rtw_8822be 0000:07:00.0: failed to leave ips state
> [ 3454.027555] rtw_8822be 0000:07:00.0: failed to leave idle state
> [ 3454.027690] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[0]
> [ 3454.027777] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[1]
> [ 3454.027862] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[2]
> [ 3454.027949] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[3]
> [ 3454.028035] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[5]
> [ 3454.028122] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[6]
> [ 3454.028218] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[0]
> [ 3454.028305] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[1]
> [ 3454.028391] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[2]
> [ 3454.028478] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[3]
> [ 3454.028563] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[5]
> [ 3454.028650] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[6]
> [ 3457.930155] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[0]
> [ 3457.930257] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[1]
> [ 3457.930367] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[2]
> [ 3457.930455] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[3]
> [ 3457.930542] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[5]
> [ 3457.930631] rtw_8822be 0000:07:00.0: timed out to flush pci tx ring[6]

Try to run following commands.

```
sudo modprope -r rtw_8822be
sudo modprobe rtw_pci disable_aspm=Y
sudo modprobe rtw_8822be
```

If issue solved with above commands, create a configuration file with following command.

`/bin/bash -c "echo 'options rtw_pci disable_aspm=Y' | sudo tee /etc/modprobe.d/rtw_pci.conf"`

sribooshan, 2021-09-10:
@kelebek333 Wi-Fi started working when I restarted my system. Thank you very much for your help. If it does not work again sometime later, I will try this latest solution of yours.

> Try to run following commands.
>
> ```
> sudo modprope -r rtw_8822be
> sudo modprobe rtw_pci disable_aspm=Y
> sudo modprobe rtw_8822be
> ```
> If issue solved with above commands, create a configuration file [...]