lwfinger/rtw88 #68: Wifi adapter not found

kishor1010, 2021-09-21:
I have tried everything but laptop is not recognising wifi driver. I tried to install the drivers but could not. Please help me !!!
[image]

kishor1010, 2021-09-21:
[image]

kishor1010, 2021-09-21:
[image]

Lollixzc, 2021-09-21:
Have you read the readme? It seems that you don't have. Please at least say what model of card you have, but you don't need to open an issue if you cannot install the devices. Issues are made to report bugs on this repo, it seems you cannot install the driver so read the readme file first

kelebek333, 2021-09-21:
Wrong driver, use this. https://github.com/lwfinger/rtw89

lwfinger, 2021-09-21:
The last suggestion is correct. You need rtw89,. NOT rtw88.

closed 2021-09-21 by lwfinger