lwfinger/rtw88 #69: Random full lock-ups on HP 250 G3 / rtw_8821ce, on all recent kernel versions

mdartmann, 2021-10-05:
Hi,

when using your driver and connected to a wireless network, I get frequent freezes, to a point where not even a SysRQ is possible. 
I have experienced the issue since Linux 5.13, on both my self-compiled rc kernel as well as Archs' linux-zen package. 

So far, I have not been able to detect a pattern in the freezes, except for being caused by the WiFi driver as they only occur while connected to a wireless network and if I don't use an external network card.

I want to help you fix this. Please let me know what logs I can provide, I will work on trying to get a pattern to show. I will reply as soon as I can reproduce the crash and attach a syslog.

mdartmann, 2021-10-05:
I managed to reproduce the crash.
It occured as I had my system under heavy load and opened a few tabs in a browser.
Syslog:
```
Oct  5 12:43:38 finja kernel: rtw_8821ce 0000:02:00.0: timed out to flush queue 1
Oct  5 12:43:39 finja kernel: rtw_8821ce 0000:02:00.0: timed out to flush queue 1
Oct  5 12:43:40 finja kernel: rtw_8821ce 0000:02:00.0: timed out to flush queue 1
Oct  5 12:43:40 finja kernel: rtw_8821ce 0000:02:00.0: timed out to flush queue 1
Oct  5 12:43:40 finja kernel: rtw_8821ce 0000:02:00.0: timed out to flush queue 1
Oct  5 12:43:41 finja kernel: rtw_8821ce 0000:02:00.0: timed out to flush queue 1
Oct  5 12:43:41 finja kernel: rtw_8821ce 0000:02:00.0: timed out to flush queue 1
Oct  5 12:43:42 finja kernel: rtw_8821ce 0000:02:00.0: timed out to flush queue 1
Oct  5 12:43:42 finja kernel: rtw_8821ce 0000:02:00.0: timed out to flush queue 1
Oct  5 12:43:43 finja kernel: rtw_8821ce 0000:02:00.0: timed out to flush queue 1
Oct  5 12:43:44 finja kernel: rtw_8821ce 0000:02:00.0: timed out to flush queue 1
Oct  5 12:43:48 finja kernel: rtw_8821ce 0000:02:00.0: timed out to flush queue 1
``` 
... and then it freezes.
This is on 5.15-rc3. I am building rc4 as I'm writing this and will try to reproduce it again.

lwfinger, 2021-10-05:
Those "timed out to flush queue" messages are fairly common, but I am unaware that they have caused a crash. Was the caps-lock light flashing, or did the system just become unresponsive? If the latter, is there a CPU running full blast? A loud system fan would be the clue. Are you running the kernel version of the driver, or the one from this repo. The "rtw_8821ce" entry in the logged messages suggest that you are using this one. If so, why? This one is just copied from the kernel, but it may have mistakes.

Ideally, I would like to see whatever is logged at the time of the crash. If you are getting a fatal BUG, that gets difficult. One way would be to open a terminal and run 'sudo dmesg -tw". Leave that window open while you do web browsing, etc. The problem is sizing the terminal and browser such that both are visible at all times. When the crash happens, something should be output to the terminal. Photograph the screen messages and post that photo.

mdartmann, 2021-10-05:
The caps lock indicator is not flashing and the CPU is running at full blast (which isn't a lot, it's just an i3-7020u).

 I use the version straight from the kernel, this isn't journald but syslog-ng, that might explain why it looks different.

I will try running dmesg and get back to you. Thank you for the quick response!

dubhater, 2021-10-10:
I have the same problem: https://marc.info/?l=linux-wireless&m=162058892007833&w=4

hidjgr, 2021-10-10:
I have a similar problem, my computer freezes within a few minutes of being unplugged.
Here is the output of `journalctl -xe -p3 -b-1`
```
Oct 11 00:03:33 hdshp kernel: rtw_8821ce 0000:03:00.0: failed to send h2c command
```

mdartmann, 2021-10-10:
Update: I wasn't able to get any additional dmesg output. Let me know if there are an more verbose output modes I can enable somewhere.

NBoumakis, 2021-11-06:
I had the same problem with the default driver (rtw_8821ce) provided by the kernel 5.13.0-20. The laptop is an HP 15-da1018nv. Looked around and found a bug [report](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1927808) on the kernel bug tracker. Worked around it by disabling Power Management and setting Power Management default to Off.

tt-thoma, 2025-09-24:
Can confirm with latest release using dkms and linux mint zara:
```
thom@Gallium:~$ neofetch
             ...-:::::-...                 thom@Gallium 
          .-MMMMMMMMMMMMMMM-.              ------------ 
      .-MMMM`..-:::::::-..`MMMM-.          OS: Linux Mint 22.2 x86_64 
    .:MMMM.:MMMMMMMMMMMMMMM:.MMMM:.        Host: HP 250 G8 Notebook PC 
   -MMM-M---MMMMMMMMMMMMMMMMMMM.MMM-       Kernel: 6.8.0-84-generic 
 `:MMM:MM`  :MMMM:....::-...-MMMM:MMM:`    Uptime: 2 mins 
 :MMM:MMM`  :MM:`  ``    ``  `:MMM:MMM:    Packages: 3192 (dpkg), 39 (flatpak) 
.MMM.MMMM`  :MM.  -MM.  .MM-  `MMMM.MMM.   Shell: bash 5.2.21 
:MMM:MMMM`  :MM.  -MM-  .MM:  `MMMM-MMM:   Resolution: 1366x768 
:MMM:MMMM`  :MM.  -MM-  .MM:  `MMMM:MMM:   DE: Cinnamon 6.4.8 
:MMM:MMMM`  :MM.  -MM-  .MM:  `MMMM-MMM:   WM: Mutter (Muffin) 
.MMM.MMMM`  :MM:--:MM:--:MM:  `MMMM.MMM.   WM Theme: Mint-Y-Dark-Pink (Mint-Y) 
 :MMM:MMM-  `-MMMMMMMMMMMM-`  -MMM-MMM:    Theme: Mint-Y-Dark-Pink [GTK2/3] 
  :MMM:MMM:`                `:MMM:MMM:     Icons: Mint-Y-Pink [GTK2/3] 
   .MMM.MMMM:--------------:MMMM.MMM.      Terminal: gnome-terminal 
     '-MMMM.-MMMMMMMMMMMMMMM-.MMMM-'       CPU: Intel i3-1005G1 (4) @ 3.400GHz 
       '.-MMMM``--:::::--``MMMM-.'         GPU: Intel Iris Plus Graphics G1 
            '-MMMMMMMMMMMMM-'              Memory: 2692MiB / 7725MiB 
               ``-:::::-``

```
I modified one line of the adapter's config to prevent it from changing its power mode and remove one of the messages spammed in logs.
My system freezes at random (it can be on the login screen as well as in xorg itself, and maybe even in other ttys) with to my knowledge no trace in journalctl.
It becomes impossible to do ANYTHING and I have to hold the power button to turn it off.
This gets fixed as soon as I disable the driver.

dubhater, 2025-09-24:
@tt-thoma Please try:

```
options rtw_pci disable_aspm=1
options rtw88_pci disable_aspm=1
```

After you edit the configuration file make sure to shut down the laptop. For this parameter it's not enough to reboot it.

If that makes it stop freezing then attach the output of ``sudo dmidecode`` so we can enable that module parameter for your laptop permanently.

mikepurvis, 2026-02-12:
I'm experiencing symptoms like these with an 8822ce PCIe card in a Jetson Orin with 5.15.136-tegra.

When I blacklist the driver, everything is fine, but when I run this:
```
sudo modprobe rtw_pci disable_msi=1 disable_aspm=1
sudo modprobe rtw_8822ce
```

Within a minute or two, the whole system has hard-locked (no capslock light flashing).