lwfinger/rtw88 #77: [ RTL8821CE ] Spontaneous disconnections from WiFi

Armiixteryx, 2022-01-29:
Hello,

I am getting an issue with this WiFi card that is included in my laptop.

Sometimes I lost connection to internet, but system WiFi keeps connected to the router. It is not an issue in my ISP because my other devices (mobile, tablet, Smart TV, etc) keeps working with internet just fine.

When that happens, I check the system logs (`journalctl -f`), and I get this message repeatedly:

`wpa_supplicant[697]: wlp2s0: CTRL-EVENT-BEACON-LOSS`

and in other moments I get following log, showing that the connection gets disconnected and system reconnects it:

```
kernel: wlp2s0: deauthenticated from 58:xx:xx:xx:xx:21 (Reason: 6=CLASS2_FRAME_FROM_NONAUTH_STA)
rtw_8821ce 0000:02:00.0: sta 58:xx:xx:xx:xx:21 with macid 0 left
wpa_supplicant[697]: wlp2s0: CTRL-EVENT-DISCONNECTED bssid=58:xx:xx:xx:xx:21 reason=6
wpa_supplicant[697]: wlp2s0: SME: Trying to authenticate with 58:xx:xx:xx:xx:21 (SSID='xxxx' freq=2427 MHz)
kernel: wlp2s0: authenticate with 58:xx:xx:xx:xx:21
wpa_supplicant[697]: wlp2s0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
kernel: wlp2s0: send auth to 58:xx:xx:xx:xx:21 (try 1/3)
NetworkManager[488]: <info>  [1643497538.7599] device (wlp2s0): supplicant interface state: completed -> authenticating
kernel: wlp2s0: authenticated
wpa_supplicant[697]: wlp2s0: Trying to associate with 58:xx:xx:xx:xx:21 (SSID='xxxx' freq=2427 MHz)
NetworkManager[488]: <info>  [1643497538.7679] device (wlp2s0): supplicant interface state: authenticating -> associating
kernel: wlp2s0: associate with 58:xx:xx:xx:xx:21 (try 1/3)
kernel: wlp2s0: RX AssocResp from 58:xx:xx:xx:xx:21 (capab=0x431 status=0 aid=2)
kernel: rtw_8821ce 0000:02:00.0: sta 58:xx:xx:xx:xx:21 joined with macid 0
kernel: wlp2s0: associated
wpa_supplicant[697]: wlp2s0: Associated with 58:xx:xx:xx:xx:21
wpa_supplicant[697]: wlp2s0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
NetworkManager[488]: <info>  [1643497538.7799] device (wlp2s0): supplicant interface state: associating -> 4way_handshake
wpa_supplicant[697]: wlp2s0: WPA: Key negotiation completed with 58:xx:xx:xx:xx:21 [PTK=CCMP GTK=CCMP]
wpa_supplicant[697]: wlp2s0: CTRL-EVENT-CONNECTED - Connection to 58:xx:xx:xx:xx:21 completed [id=0 id_str=]
NetworkManager[488]: <info>  [1643497538.8264] device (wlp2s0): supplicant interface state: 4way_handshake -> completed
kernel: rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
systemd-resolved[470]: Using degraded feature set UDP instead of UDP+EDNS0 for DNS server 192.168.0.1.
systemd-resolved[470]: Using degraded feature set UDP instead of UDP+EDNS0 for DNS server 192.168.0.1.
```

lwfinger, 2022-01-30:
kernel: wlp2s0: deauthenticated from 58:xx:xx:xx:xx:21 (Reason: 6=CLASS2_FRAME_FROM_NONAUTH_STA)
rtw_8821ce 0000:02:00.0: sta 58:xx:xx:xx:xx:21 with macid 0 left
wpa_supplicant[697]: wlp2s0: CTRL-EVENT-DISCONNECTED bssid=58:xx:xx:xx:xx:21 reason=6

That CLASS2_FRAME_FROM_NONAUTH_STA reason for disconnecting is troubling. It makes me wonder if your AP is doing something that rtw88 is not expecting.

It is OK to obfuscate the MAC listings in stuff you post, but please leave the first 3 octants in the clear. They indicate the vendor of your equipment. The last 3 indicate the serial number of the device, and it is OK to replace then with x's, but the vendor of the AP is valuable information.

I will check the current status of rtw88 in the kernel, and update this repo accordingly.

lwfinger, 2022-01-30:
As I said, the disconnect reason is troubling; however, most loss of beacon problems are due to AP and STA not handling sleep mode correctly. This is not the sleep that happens when you close the cover on your laptop, but a means of reducing power usage when wireless traffic is low.

To check this, we will disable that sleep mode. As root, create a file /etc/modprobe.d/50-rtw88.conf containing the following 2 lines:

options rtw_core disable_lps_deep=y
options rtw88_core disable_lps_deep=y

The first line handles drivers from this repo, and the second for rtw88 drivers built into the kernel.

Armiixteryx, 2022-01-31:
> kernel: wlp2s0: deauthenticated from 58:xx:xx:xx:xx:21 (Reason: 6=CLASS2_FRAME_FROM_NONAUTH_STA) rtw_8821ce 0000:02:00.0: sta 58:xx:xx:xx:xx:21 with macid 0 left wpa_supplicant[697]: wlp2s0: [...]

Here are the first octects: `58:d9:d5`

Armiixteryx, 2022-01-31:
> As I said, the disconnect reason is troubling; however, most loss of beacon problems are due to AP and STA not handling sleep mode correctly. This is not the sleep that happens when you close the [...]

Thank you, I have created the file. I will test and update this issue on results.

lwfinger, 2022-01-31:
Your AP vendor is Tenda Technology Co.,Ltd.Dongguan Branch.

lwfinger, 2022-04-22:
I just pushed an update to bring the code up to the kernel 5.18 value. Perhaps the changes will help your issue.

lucassith, 2022-05-16:
@lwfinger Is that something related to the AP Vendor? I have Tenda WiFi mesh, even though I don't experience disconnection issues I do have dmesg spammed with:

`rtw_8822ce 0000:03:00.0: failed to send h2c command`

followed by

```
[  169.955914] ------------[ cut here ]------------
[  169.955917] firmware crash, start reset and recover
[  169.955940] WARNING: CPU: 3 PID: 10 at /var/lib/dkms/rtw88/r131.ce97958/build/main.c:598 rtw_fw_recovery_work+0x17b/0x280 [rtw_core]
[  169.955952] Modules linked in: cdc_acm ccm rfcomm snd_seq_dummy snd_hrtimer snd_seq xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c br_netfilter bridge stp llc overlay hid_logitech_hidpp hid_logitech_dj cmac algif_hash algif_skcipher snd_usb_audio af_alg snd_usbmidi_lib snd_rawmidi usbhid snd_seq_device bnep uvcvideo videobuf2_vmalloc btusb videobuf2_memops videobuf2_v4l2 btrtl btbcm videobuf2_common btintel videodev btmtk uas qrtr usb_storage mc bluetooth ecdh_generic intel_rapl_msr intel_rapl_common amdgpu snd_sof_amd_renoir uinput joydev snd_acp_config mousedev rtw_8822ce(OE) nvidia_uvm(POE) snd_sof_amd_acp edac_mce_amd asus_nb_wmi rtw_8822c(OE) asus_wmi rtw_pci(OE) kvm_amd sparse_keymap snd_sof_pci hid_multitouch platform_profile wmi_bmof snd_sof snd_hda_codec_realtek rtw_core(OE) snd_hda_codec_generic snd_soc_acpi snd_hda_codec_hdmi kvm mac80211 snd_hda_intel vfat
[  169.955983]  ledtrig_audio fat snd_intel_dspcfg irqbypass snd_soc_core snd_intel_sdw_acpi libarc4 crct10dif_pclmul snd_hda_codec crc32_pclmul ghash_clmulni_intel nvidia_drm(POE) snd_compress aesni_intel snd_hda_core ac97_bus cfg80211 snd_hwdep crypto_simd snd_pcm_dmaengine nvidia_modeset(POE) cryptd snd_pci_acp5x rapl snd_pcm gpu_sched sp5100_tco ucsi_acpi tpm_crb snd_rn_pci_acp3x drm_ttm_helper snd_timer typec_ucsi tpm_tis snd pcspkr typec nvidia(POE) tpm_tis_core r8168(OE) k10temp i2c_piix4 ccp snd_pci_acp3x ttm soundcore i2c_nvidia_gpu rfkill wmi roles video i2c_hid_acpi tpm i2c_hid asus_wireless mac_hid pinctrl_amd rng_core acpi_cpufreq squashfs loop fuse crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 serio_raw atkbd libps2 i8042 xhci_pci crc32c_intel xhci_pci_renesas serio
[  169.956017] CPU: 3 PID: 10 Comm: kworker/u32:1 Tainted: P        W  OE     5.17.6-1-MANJARO #1 1e1d23d0a05f307a20d237ae367072b447c14fc7
[  169.956019] Hardware name: ASUSTeK COMPUTER INC. ASUS TUF Gaming A17 FA706IU_FA706IU/FA706IU, BIOS FA706IU.316 03/12/2021
[  169.956021] Workqueue: phy0 rtw_fw_recovery_work [rtw_core]
[  169.956028] RIP: 0010:rtw_fw_recovery_work+0x17b/0x280 [rtw_core]
[  169.956037] Code: c7 83 e0 eb ff ff 00 00 00 00 48 c7 83 d8 eb ff ff 00 00 00 00 48 8b 40 70 e8 b1 c8 01 f7 48 c7 c7 68 79 a0 c4 e8 92 4d bd f6 <0f> 0b e8 4e 7a 14 f6 48 8b bb b0 b5 ff ff 4c 89 e1 31 f6 48 c7 c2
[  169.956038] RSP: 0018:ffffaada00137e68 EFLAGS: 00010246
[  169.956040] RAX: 0000000000000000 RBX: ffff92f373a06b30 RCX: 0000000000000000
[  169.956041] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[  169.956041] RBP: ffffaada164eb010 R08: 0000000000000000 R09: 0000000000000000
[  169.956042] R10: 0000000000000000 R11: 0000000000000000 R12: ffff92f373a020e0
[  169.956043] R13: ffff92f373a06a40 R14: 0000000000003000 R15: ffff92f34917c485
[  169.956044] FS:  0000000000000000(0000) GS:ffff92f65f6c0000(0000) knlGS:0000000000000000
[  169.956045] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  169.956046] CR2: 00007f332f8ed000 CR3: 000000014ce10000 CR4: 0000000000350ee0
[  169.956047] Call Trace:
[  169.956050]  <TASK>
[  169.956051]  process_one_work+0x1e5/0x3b0
[  169.956055]  ? rescuer_thread+0x3a0/0x3a0
[  169.956056]  worker_thread+0x50/0x3a0
[  169.956058]  ? rescuer_thread+0x3a0/0x3a0
[  169.956059]  kthread+0xd8/0x100
[  169.956061]  ? kthread_complete_and_exit+0x20/0x20
[  169.956062]  ret_from_fork+0x22/0x30
[  169.956065]  </TASK>
[  169.956066] ---[ end trace 0000000000000000 ]---
```

pantr19, 2022-06-12:
Hello,
I would like to add up to this issue.
My device HP Laptop 15s-du2xxx runs RTL8821CE chip-set and i face the same issue.
Currently on 5.18.3-arch1-1 kernel on Arch Linux the driver has become fairly usable by adding 'options rtw88_core disable_lps_deep=y' as suggested by you but the random disconnects do occur.
The gnome Panel WI-FI indicator shows sudden drop in strength and then its back up to full strength [behavior not observed on tomaspinho-rtl8821ce].
Also the dmesg gets flooded, attaching output below. 

[  +0.001058] rtw_8821ce 0000:02:00.0: enabling device (0000 -> 0003)
[  +0.007029] rtw_8821ce 0000:02:00.0: Firmware version 24.11.0, H2C version 12
[  +0.035947] rtw_8821ce 0000:02:00.0 wlo1: renamed from wlan0
[  +0.426371] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.529518] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.506201] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.506297] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.556258] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[Jun12 14:41] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.456592] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.140079] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.403404] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.133360] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.406716] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.516761] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.516804] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.136748] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.406668] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.133366] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.410125] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.133382] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.273571] rtw_8821ce 0000:02:00.0: timed out to flush pci tx ring[1]
[  +0.136431] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.000267] rtw_8821ce 0000:02:00.0: timed out to flush pci tx ring[1]
[ +37.867532] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.515781] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.501118] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.509399] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.479224] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.509710] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.529863] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.506714] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.535295] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.526890] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.543171] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.536429] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.513243] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.506485] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.557922] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.516649] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.517681] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.513313] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.512547] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.519992] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.514768] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.503317] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.519371] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.573456] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.520013] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.413680] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.506546] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.466348] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.503233] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.579394] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.520291] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.525578] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.371606] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.506653] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.395046] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.526648] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.503176] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.368833] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[Jun12 15:27] rtw_8821ce 0000:02:00.0: timed out to flush queue 1
[  +0.513372] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.524130] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.574355] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.506641] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[... cut ...]  failed to get tx report from firmware
[  +0.519709] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.462917] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.506890] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.581541] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.526563] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware
[  +0.390293] rtw_8821ce 0000:02:00.0: failed to get tx report from firmware

bigangryrobot, 2022-07-31:
related https://github.com/sebanc/brunch/issues/855 I've also got a bunch of boxes in a network doing the same

bigangryrobot, 2022-07-31:
possibly related stack
```
22.06.21 21:30	wpa_supplicant	wlp1s0: CTRL-EVENT-BEACON-LOSS
22.06.21 21:30	kernel	------------[ cut here ]------------
22.06.21 21:30	kernel	purge skb(s) not reported by firmware
22.06.21 21:30	kernel	WARNING: CPU: 0 PID: 132630 at drivers/net/wireless/realtek/rtw88/tx.c:161 rtw_tx_report_purge_timer+0x20/0x50 [rtw88_core]
22.06.21 21:30	kernel	Modules linked in: ccm rfcomm cmac algif_hash algif_skcipher af_alg bnep btusb btrtl btbcm btintel bluetooth uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev ecdh_generic ecc mc squashfs loop snd_hda_codec_realtek snd_hda_codec_generic snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg soundwire_intel soundwire_generic_allocation soundwire_cadence snd_hda_codec rtw88_8822ce rtw88_8822c rtw88_pci snd_hda_core amdgpu rtw88_core joydev snd_hwdep mousedev soundwire_bus mac80211 vfat snd_soc_core fat edac_mce_amd kvm_amd snd_compress gpu_sched ac97_bus i2c_algo_bit snd_pcm_dmaengine huawei_wmi kvm hid_multitouch ledtrig_audio wmi_bmof cfg80211 ttm sparse_keymap irqbypass snd_pcm crct10dif_pclmul crc32_pclmul snd_timer drm_kms_helper ghash_clmulni_intel snd aesni_intel cec syscopyarea crypto_simd sysfillrect cryptd sysimgblt rfkill pcspkr glue_helper snd_rn_pci_acp3x sp5100_tco ccp libarc4 fb_sys_fops soundcore snd_pci_acp3x rapl i2c_piix4 wmi
22.06.21 21:30	kernel	 k10temp tpm_crb tpm_tis video tpm_tis_core tpm i2c_hid mac_hid rng_core pinctrl_amd acpi_cpufreq drm uinput sg fuse crypto_user agpgart ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 crc32c_intel serio_raw xhci_pci
22.06.21 21:30	kernel	CPU: 0 PID: 132630 Comm: ksystemlog Tainted: G        W         5.10.42-1-MANJARO #1
22.06.21 21:30	kernel	Hardware name: HUAWEI KLVL-WXX9/KLVL-WXX9-PCB, BIOS 1.10 12/31/2020
22.06.21 21:30	kernel	RIP: 0010:rtw_tx_report_purge_timer+0x20/0x50 [rtw88_core]
22.06.21 21:30	kernel	Code: 0f 1f 84 00 00 00 00 00 66 90 0f 1f 44 00 00 8b 47 f0 85 c0 75 01 c3 41 54 55 53 48 89 fb 48 c7 c7 18 7c e4 c0 e8 fd ff 9c e5 <0f> 0b 4c 8d 63 d8 4c 89 e7 e8 62 e9 a2 e5 48 8d 7b e0 48 89 c5 e8
22.06.21 21:30	kernel	RSP: 0000:ffffa5dec0003ec8 EFLAGS: 00010282
22.06.21 21:30	kernel	RAX: 0000000000000000 RBX: ffff8dec95086308 RCX: 0000000000000027
22.06.21 21:30	kernel	RDX: ffff8def7f418bb8 RSI: 0000000000000001 RDI: ffff8def7f418bb0
22.06.21 21:30	kernel	RBP: ffff8dec95086308 R08: 0000000000000000 R09: ffffa5dec0003d00
22.06.21 21:30	kernel	R10: ffffa5dec0003cf8 R11: ffffffffa74c9948 R12: ffffffffc0e32480
22.06.21 21:30	kernel	R13: dead000000000122 R14: ffffffffc0e32480 R15: ffff8def7f41cf80
22.06.21 21:30	kernel	FS:  00007f1329ff0880(0000) GS:ffff8def7f400000(0000) knlGS:0000000000000000
22.06.21 21:30	kernel	CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
22.06.21 21:30	kernel	CR2: 00002e5304b5d468 CR3: 000000011a5b0000 CR4: 0000000000350ef0
22.06.21 21:30	kernel	Call Trace:
22.06.21 21:30	kernel	 <IRQ>
22.06.21 21:30	kernel	 ? rtw_tx_write_data_h2c_get+0x90/0x90 [rtw88_core]
22.06.21 21:30	kernel	 call_timer_fn+0x29/0x130
22.06.21 21:30	kernel	 __run_timers.part.0+0x1cf/0x260
22.06.21 21:30	kernel	 ? rebalance_domains+0x108/0x3f0
22.06.21 21:30	kernel	 run_timer_softirq+0x26/0x50
22.06.21 21:30	kernel	 __do_softirq+0xcb/0x2ba
22.06.21 21:30	kernel	 asm_call_irq_on_stack+0x12/0x20
22.06.21 21:30	kernel	 </IRQ>
22.06.21 21:30	kernel	 do_softirq_own_stack+0x37/0x40
22.06.21 21:30	kernel	 irq_exit_rcu+0x9c/0xd0
22.06.21 21:30	kernel	 sysvec_apic_timer_interrupt+0x36/0x80
22.06.21 21:30	kernel	 ? asm_sysvec_apic_timer_interrupt+0xa/0x20
22.06.21 21:30	kernel	 asm_sysvec_apic_timer_interrupt+0x12/0x20
22.06.21 21:30	kernel	RIP: 0033:0x7f132e420511
22.06.21 21:30	kernel	Code: 66 0f 1f 84 00 00 00 00 00 c5 fe 7f 40 80 48 83 e8 80 c5 fe 7f 80 20 ff ff ff c5 fe 7f 80 40 ff ff ff c5 fe 7f 80 60 ff ff ff <48> 39 d0 76 da 48 83 c6 80 48 83 e6 80 48 8d 3c 31 48 89 d6 48 29
22.06.21 21:30	kernel	RSP: 002b:00007ffe67b38158 EFLAGS: 00000207
22.06.21 21:30	kernel	RAX: 00007f12bbb37714 RBX: 0000000000000450 RCX: 00007f12bbb36594
22.06.21 21:30	kernel	RDX: 00007f12bbb38644 RSI: 0000000000002130 RDI: 00007f12bbb36514
22.06.21 21:30	kernel	RBP: 00007f12bbb36514 R08: 00000000000021c0 R09: 0000000000ff0000
22.06.21 21:30	kernel	R10: 000000000000ff00 R11: 000001007f000100 R12: 00000000000021c0
22.06.21 21:30	kernel	R13: 000000000000023f R14: 00000000ffffffff R15: 000000000000084c
22.06.21 21:30	kernel	---[ end trace adbe2be71686001e ]---
22.06.21 21:30	kernel	rtw_8822ce 0000:01:00.0: timed out to flush queue 1
22.06.21 21:30	kernel	rtw_8822ce 0000:01:00.0: timed out to flush queue 2
22.06.21 21:30	kernel	rtw_8822ce 0000:01:00.0: sta 30:b1:xx:xx:xx:xx with macid 0 left
22.06.21 21:30	wpa_supplicant	wlp1s0: CTRL-EVENT-DISCONNECTED bssid=30:b1:xx:xx:xx:xx reason=4 locally_generated=1
22.06.21 21:30	wpa_supplicant	wlp1s0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
22.06.21 21:30	NetworkManager	<info>  [1624390201.0714] device (wlp1s0): supplicant interface state: completed -> disconnected
22.06.21 21:30	wpa_supplicant	wlp1s0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
22.06.21 21:30	NetworkManager	<info>  [1624390201.3536] device (wlp1s0): supplicant interface state: disconnected -> scanning
22.06.21 21:30	wpa_supplicant	wlp1s0: SME: Trying to authenticate with 30:b2:xx:xx:xx:xx (SSID='mySSID' freq=5180 MHz)
22.06.21 21:30	kernel	wlp1s0: authenticate with 30:b2:xx:xx:xx:xx
[... cut ...]
22.06.21 21:30	systemd	NetworkManager-dispatcher.service: Deactivated successfully.
22.06.21 21:30	audit	SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
22.06.21 21:30	kernel	audit: type=1131 audit(1624390212.660:501): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
```

bigangryrobot, 2022-07-31:
reviewing your pr you mention above https://github.com/lwfinger/rtw88/commit/f7e18c5d5cd43bac548674ea9fc277b307ccdfbd i think it will address this issue. I will find a box to run this on and report back

bigangryrobot, 2022-07-31:
also possibly related https://github.com/lwfinger/rtw88/issues/82

bigangryrobot, 2022-07-31:
and https://github.com/lwfinger/rtw88/issues/69

bigangryrobot, 2022-07-31:
also related https://github.com/lwfinger/rtw88/issues/61 and https://github.com/lwfinger/rtw88/issues/72

lwfinger, 2022-08-01:
I  have said it before, and I will likely have to say it again that I did not write this driver and I do not have any knowledge of the internals of the chip other than what I have learned from reading the code. If there is a build issue with a certain kernel, that is something I can usually handle.

You need to take the issue to the authors of the code as described in the final paragraph of README.md. When you submit the E-mail, be absolutely that it is plain text. Any HTML mail will be rejected by the server.

lucassith, 2022-08-01:
Unfortunately, the driver creators are lazy/unresponsive.

I wrote an email explaining my bug from another issue but so far have not received any reply. Such is unfortunately the charm of closed software that has problems.

For me, none of the driver issues were solved so I bought a WIFI transmitter that has open software and disabled the RTL drivers in the kernel. This is how my adventure with this device ended.

lwfinger, 2022-08-01:
The author is not lazy, but he has many responsibilities. The software is no more closed than code for Intel or any other devices. The intellectual property in the implementation of the very complicated procedures of wifi5 or wifi6 is very valuable, thus the internals of the devices and the associated firmware are not made public.

closed 2022-08-01 by lwfinger

bigangryrobot, 2022-08-01:
this is a very visible bug showing in a large number of cases and needs addressed somehow. I will try to contact the authors as suggested. If you are going to close this bug then close the associated ones too so we can at-least consolidate efforts.

ObserverOfTime, 2022-08-02:
This bug has been present in both this driver and the rtl8821ce one for years.
The authors are surely aware of it but they don't seem to be doing anything to fix it.

dubhater, 2022-10-02:
I have not had any beacon loss for three days now since I made this change to the driver:
```diff
diff --git a/phy.c b/phy.c
index 6d8f006..d4f9d28 100644
--- a/phy.c
+++ b/phy.c
@@ -766,8 +766,8 @@ void rtw_phy_dynamic_mechanism(struct rtw_dev *rtwdev)
 {
 	/* for further calculation */
 	rtw_phy_statistics(rtwdev);
-	rtw_phy_dig(rtwdev);
-	rtw_phy_cck_pd(rtwdev);
+// 	rtw_phy_dig(rtwdev);
+// 	rtw_phy_cck_pd(rtwdev);
 	rtw_phy_ra_track(rtwdev);
 	rtw_phy_tx_path_diversity(rtwdev);
 	rtw_phy_cfo_track(rtwdev);
```
Also no disconnections and no delays in the browser.

So far I haven't figured out why this helps. The two functions seem to do the same things that the old driver does, and I never had problems with that one. It's really strange.

lwfinger, 2022-10-02:
The first of those two lines come from the original submission of the driver, thus the record has no information on why it was selected.

The commit message associated with the second line was as follows:
    rtw88: add dynamic rrsr configuration
    
    Register rrsr determines the response rate we send.
    In field tests, using rate higher than current tx rate could lead
    to difficulty for the receiving end to receive management/control
    frames. Calculate current modulation level by tx rate then cross out
    rate higher than those.

Adding that change to the repo could be dangerous. The code in phy.c is used by ALL the drivers in rtw88, and you have tested only on 8821ce. In addition, I have tested that device on my computer, and did not have a single disconnect in 4 days. If you want to have that change in your repo, use git to add and commit it locally. That way, any upstream changes will not overwrite it.

dubhater, 2022-10-03:
I was just hoping other people affected by this problem can test this "fix". Maybe it only helps on my system.

By the way, @lwfinger what RFE type is your device? What about everyone who's affected by this problem? Mine is RFE 2.

(Also, you looked at the wrong commit in the blame, but it doesn't matter anyway.)

lwfinger, 2022-10-03:
I think mine is RFE 1, but I could not find it in my logs.

The blame I looked at was from the mainline head. I think I got it right.

mrfixit2001, 2023-10-05:
@lwfinger 

FWIW - I can confirm commenting out those lines in rtw_phy_dynamic_mechanism appears to resolve the same reported issue here but with the rtl8821cs. 

I'm testing with kernel 6.5. I also tried using this repo's code as well as tried the latest updates in rtw88 from v6.6-rc4 upstream. The problem persisted among all driver variations. I even went as far as testing by patching the kernel to basically completely ignore beacon loss - which made the disconnections less often but they were still happening consistently. So far this patch is the only thing that seems to have completely resolved the issue for me.

If you have any contacts in the wireless-next dev team, would you please alert them of this issue? I'm sure removing these lines isn't the "right" fix, but this hack DOES work for at least a few cases. Happy to do testing for them if they reach out.

lwfinger, 2023-10-05:
Please be specific as to the lines you commented out. I do have contacts at Realtek, but I need specifics.

mrfixit2001, 2023-10-05:
> Please be specific as to the lines you commented out. I do have contacts at Realtek, but I need specifics.

I was referring to the listed patch above:
https://github.com/lwfinger/rtw88/issues/77#issuecomment-1264617163

With these two lines commented out:
  rtw_phy_dig(rtwdev);
  rtw_phy_cck_pd(rtwdev);

lwfinger, 2023-10-05:
Thank you. I could make those lines go away for any RTW8821C device (U, S, or E). I will report it to Realtek.

sbwml, 2023-10-09:
RTL8812BU also has the problem of WiFi disconnection (Client). I use rtw88 from linux-6.5 on OpenWrt 23.05.0-rc4.

```
[19397.557017] rtw_8822bu 1-1:1.0: failed to get tx report from firmware
```

I'm trying patch [77#issuecomment-1264617163](https://github.com/lwfinger/rtw88/issues/77#issuecomment-1264617163) and it's been using it for 3 hours and the WiFi disconnected issue doesn't seem to be happening again.

----------

In addition to this, I received another kernel panic, but it didn't affect WiFi connectivity.

```
[  262.264815] usb 1-1: new high-speed USB device number 3 using xhci-hcd
[  262.459058] rtw_8822bu 1-1:1.0: Firmware version 27.2.0, H2C version 13
[  410.332966] ------------[ cut here ]------------
[  410.332983] write RF mode table fail
[  410.333022] WARNING: CPU: 3 PID: 26242 at backports-6.5/drivers/net/wireless/realtek/rtw88/rtw8822b.c:823 0xffff800001a920b0
[  410.333047] Modules linked in: rtw88_8822cu rtw88_8822ce rtw88_8822c rtw88_8822bu rtw88_8822be rtw88_8822b rtw88_8821cu rtw88_8821ce rtw88_8821c rtw88_8723du rtw88_8723de rtw88_8723d rtw88_usb rtw88_pci rtw88_core mac80211 cfg80211 compat oaf pppoe ppp_async l2tp_ppp wireguard rockchipdrm rndis_host pppox ppp_generic nft_fib_inet nf_flow_table_inet lz4 libchacha20poly1305 ipt_REJECT dw_hdmi_cec dw_hdmi drm_dma_helper chacha_neon cec cdc_ether zstd xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbnet usblp ts_fsm ts_bm tls tcp_bbr slhc sch_cake rc_core r8152 r8125 poly1305_neon phy_rockchip_inno_hdmi panfrost nft_tproxy nft_socket nft_reject_ipv6
[  410.333244]  nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_fullcone nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_compat nft_chain_nat nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan lzo_rle lzo lz4hc_compress lz4_decompress lz4_compress libcurve25519_generic libchacha iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables gpu_sched dw_mipi_dsi drm_shmem_helper drm_kms_helper drm_display_helper crc_ccitt br_netfilter asn1_decoder ntfs3 sch_teql sch_sfq sch_multiq sch_gred sch_fq sch_dsmark sch_codel
[  410.333400]  em_text em_nbyte em_meta em_cmp act_skbmod act_simple act_pedit act_csum cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs pwm_fan cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb dummy sit l2tp_netlink l2tp_core ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 xsk_diag netlink_diag tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep
[  410.333573]  snd_compress snd_pcm snd_timer snd soundcore xfrm_user xfrm_ipcomp af_key xfrm_algo nls_utf8 nls_cp950 nls_cp936 nls_cp437 zram zsmalloc vxlan udp_tunnel ip6_udp_tunnel shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha1_generic seqiv drbg md5 kpp hmac echainiv des_generic libdes deflate cmac cbc authencesn authenc crypto_acompress sysimgblt sysfillrect syscopyarea fb_sys_fops cfbimgblt cfbfillrect cfbcopyarea fb font drm drm_panel_orientation_quirks fsl_mph_dr_of ehci_fsl gpio_button_hotplug xfs vfat fat exfat btrfs xor zstd_decompress zstd_compress zstd_common xor_neon raid6_pq lzo_decompress lzo_compress
[  410.333718] CPU: 3 PID: 26242 Comm: iw Not tainted 6.1.56 #0
[  410.333725] Hardware name: FriendlyElec NanoPi R4S (DT)
[  410.333729] pstate: 40400009 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  410.333735] pc : 0xffff800001a920b0
[  410.333744] lr : 0xffff800001a920b0
[  410.333748] sp : ffff800013ddb790
[  410.333751] x29: ffff800013ddb790 x28: ffff000008b53900 x27: ffff800013ddbdc8
[  410.333760] x26: 0000000000000003 x25: 0000000000000001 x24: ffff800001a94268
[  410.333768] x23: ffff800001a94010 x22: 0000000000000003 x21: 0000000000000000
[  410.333776] x20: 00000000000aeaea x19: ffff00004f712080 x18: fffffffffff8a260
[  410.333784] x17: 0000000000000000 x16: 0000000000000000 x15: fffffffffff8a288
[  410.333792] x14: ffff8000094cb130 x13: ffff8000094cb188 x12: 0000000000000711
[  410.333799] x11: 000000000000025b x10: ffff80000962b188 x9 : ffff8000094cb188
[... cut ...]
[  417.372589] phy0-sta0: authenticate with 6a:16:51:0b:03:e3
[  418.571077] phy0-sta0: send auth to 6a:16:51:0b:03:e3 (try 1/3)
[  418.572081] phy0-sta0: authenticated
[  418.580644] phy0-sta0: associate with 6a:16:51:0b:03:e3 (try 1/3)
[  418.605694] phy0-sta0: RX AssocResp from 6a:16:51:0b:03:e3 (capab=0x11 status=0 aid=29)
[  418.621763] phy0-sta0: associated
[  418.712980] phy0-sta0: Limiting TX power to 23 (23 - 0) dBm as advertised by 6a:16:51:0b:03:e3
[  418.772957] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-sta0: link becomes ready
```

dubhater, 2024-02-23:
Hello again!

If someone still cares about this bug, I believe this patch fixes it: 
[0001-wifi-rtw88-8821c-Fix-beacon-loss-and-disconnect.txt](https://github.com/lwfinger/rtw88/files/14390629/0001-wifi-rtw88-8821c-Fix-beacon-loss-and-disconnect.txt)

It's a real fix this time, not just papering over the problem.

lwfinger, 2024-02-24:
That fix will apply to not only RTW8821CE, but also to RTW8821CU and RTW8821CS. It is definitely needed; however, it breaks RTW8821BU and probably other chips in the RTW8821XY family. The message is
rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -19

I have an idea on how to fix the problem and will post a revissed patch.

dubhater, 2024-02-25:
```
rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -19
```
That comes from another patch I posted here: https://github.com/morrownr/8821cu-20210916/issues/115#issuecomment-1954105239

You don't need to worry about it, though. I already revised the patch to get rid of that message, I just didn't think to post the new version.

everschen, 2024-03-10:
> Hello again!
>
> If someone still cares about this bug, I believe this patch fixes it: [...]

在openwrt上添加了这个patch，并且看到build的文件确实也修改了，编译生成了固件，可是好像还是存在这个问题。
还有什么其他的调试方式吗？

dubhater, 2024-03-10:
We need to know the RFE type to know if that patch has any effect for your card:
```diff
diff --git a/drivers/net/wireless/realtek/rtw88/rtw8821c.c b/drivers/net/wireless/realtek/rtw88/rtw8821c.c
index fe5d8e188350..788735f5b234 100644
--- a/drivers/net/wireless/realtek/rtw88/rtw8821c.c
+++ b/drivers/net/wireless/realtek/rtw88/rtw8821c.c
@@ -54,6 +54,8 @@ static int rtw8821c_read_efuse(struct rtw_dev *rtwdev, u8 *log_map)
 
 	map = (struct rtw8821c_efuse *)log_map;
 
+	pr_warn("%s: RFE option from efuse: %#x\n", __func__, map->rfe_option);
+
 	efuse->rfe_option = map->rfe_option & 0x1f;
 	efuse->rf_board_option = map->rf_board_option;
 	efuse->crystal_cap = map->xtal_k;
```

everschen, 2024-03-10:
will try and let you know the result, the log will be appeared in dmesg output?

everschen, 2024-03-10:
[dmesg.txt](https://github.com/lwfinger/rtw88/files/14550151/dmesg.txt)
I don't see the message? where should I go to find it? should i enable debug anywhere?

dubhater, 2024-03-10:
Yes, it should appear in dmesg. Maybe change pr_warn for pr_err or raise the kernel's console debug level. I don't remember the option for that.

But if your problem is with „error beacon valid” and „failed to download drv rsvd page”, I don't think my patch will help with that.

everschen, 2024-03-10:
do you know which patch can fix this issue?

---- Replied Message ----
| From | ***@***.***> |
| Date | 03/10/2024 23:36 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [lwfinger/rtw88] [ RTL8821CE ] Spontaneous disconnections from WiFi (Issue #77) |

Yes, it should appear in dmesg. Maybe change pr_warn for pr_err or raise the kernel's console debug level. I don't remember the option for that.

But if your problem is with „error beacon valid” and „failed to download drv rsvd page”, I don't think my patch will help with that.

dubhater, 2024-03-11:
I think there is no patch for that yet.

everschen, 2024-03-11:
8821cu always disconnect， no patch?

---- Replied Message ----
| From | ***@***.***> |
| Date | 03/11/2024 20:25 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [lwfinger/rtw88] [ RTL8821CE ] Spontaneous disconnections from WiFi (Issue #77) |

I think there is no patch for that yet.

lwfinger, 2024-03-11:
There are a number of patches that improve the stability of 8821c, and they are included in wireless-next, and in rtw88. They improved the performance of my RTW8821CU, which does not disconnect. Have you done a 'git pull' and built the latest version. What kernel are you running ('uname -r')? What device do you have? What does 'dmesg' say? Your bare mention that it does not work does no good without further documentation.

everschen, 2024-03-11:
i alreay provided dmesg if you look at the comment above.
i believe it included all the infomation you wanted.

---- Replied Message ----
| From | ***@***.***> |
| Date | 03/12/2024 01:17 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [lwfinger/rtw88] [ RTL8821CE ] Spontaneous disconnections from WiFi (Issue #77) |

There are a number of patches that improve the stability of 8821c, and they are included in wireless-next, and in rtw88. They improved the performance of my RTW8821CU, which does not disconnect. Have you done a 'git pull' and built the latest version. What kernel are you running ('uname -r')? What device do you have? What does 'dmesg' say? Your bare mention that it does not work does no good without further documentation.

dubhater, 2025-03-08:
@cculianu That's probably https://github.com/lwfinger/rtw88/issues/205, fixed in kernel 6.14. Until you get that kernel, you can install the driver from this repository.

cculianu, 2025-03-08:
Oh man. The last time I installed extra-kernel drivers was back in like 2005 on Linux 2.x series. Is that trivially easier to do nowadays? Back then you basically had to patch your kernel source tree and rebuild your kernel manually. I'm completely out of the loop on latest stuff. Thanks for the tip though -- I will do some research on how do to that and hopefully get the new driver in there! Thanks for quick response!

dubhater, 2025-03-08:
It's easier than that. There are some instructions here: https://github.com/lwfinger/rtw88/blob/master/README.md

Or you could try the upcoming Ubuntu 25.04, which seems to have kernel 6.14.

cculianu, 2025-03-08:
OMFG I just read the guide. I'm so happy kernel APIs are stable now and that it's just this easy to install out-of-kernel drivers. Back in my day, when I still messed around heavily with this stuff, it was really not as well organized. Glad that in the intervening 20+ years of development some of the problems were solved! Ha ha... thanks for the guide. You are a lifesaver.

cculianu, 2025-03-08:
OK I did the install. I had to turn off secureboot since I have no idea how that works.. the driver has the exact same identical dmesg message in terms of versions as my previous one on my 6.8 kernel.

```
rtw_8821cu 3-1:1.0: Firmware version 24.11.0, H2C version 12
```

This was what I saw before in dmesg.. but I did get warnings about "out of kernel" drivers so I think it did load the right one. Should I be worried it loaded the old one or is this fine?  Thanks for your help.. I haven't done this in 20 years..

dubhater, 2025-03-08:
That's normal because it's using the same firmware files.

You can check lsmod. If you see modules with "rtw_" prefix, you are using those compiled from this repository. If you see modules with "rtw88_" prefix, you are using the modules that came with the kernel.

Note that it's possible to load a mix of both, which is a bad idea. But you should already have a file in /etc/modprobe.d/ which prevents that.

(The messages in dmesg don't follow this rule, the prefix is always "rtw_" there.)

cculianu, 2025-03-08:
Ahhh! Makes sense. Yes, you are right before I was using `rtw88_` prefixed modules (I remember this from messing around before), and indeed now they are `rtw_` without the 88. Great!

W00!

I'm hoping my conn drop issues are nuked. Will have to let this thing run for a while and see.. thanks man for the "tech support", ha ha.  I'm actually a C++ programmer but I haven't run Linux as a primary OS in some time (decade or more) so I'm rusty with what the latest and greatest state of the art is as far as system config goes, etc. This changed somewhat and matured since I was doing this back in the 2000's.

 Best!