lwfinger/rtw88 #79: make failed (fsleep error)

truezed, 2022-03-11:
Hello, I have encountered this error. 

```
truezed@cerberus:~/Downloads/rtw88$ make
make -C /lib/modules/5.4.0-100-generic/build M=/home/truezed/Downloads/rtw88 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-100-generic'
  CC [M]  /home/truezed/Downloads/rtw88/rtw8822c.o
/home/truezed/Downloads/rtw88/rtw8822c.c: In function ‘rtw8822c_txgapk_read_offset’:
/home/truezed/Downloads/rtw88/rtw8822c.c:1479:2: error: implicit declaration of function ‘fsleep’; did you mean ‘ssleep’? [-Werror=implicit-function-declaration]
  fsleep(1000);
  ^~~~~~
  ssleep
cc1: some warnings being treated as errors
scripts/Makefile.build:270: recipe for target '/home/truezed/Downloads/rtw88/rtw8822c.o' failed
make[2]: *** [/home/truezed/Downloads/rtw88/rtw8822c.o] Error 1
Makefile:1762: recipe for target '/home/truezed/Downloads/rtw88' failed
make[1]: *** [/home/truezed/Downloads/rtw88] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-100-generic'
Makefile:80: recipe for target 'all' failed
make: *** [all] Error 2
```
```
truezed@cerberus:~/Downloads/rtw88$ neofetch 
            .-/+oossssoo+/-.               truezed@cerberus 
        `:+ssssssssssssssssss+:`           ---------------- 
      -+ssssssssssssssssssyyssss+-         OS: Ubuntu 18.04.6 LTS x86_64 
    .ossssssssssssssssssdMMMNysssso.       Host: B550 AORUS ELITE AX V2 
   /ssssssssssshdmmNNmmyNMMMMhssssss/      Kernel: 5.4.0-100-generic 
  +ssssssssshmydMMMMMMMNddddyssssssss+     Uptime: 18 mins 
 /sssssssshNMMMyhhyyyyhmNMMMNhssssssss/    Packages: 2072 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Shell: bash 4.4.20 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   Resolution: 1920x1080 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   DE: GNOME 3.28.4 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   WM: GNOME Shell 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   WM Theme: Adwaita 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Theme: Ambiance [GTK2/3] 
 /sssssssshNMMMyhhyyyyhdNMMMNhssssssss/    Icons: Ubuntu-mono-dark [GTK2/3] 
  +sssssssssdmydMMMMMMMMddddyssssssss+     Terminal: gnome-terminal 
   /ssssssssssshdmNNNNmyNMMMMhssssss/      CPU: AMD Ryzen 5 5600X 6- (12) @ 3.700GHz 
    .ossssssssssssssssssdMMMNysssso.       GPU: NVIDIA GeForce GT 1030 
      -+sssssssssssssssssyyyssss+-         Memory: 2872MiB / 32117MiB 
        `:+ssssssssssssssssss+:` 
            .-/+oossssoo+/-.                                       

```

If you need more information, I will provide. Just tell me which commands you need to execute, I'm not very good at this (:

Thank you!

maxxchildress, 2022-03-16:
Having this same problem on MX Linux 19.4 Debian 10.11
```

/home/tinystranger/rtw88/rtw8822c.c: In function ‘rtw8822c_txgapk_read_offset’:
/home/tinystranger/rtw88/rtw8822c.c:1479:2: error: implicit declaration of function ‘fsleep’; did you mean ‘ssleep’? [-Werror=implicit-function-declaration]
  fsleep(1000);
  ^~~~~~
  ssleep
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-4.19.0-19-common/scripts/Makefile.build:309: /home/tinystranger/rtw88/rtw8822c.o] Error 1
make[3]: *** [/usr/src/linux-headers-4.19.0-19-common/Makefile:1561: _module_/home/tinystranger/rtw88] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-19-amd64'
make: *** [Makefile:80: all] Error 2
```

lwfinger, 2022-03-17:
There was a bug in main.h. The code was supposed to create fsleep for kernels older than 5.8 and certain older Fedora releases. The code was not testing to see if this was actually a Fedora system, and failed for your system.

I think I fixed it now. Do a 'git pull', 'make' and 'sudo make install'.

closed 2022-03-17 by truezed