lwfinger/rtw88 #84: mac80211.c make fail

1256-bits, 2022-05-07:
make80211.c fails to compile. It used to work on the same distro a few months ago.
Log:
` /lib/modules/5.4.191_1/build M=/home/user/rtw88 modules
make[1]: Entering directory '/usr/src/kernel-headers-5.4.191_1'
  CC [M]  /home/iris/rtw88/mac80211.o
/home/iris/rtw88/mac80211.c:898:19: warning: 'struct cfg80211_sar_specs' declared inside parameter list will not be visible outside of this definition or declaration
  898 |      const struct cfg80211_sar_specs *sar)
      |                   ^~~~~~~~~~~~~~~~~~
/home/iris/rtw88/mac80211.c: In function 'rtw_ops_set_sar_specs':
/home/iris/rtw88/mac80211.c:902:2: error: implicit declaration of function 'rtw_set_sar_specs'; did you mean 'rtw_ops_set_sar_specs'? [-Werror=implicit-function-declaration]
  902 |  rtw_set_sar_specs(rtwdev, sar);
      |  ^~~~~~~~~~~~~~~~~
      |  rtw_ops_set_sar_specs
At top level:
/home/iris/rtw88/mac80211.c:897:12: warning: 'rtw_ops_set_sar_specs' defined but not used [-Wunused-function]
  897 | static int rtw_ops_set_sar_specs(struct ieee80211_hw *hw,
      |            ^~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:262: /home/user/rtw88/mac80211.o] Error 1
make[1]: *** [Makefile:1734: /home/user/rtw88] Error 2
make[1]: Leaving directory '/usr/src/kernel-headers-5.4.191_1'
make: *** [Makefile:83: all] Error 2
`

1256-bits, 2022-05-08:
The issue doesn't seem to be present in April 7 commit.

lwfinger, 2022-05-09:
I found the problem and pushed a new version. Sorry for the problem.