lwfinger/rtw88 #85: "failed to send h2c command" errors in dmesg

lucassith, 2022-05-16:
Hi, I have noticed a drop in network performance when I connect via wifi on my laptop. I don't know when the driver stopped working but recently I noticed that wireless devices have a delay (mouse, keyboard, headphones). They have started to disconnect. After consulting 'dmesg' I noticed that the entire log is cluttered with the following entries:
`rtw_8822ce 0000:03:00.0: failed to send h2c command`.
`ieee80211 phy0: Hardware restart was requested`
and then the logs about the firmware crash dump:

```
[ 3636.253583] ------------[ cut here ]------------
[ 3636.253586] firmware crash, start reset and recover
[ 3636.253593] WARNING: CPU: 12 PID: 10 at /var/lib/dkms/rtw88/r131.ce97958/build/main.c:598 rtw_fw_recovery_work+0x17b/0x280 [rtw_core]
[ 3636.253605] Modules linked in: rtw_8822ce(OE) rtw_8822c(OE) rtw_pci(OE) rtw_core(OE) mac80211 libarc4 cfg80211 rndis_host cdc_ether usbnet mii cdc_acm ccm rfcomm snd_seq_dummy snd_hrtimer snd_seq xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c br_netfilter bridge stp llc overlay hid_logitech_hidpp hid_logitech_dj cmac algif_hash algif_skcipher snd_usb_audio af_alg snd_usbmidi_lib snd_rawmidi usbhid snd_seq_device bnep uvcvideo videobuf2_vmalloc btusb videobuf2_memops videobuf2_v4l2 btrtl btbcm videobuf2_common btintel videodev btmtk uas qrtr usb_storage mc bluetooth ecdh_generic intel_rapl_msr intel_rapl_common amdgpu snd_sof_amd_renoir uinput joydev snd_acp_config mousedev nvidia_uvm(POE) snd_sof_amd_acp edac_mce_amd asus_nb_wmi asus_wmi kvm_amd sparse_keymap snd_sof_pci hid_multitouch platform_profile wmi_bmof snd_sof snd_hda_codec_realtek snd_hda_codec_generic snd_soc_acpi
[ 3636.253644]  snd_hda_codec_hdmi kvm snd_hda_intel vfat ledtrig_audio fat snd_intel_dspcfg irqbypass snd_soc_core snd_intel_sdw_acpi crct10dif_pclmul snd_hda_codec crc32_pclmul ghash_clmulni_intel nvidia_drm(POE) snd_compress aesni_intel snd_hda_core ac97_bus snd_hwdep crypto_simd snd_pcm_dmaengine nvidia_modeset(POE) cryptd snd_pci_acp5x rapl snd_pcm gpu_sched sp5100_tco ucsi_acpi tpm_crb snd_rn_pci_acp3x drm_ttm_helper snd_timer typec_ucsi tpm_tis snd pcspkr typec nvidia(POE) tpm_tis_core r8168(OE) k10temp i2c_piix4 ccp snd_pci_acp3x ttm soundcore i2c_nvidia_gpu rfkill wmi roles video i2c_hid_acpi tpm i2c_hid asus_wireless mac_hid pinctrl_amd rng_core acpi_cpufreq squashfs loop fuse crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 serio_raw atkbd libps2 i8042 xhci_pci crc32c_intel xhci_pci_renesas serio [last unloaded: libarc4]
[ 3636.253678] CPU: 12 PID: 10 Comm: kworker/u32:1 Tainted: P        W  OE     5.17.6-1-MANJARO #1 1e1d23d0a05f307a20d237ae367072b447c14fc7
[ 3636.253680] Hardware name: ASUSTeK COMPUTER INC. ASUS TUF Gaming A17 FA706IU_FA706IU/FA706IU, BIOS FA706IU.316 03/12/2021
[ 3636.253682] Workqueue: phy0 rtw_fw_recovery_work [rtw_core]
[ 3636.253689] RIP: 0010:rtw_fw_recovery_work+0x17b/0x280 [rtw_core]
[ 3636.253696] Code: c7 83 e0 eb ff ff 00 00 00 00 48 c7 83 d8 eb ff ff 00 00 00 00 48 8b 40 70 e8 b1 48 ca f8 48 c7 c7 68 f9 d7 c2 e8 92 cd 85 f8 <0f> 0b e8 4e fa dc f7 48 8b bb b0 b5 ff ff 4c 89 e1 31 f6 48 c7 c2
[ 3636.253697] RSP: 0018:ffffaada00137e68 EFLAGS: 00010246
[ 3636.253699] RAX: 0000000000000000 RBX: ffff92f3a9886b30 RCX: 0000000000000000
[ 3636.253699] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[ 3636.253700] RBP: ffffaada10411010 R08: 0000000000000000 R09: 0000000000000000
[ 3636.253701] R10: 0000000000000000 R11: 0000000000000000 R12: ffff92f3a98820e0
[ 3636.253701] R13: ffff92f3a9886a40 R14: 0000000000003000 R15: ffff92f343f23425
[ 3636.253702] FS:  0000000000000000(0000) GS:ffff92f65f900000(0000) knlGS:0000000000000000
[ 3636.253703] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 3636.253704] CR2: 0000221500a45000 CR3: 000000014ce10000 CR4: 0000000000350ee0
[ 3636.253705] Call Trace:
[ 3636.253707]  <TASK>
[ 3636.253709]  process_one_work+0x1e5/0x3b0
[ 3636.253713]  worker_thread+0x50/0x3a0
[ 3636.253714]  ? rescuer_thread+0x3a0/0x3a0
[ 3636.253715]  kthread+0xd8/0x100
[ 3636.253717]  ? kthread_complete_and_exit+0x20/0x20
[ 3636.253719]  ret_from_fork+0x22/0x30
[ 3636.253722]  </TASK>
[ 3636.253723] ---[ end trace 0000000000000000 ]---
```

System info:

```
 λ System:
  Host: Asus-TUF-17 Kernel: 5.17.6-1-MANJARO arch: x86_64 bits: 64
    Desktop: Xfce v: 4.16.0 Distro: Manjaro Linux
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet
    driver: r8168
  IF: enp2s0 state: down mac: f0:2f:74:4b:0c:f7
  Device-2: Realtek RTL8822CE 802.11ac PCIe Wireless Network Adapter
    driver: rtw_8822cesudo modinfo rtw88_8822ce
```

modinfo:

```
filename:       /lib/modules/5.17.6-1-MANJARO/updates/dkms/rtw_8822ce.ko.xz
license:        Dual BSD/GPL
description:    Realtek 802.11ac wireless 8822ce driver
author:         Realtek Corporation
srcversion:     BF9D29A6BD5A0222292439C
alias:          pci:v000010ECd0000C82Fsv*sd*bc*sc*i*
alias:          pci:v000010ECd0000C822sv*sd*bc*sc*i*
depends:        rtw_pci,rtw_8822c
retpoline:      Y
name:           rtw_8822ce
vermagic:       5.17.6-1-MANJARO SMP preempt mod_unload
```

rtw88 built from git

lucassith, 2022-05-16:
I have read the information here:
https://www.spinics.net/lists/linux-wireless/msg222874.html

I did the update to 5.18 and I've added kernel modules to blacklist:

```
blacklist rtw88_8822ce
blacklist rtw88_8822c
blacklist rtw88_pci
blacklist rtw88_core
```

I then pulled the repository again, I did `make` then `sudo make install` + reboot

Now, my lsmod looks like that:
```
 λ _ lsmod | grep 88
amdgpu               8921088  2
rtw_8822ce             16384  0
rtw_8822c             491520  1 rtw_8822ce
rtw_pci                36864  1 rtw_8822ce
rtw_core              278528  2 rtw_8822c,rtw_pci
```

So, from what I understand, I now use the module from Github, not the kernel ones.
However, this issue still persists. And again, I do have WIFI but the performance has clearly degraded, sometimes it takes a lot of time to load the page, the speedtest hangs for a second. The same happens for a wireless mouse, the cursor has some latency every 5-10 seconds.

Logs from dmesg:

```
[  287.697720] ieee80211 phy0: Hardware restart was requested
[  288.137955] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  288.140980] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  288.144261] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  288.147281] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  288.150439] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  288.153528] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  288.156549] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  288.159563] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  289.328117] ------------[ cut here ]------------
[  289.328120] firmware crash, start reset and recover
[  289.328147] WARNING: CPU: 3 PID: 135 at /home/lucassith/temps/rtw88/main.c:598 rtw_fw_recovery_work+0x17b/0x280 [rtw_core]
[  289.328162] Modules linked in: rfcomm ccm snd_seq_dummy snd_hrtimer snd_seq hid_logitech_hidpp hid_logitech_dj uas xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat cmac nf_conntrack algif_hash algif_skcipher nf_defrag_ipv6 af_alg nf_defrag_ipv4 libcrc32c snd_usb_audio snd_usbmidi_lib snd_rawmidi snd_seq_device br_netfilter usbhid bnep bridge btusb uvcvideo btrtl btbcm stp videobuf2_vmalloc llc btintel videobuf2_memops btmtk videobuf2_v4l2 videobuf2_common bluetooth videodev mc usb_storage ecdh_generic overlay amdgpu snd_sof_amd_renoir snd_sof_amd_acp rtw_8822ce(OE) snd_sof_pci qrtr rtw_8822c(OE) snd_sof snd_hda_codec_realtek intel_rapl_msr snd_sof_utils snd_hda_codec_generic intel_rapl_common snd_soc_core rtw_pci(OE) ledtrig_audio snd_hda_codec_hdmi snd_compress ac97_bus snd_pcm_dmaengine rtw_core(OE) gpu_sched snd_hda_intel uinput snd_pci_acp5x edac_mce_amd snd_intel_dspcfg drm_ttm_helper snd_rn_pci_acp3x ttm snd_intel_sdw_acpi
[  289.328199]  snd_acp_config nvidia_uvm(POE) snd_hda_codec drm_dp_helper kvm_amd snd_soc_acpi snd_pci_acp3x ccp cdc_acm snd_hda_core joydev mousedev mac80211 snd_hwdep kvm asus_nb_wmi snd_pcm asus_wmi irqbypass sparse_keymap hid_multitouch snd_timer vfat libarc4 crct10dif_pclmul wmi_bmof platform_profile squashfs crc32_pclmul nvidia_drm(POE) fat snd nvidia_modeset(POE) loop ghash_clmulni_intel cfg80211 i2c_nvidia_gpu soundcore ucsi_acpi aesni_intel sp5100_tco typec_ucsi crypto_simd cryptd rapl nvidia(POE) pcspkr typec tpm_crb rfkill k10temp roles i2c_piix4 i2c_hid_acpi tpm_tis i2c_hid tpm_tis_core tpm mac_hid video pinctrl_amd asus_wireless acpi_cpufreq wmi rng_core fuse crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 serio_raw atkbd libps2 vivaldi_fmap crc32c_intel xhci_pci xhci_pci_renesas i8042 serio
[  289.328243] CPU: 3 PID: 135 Comm: kworker/u32:4 Tainted: P        W  OE     5.18.0-1-MANJARO #1 152a9506397b9bb0938490486f4c9545c016b805
[  289.328245] Hardware name: ASUSTeK COMPUTER INC. ASUS TUF Gaming A17 FA706IU_FA706IU/FA706IU, BIOS FA706IU.316 03/12/2021
[  289.328248] Workqueue: phy0 rtw_fw_recovery_work [rtw_core]
[  289.328256] RIP: 0010:rtw_fw_recovery_work+0x17b/0x280 [rtw_core]
[  289.328265] Code: c7 83 e0 eb ff ff 00 00 00 00 48 c7 83 d8 eb ff ff 00 00 00 00 48 8b 40 70 e8 e1 ef 6f f7 48 c7 c7 18 75 32 c3 e8 05 32 2c f7 <0f> 0b e8 fe a2 82 f6 48 8b bb b0 b5 ff ff 4c 89 e1 31 f6 48 c7 c2
[  289.328266] RSP: 0018:ffff9d1fc069fe68 EFLAGS: 00010246
[  289.328268] RAX: 0000000000000000 RBX: ffff905ca9a36b30 RCX: 0000000000000000
[  289.328269] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[  289.328270] RBP: ffff9d1fd4647010 R08: 0000000000000000 R09: 0000000000000000
[  289.328271] R10: 0000000000000000 R11: 0000000000000000 R12: ffff905ca9a320e0
[  289.328272] R13: ffff905ca9a36a40 R14: 0000000000003000 R15: ffff905c892ff2c5
[  289.328273] FS:  0000000000000000(0000) GS:ffff905f9f6c0000(0000) knlGS:0000000000000000
[  289.328274] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  289.328275] CR2: 00003dac0037c000 CR3: 000000015ee10000 CR4: 0000000000350ee0
[  289.328276] Call Trace:
[  289.328279]  <TASK>
[  289.328281]  process_one_work+0x1e5/0x3b0
[  289.328284]  ? rescuer_thread+0x3a0/0x3a0
[  289.328286]  worker_thread+0x50/0x3b0
[  289.328288]  ? rescuer_thread+0x3a0/0x3a0
[  289.328289]  kthread+0xd8/0x100
[  289.328291]  ? kthread_complete_and_exit+0x20/0x20
[  289.328293]  ret_from_fork+0x22/0x30
[  289.328297]  </TASK>
[  289.328298] ---[ end trace 0000000000000000 ]---
[  289.929194] ieee80211 phy0: Hardware restart was requested
[... 13 lines cut ...]
[... cut ...]
[  296.085111] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  296.085112] CR2: 00002ea701e83000 CR3: 000000015ee10000 CR4: 0000000000350ee0
[  296.085113] Call Trace:
[  296.085114]  <TASK>
[  296.085116]  process_one_work+0x1e5/0x3b0
[  296.085119]  worker_thread+0x50/0x3b0
[  296.085121]  ? rescuer_thread+0x3a0/0x3a0
[  296.085122]  kthread+0xd8/0x100
[  296.085124]  ? kthread_complete_and_exit+0x20/0x20
[  296.085126]  ret_from_fork+0x22/0x30
[  296.085129]  </TASK>
[  296.085129] ---[ end trace 0000000000000000 ]---
```

lwfinger, 2022-05-16:
I do not have any knowledge of the internals of the chips, thus I have no means of interpreting your dumps. I can fix kernel BUGS and WARNINGS, but not these.

You need to follow the last paragraph of README.md to prepare a report that the Realtek engineers will see.

lucassith, 2022-05-16:
@lwfinger I understand, thank you. I will do as you say.

closed 2022-05-16 by lucassith