lwfinger/rtw88 #86: Fail on kernel 4.18.0-372.9.1

ak-ferhat, 2022-05-27:
Hi
I am on Almalinux 8.5 
The last update upgrades the kernel to 4.18.0-372.9.1 and the "make" fails. Before all was OK with4.18.0-348.23.1
Here is the output of make:

make -C /lib/modules/4.18.0-372.9.1.el8.x86_64/build M=/home/abdou_2022/8821CE_new/rtw88 modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-372.9.1.el8.x86_64'
  CC [M]  /home/abdou_2022/8821CE_new/rtw88/main.o
In file included from /home/abdou_2022/8821CE_new/rtw88/main.c:7:
/home/abdou_2022/8821CE_new/rtw88/main.h:41:20: error: redefinition of ‘fsleep’
 static inline void fsleep(unsigned long usecs)
                    ^~~~~~
In file included from ./include/linux/netdevice.h:30,
                 from ./include/linux/etherdevice.h:26,
                 from ./include/linux/ieee80211.h:23,
                 from ./include/net/mac80211.h:20,
                 from /home/abdou_2022/8821CE_new/rtw88/main.h:15,
                 from /home/abdou_2022/8821CE_new/rtw88/main.c:7:
./include/linux/delay.h:68:20: note: previous definition of ‘fsleep’ was here
 static inline void fsleep(unsigned long usecs)
                    ^~~~~~
make[2]: *** [scripts/Makefile.build:316: /home/abdou_2022/8821CE_new/rtw88/main.o] Error 1
make[1]: *** [Makefile:1577: _module_/home/abdou_2022/8821CE_new/rtw88] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-372.9.1.el8.x86_64'
make: *** [Makefile:83: all] Error 2

Lollixzc, 2022-06-03:
Short answer: You need to use a newer kernel >=5.4 because there are some dependencies needed for this module to compile, as written in [README.md](https://github.com/lwfinger/rtw88/blob/master/README.md).
See #83 for more information. 
A solution is to upgrade to a newer kernel or to use the [backports project](https://backports.wiki.kernel.org/index.php/Main_Page).

ak-ferhat, 2022-06-04:
> Short answer: You need to use a newer kernel >=5.4 because there are some dependencies needed for this module to compile, as written in [README.md](https://github.com/lwfinger/rtw88/blob/master/README.md). See #83 for more information. A solution is to upgrade to a newer kernel or to use the [backports project](https://backports.wiki.kernel.org/index.php/Main_Page).
> 
> For @lwfinger this issue seems related to #73. It suggest to remove fsleep redefinition on [main.h](https://github.com/lwfinger/rtw88/blob/master/main.h#L41) but I don't think that's the solution

When I remove the redifintion of fsleep in main.h

[image]

And when I make the change

#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 14, 0)
static void rtw_ops_mgd_prepare_tx(struct ieee80211_hw *hw,
				   struct ieee80211_vif *vif,
    #if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,14,0))
				   struct ieee80211_prep_tx_info *info)
    #else
				   u16 duration)
    #endif
#else
static void rtw_ops_mgd_prepare_tx(struct ieee80211_hw *hw,
				   struct ieee80211_vif *vif)
#endif 

in mac80211.c line 671, the module is built but when I load it the system freezes.

lwfinger, 2022-06-04:
Funny, my code in that section looks like

#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0)
static void rtw_ops_mgd_prepare_tx(struct ieee80211_hw *hw,
                                   struct ieee80211_vif *vif,
    #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,14,0))
                                   struct ieee80211_prep_tx_info *info)
    #else
                                   u16 duration)
    #endif
#else
static void rtw_ops_mgd_prepare_tx(struct ieee80211_hw *hw,
                                   struct ieee80211_vif *vif)
#endif

I am not sure it will run, but did you do a 'git pull'?

ak-ferhat, 2022-06-04:
Yes I did.
No build error but with modprobe the system freezes. It say panic kernel

lwfinger, 2022-06-05:
Unless you can provide me with a netconsole dump of that panic, or some other listing of the reason or backtrace, there is little I can do. 

I am actually surprised that a laptop new enough to have an rtw88 chip will actually work with 4.18. I would expect that the features of the CPU would not be supported.

ak-ferhat, 2022-06-05:
> Unless you can provide me with a netconsole dump of that panic, or some other listing of the reason or backtrace, there is little I can do.
>
> I am actually surprised that a laptop new enough to have [...]

Hi 
All Redhat 8 derivative work with kernel 4.18. i think Redhat 9 work with 5.14
What I have done after modifying main.h and mac80211.c is 
> make
The make is done with success. When I do
> make install
I got error first time and the second time I got success. Surprising.

[image]

When I do
> modprobe rtw_8821ce
the system freezes and CapsLock blinks.
When I reboot, the system reach a point wher it freezes with the following message

[image]

That all.
With the preceding kernel 4.18.0-348 there is no problem. It is my rescue kernel

Best regards

ak-ferhat, 2022-07-08:
Hi
With the last kernel update to the version "4.18.0-372.13.1", the build is OK.
I have just to make the cited  modifications  in main.h and mac80211.c 
Thank you

lwfinger, 2022-07-08:
Please run the command 'git diff > diffs.txt' and attach the file diffs.txt. I want to see what works for you.